syssi/esphome-jk-bms #188: B2A8S20P wich port )

claib, 2022-10-10:
hi there
got a brand new B2A8S20P which has 2 Ports : RS485 (3 Pin ) and GPS ( 4 Pin )
which to use ? and which pins ?
Both connectors do have ca 1mm pitch

best regards
Christoph

syssi, 2022-10-10:
Could you provide a photo of the connectors? I'm confused. ;-)

claib, 2022-10-10:
[image]

claib, 2022-10-10:
the RS485 Port is a 3 pin, and i'm wondering which is A, B and ground ;-)

syssi, 2022-10-20:
Good question? I will ask Hankzor about some details of the new connector.

claib, 2022-10-20:
i can confirm that it is the 4 port connector labeled " GPS ". The pins ( view from upside ) are: GND, TX, RX, V+
it is a TTL level interface, preferably connected isolated.

syssi, 2022-10-22:
The GPS (= UART-TTL) port is well tested and supported by this component. If you are looking for a short term solution you should use this port.

syssi, 2022-11-07:
I talked to Hankzor BMS but didn't got infos about RS485 labeled port.

Let's close this issue for now.

closed 2022-11-07 by syssi