syssi/esphome-jk-bms #189: esphome-jk-bms - does not work with 2 jkbms managed by 1 esp32

Manu0171discord, 2022-10-11:
Hi,
I tried the example yaml file to manage 2 jkbms with 1 esp32
jkbms is compliant as described into the yaml file

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example-multiple-devices.yaml

The first jkbms which is kept by the BT connection of the esp32 is the only one
no connection with the second jkbms (different mac address from the 1st one)

In summary, only can have data into HA/esphome for 1 jkbms (bms0 and bms1 values are the same)

syssi, 2022-10-11:
Please check your ESPHome version. There was a bug in the past which leads to the described behavior. Please update ESPHome to the latest release + compile & flash the node again.

Bandit-17, 2022-10-11:
Hello Sebastian,
we are using the dev version of esphome with anaconda (version: 2022.10.0-dev)

here is the file we are working on
[teste2bmsble.txt](https://github.com/syssi/esphome-jk-bms/files/9756452/teste2bmsble.txt)

unfortunately we have the values ​​of the 1st BMS which is grafted onto the 2nd (see capture)
[image]

Bandit-17, 2022-10-11:
we try to integrate the BLE part of the JKBMS in the Pvbrain, to recover the information not available with the RS485.

syssi, 2022-10-11:
This is strange. Could you try the IDF framework instead if Arduino?

syssi, 2022-10-11:
Another idea: May be multiple BLE clients are broken (again) at ESPHome 2022.10.0. Please downgrade to `2022.8.3` by:

``` 
pip3 install esphome==2022.8.3
```

And use the ESPHome 2022.8.3 compatible release of this external component:

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@1.3.0
```

Are you able to monitor two JK-BMS simultaneously now?

Manu0171discord, 2022-10-11:
We, bandit17 and I tried with your instructions but it does not work neither this way.

syssi, 2022-10-11:
Alright. I'm back home in a few days and will try to reproduce the issue on my testbench.

Manu0171discord, 2022-10-11:
We thank you very much for your time and work you've done
there is no emergency for sure but if you can reproduce the case, it is very nice.

syssi, 2022-10-17:
@proasnet May be this issue is relevant for you too or you are able to verify multiple BLE client connections are still working.

Long story short: @Bandit-17 and @Manu0171discord are trying to talk to 2 BMS using one ESP32. If I remember correctly you are doing the same successfully(?).

proasnet, 2022-10-17:
Yes, right.

proasnet, 2022-10-17:
But, one month before some ESPHome update, both BMS worked via BT on the one ESP32. Now not.

syssi, 2022-10-17:
```
[12:49:57][I][app:102]: ESPHome version 2022.9.1 compiled on Sep 22 2022, 12:47:24
# https://github.com/syssi/esphome-jk-bms/issues/160#issuecomment-1254943112
```

proasnet, 2022-10-17:
Yes, we have addressed this. This seems to be the problem, not in the wifi, but as described in this post
@Bandit-17 and @Manu0171discord are trying to talk to 2 BMS using one ESP32.

proasnet, 2022-10-17:
As I read, is needed downgrade ESPHome?

syssi, 2022-10-17:
Could you try to compare the behavior of the most recent ESPHome version:

```
pip3 install esphome==2022.9.4 -U
```

plus

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@main
```

Versus the ESPHome version you did use successfully in the past:

```
pip3 install esphome==2022.9.1 -U
```

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@1.3.0
```

Do you see a difference?

proasnet, 2022-10-17:
I am sorry, where I use this command?
In the terminal, not working...
bash: command not found
pip3 install esphome==2022.9.1 -U
I am lama :D

syssi, 2022-10-17:
If you use the Home Assistant Addon are you able to up-/downgrade the ESPHome version? If not I will explain how to install `esphome` manually and use it (for testing) at the terminal.

proasnet, 2022-10-17:
I can not downgrade ESPHome. In addons, I can get only latest version.

syssi, 2022-10-17:
Alright. Do you have access to a linux host? You could try to follow this tutorial:

[image]

proasnet, 2022-10-17:
You mean another PC with linux? Not now, I am in work with win PC, but at home I have PC with ubuntu

proasnet, 2022-10-17:
I have the option to install ESPHome in docker in the synology server, but then I don't know how to get to his files, I don't understand these Linux things that much.
[image]

proasnet, 2022-10-19:
I am sorry, my ubuntu pc not working. I cant test it for you.
Now I am trying update ESPHome to 2022.10.0 with backup previous.

proasnet, 2022-10-25:
New status:
The second BMS has after some time the same values as the first BMS. The real values of the second BMS is not readed.
The second BMS is not really connected. LED on device flashing.

syssi, 2022-10-25:
You are using the most recent ESPHome release now because you are unable to downgrade to a previous version. Correct?

syssi, 2022-11-06:
It looks like the BLE client component of ESPHome 2022.10.1 is broken. Please be careful: https://github.com/esphome/issues/issues/3729

syssi, 2022-11-06:
I did start my second JK-BMS today. First findings:

The simultaneous communication with multiple devices is working fine:

- ESPHome 2022.8.3 & Custom component version 1.3.0
- ESPHome 2022.9.0 & Custom component version 1.3.0
- ESPHome 2022.9.1 & Custom component version 1.3.0
- ESPHome 2022.9.2 & Custom component version 1.3.0

Broken versions:

- ESPHome 2022.9.3 & Custom component version `main`
- ESPHome 2022.10.1 & Custom component version `main`
- ESPHome 2022.10.2 & Custom component version `main`

proasnet, 2022-11-06:
that's a good find.  Will it be fixed in the new version?  Or is it necessary to downgrade?
Very thanks.

syssi, 2022-11-06:
The root cause is identified and fixed here: https://github.com/esphome/esphome/commit/0e66c899ce57b304a7697bf4859806e78a07325c

It should work (again) as soon ESPHome `dev` gets released in a few days!

Softface, 2022-11-06:
gooood !
ᐧ

Fierofiend, 2022-11-07:
I am still having this issue with 2022.10.2.

Fierofiend, 2022-11-17:
I just updated to 2022.11.0 and it seems to be fixed.  I am running 10.X-W (10.03) and 8.X (8.20G).

proasnet, 2022-11-17:
Hi @Fierofiend , I am updated HA CORE, ESPHOME and jk-bms node today to last versions, but values of both bms is returned as the same from one bms, no change from previous versions. I see log from jk-bms and see the same values from both.
bms0 and bms1 has the same values, but physicaly has not the same values.

proasnet, 2022-11-17:
[image]

syssi, 2022-11-17:
@proasnet On boot-up the ESPHome node reports the used ESPHome version at the log. Could you make sure you are using a 2022.11 build? I don't think so.

proasnet, 2022-11-17:
You think this?
```
INFO Reading configuration /config/esphome/jk_bms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms.local using esphome API
INFO Successfully connected to jk-bms.local
[06:21:39][I][app:102]: ESPHome version 2022.10.2 compiled on Nov  4 2022, 09:16:55
[06:21:39][I][app:104]: Project syssi.esphome-jk-bms version 1.2.0
```

syssi, 2022-11-17:
Yes! This `2022.10.2` indicates you are not using ESPHome 2022.11 and you did not reflash the ESP recently (`Nov  4 2022, 09:16:55`). Please give it another try! :-)

proasnet, 2022-11-17:
Is interesting, that if I update ESPHome, I will see "UPDATE" on each node, I use it, but is not updated new version of ESPHome. Now, I know it :)
So, cleaned build files, installed to node again, and result is:
Working !!! 
Thank you very much !

commit eed5cf56 (2022-11-17): Require ESPHome 2022.11 to fix BLE issues (Fixes: #189)
closed 2022-11-17 by syssi