syssi/esphome-jk-bms #193: Second BMS still disconected after update

proasnet, 2022-10-17:
Hello @syssi,
after some update esp-home, my second JKBMS was connected after a long random time, and disconected.
Now, I am trying set longer time from 5s to 10s in setup, and is not possible Install to ESP, back to 5s is the same problem.

My setup:
```
substitutions:
  name: jk-bms
  bms0: "${name} bms0"
  bms1: "${name} bms1"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:8C:E4:51:FA
  bms1_mac_address: C8:47:8C:E4:52:9D
  bms0_protocol_version: JK02
  bms1_protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.2.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG
  logs:
    jk_bms_ble: INFO

api:

esp32_ble_tracker:

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 10s
    id: bms0
  - ble_client_id: client1
    protocol_version: ${bms1_protocol_version}
    throttle: 10s
    id: bms1

binary_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balancing:
      name: "${bms0} balancing"
[... 207 lines cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o: in function `esphome::ble_client::BLEClient::setup()':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:16: undefined reference to `esphome::esp32_ble_client::BLEClientBase::setup()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o: in function `esphome::ble_client::BLEClient::loop()':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:21: undefined reference to `esphome::esp32_ble_client::BLEClientBase::loop()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o: in function `esphome::ble_client::BLEClient::parse_device(esphome::esp32_ble_tracker::ESPBTDevice const&)':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:34: undefined reference to `esphome::esp32_ble_client::BLEClientBase::parse_device(esphome::esp32_ble_tracker::ESPBTDevice const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o: in function `esphome::ble_client::BLEClient::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*)':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:54: undefined reference to `esphome::esp32_ble_client::BLEClientBase::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/jk-bms/src/esphome/components/ble_client/ble_client.cpp:62: undefined reference to `esphome::esp32_ble_client::BLEService::~BLEService()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o:(.rodata._ZTVN7esphome10ble_client9BLEClientE[vtable for esphome::ble_client::BLEClient]+0x14): undefined reference to `esphome::esp32_ble_client::BLEClientBase::gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o:(.rodata._ZTVN7esphome10ble_client9BLEClientE[vtable for esphome::ble_client::BLEClient]+0x18): undefined reference to `esphome::esp32_ble_client::BLEClientBase::connect()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o:(.rodata._ZTVN7esphome10ble_client9BLEClientE[vtable for esphome::ble_client::BLEClient]+0x28): undefined reference to `esphome::esp32_ble_client::BLEClientBase::get_setup_priority() const'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/ble_client/ble_client.o:(.rodata._ZTVN7esphome10ble_client9BLEClientE[vtable for esphome::ble_client::BLEClient]+0x44): undefined reference to `non-virtual thunk to esphome::esp32_ble_client::BLEClientBase::get_setup_priority() const'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.o:(.literal._ZN7esphome10jk_bms_ble8JkBmsBle19gattc_event_handlerE20esp_gattc_cb_event_thP24esp_ble_gattc_cb_param_t+0x18): undefined reference to `esphome::esp32_ble_client::BLEClientBase::get_characteristic(unsigned short, unsigned short)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.o: in function `esphome::jk_bms_ble::JkBmsBle::write_register(unsigned char, unsigned int, unsigned char)':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:118: undefined reference to `esphome::esp32_ble_client::BLEClientBase::address_str[abi:cxx11]() const'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.o: in function `esphome::jk_bms_ble::JkBmsBle::gattc_event_handler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*)':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:144: undefined reference to `esphome::esp32_ble_client::BLEClientBase::get_characteristic(unsigned short, unsigned short)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /config/esphome/.esphome/build/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:146: undefined reference to `esphome::esp32_ble_client::BLEClientBase::address_str[abi:cxx11]() const'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/jk-bms/.pioenvs/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.o: in function `esphome::jk_bms_ble::JkBmsBle::update()':
/config/esphome/.esphome/build/jk-bms/src/esphome/components/jk_bms_ble/jk_bms_ble.cpp:342: undefined reference to `esphome::esp32_ble_client::BLEClientBase::address_str[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
*** [/data/jk-bms/.pioenvs/jk-bms/firmware.elf] Error 1
========================= [FAILED] Took 136.28 seconds =========================
```

syssi, 2022-10-17:
```
esphome clean jk_bms.yaml
esphome run jk_bms.yaml
```

should fix the build issue.

Manu0171discord, 2022-10-17:
Hello proasnet
Did you try to flash tour esp without being too near your jkbms ?
I think you need yo take your esp module and try to flash It in another place.
BT can impact your flash operation with ota.

proasnet, 2022-10-17:
> Hello proasnet Did you try to flash your esp without being too near your jkbms ? I think you need yo take your esp module and try to flash It in another place. BT can impact your flash operation with ota.

ESP is connected via USB to HA server. I am flashing by serial interface, not over OTA.

closed 2022-10-17 by proasnet

Manu0171discord, 2022-10-17:
> I used in ESP Home for jk-bms "Clean Build files" in three dots for this device, and in next, Install again, compiled, flashed without errors.

And your esp can manage both jkbms without any problems?

proasnet, 2022-10-17:
Not now. Only one working fine. The second BMS was connected sporadic. Sometime working. I have this condition for the last month. Both BMS worked perfectly for me until then. Then there was some ESPHome update, and since then there is a problem