syssi/esphome-jk-bms #199: BLE notifications & unknown messages

zbyndatvrd, 2022-10-19:
Hi,
I'm receiving plenty of notifications and unknown messages after updating the esp. It was working like a charm previously. 
Any idea, please, what could cause it? 

(example below: part of the log)

...
`[15:36:19][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.71.32.0D.31.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.33.0D.32.0D.32.0D.31.0D.31.0D.31.0D.31.0D.31.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.31.0D.03.00.02.04.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.14.D3.00.00.7A.3C.03.00.55.0F (128)
[15:36:19][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.80.29.03.00.A0.7A.03.00 (22)
[15:36:20][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3C.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.13.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.52.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:20][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.99 (22)
[15:36:20][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.72.32.0D.32.0D.33.0D.32.0D.32.0D.32.0D.33.0D.32.0D.32.0D.32.0D.31.0D.31.0D.32.0D.31.0D.31.0D.31.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.32.0D.03.00.02.04.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.1A.D3.00.00.93.3C.03.00.F2.10 (128)
[15:36:20][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.80.29.03.00.A0.7A.03.00 (22)
[15:36:20][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3C.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.15.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.57.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:20][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.61 (22)
[15:36:20][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.73.32.0D.32.0D.32.0D.33.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.32.0D.03.00.02.04.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.22.D3.00.00.E2.93.03.00.8F.12 (128)
[15:36:20][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.81.29.03.00.A0.7A.03.00 (22)
[15:36:21][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3D.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.17.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.5C.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:21][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.BC (22)
[15:36:21][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.74.32.0D.32.0D.33.0D.32.0D.32.0D.32.0D.32.0D.33.0D.33.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.32.0D.03.00.02.04.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.20.D3.00.00.0F.EB.03.00.8F.12 (128)
[15:36:21][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.82.29.03.00.A0.7A.03.00 (22)
[15:36:21][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3D.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.17.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.61.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:21][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.48 (22)
[15:36:21][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.75.32.0D.32.0D.33.0D.33.0D.32.0D.32.0D.33.0D.32.0D.33.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.32.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.32.0D.01.00.02.00.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.20.D3.00.00.77.42.04.00.2D.14 (128)
[15:36:22][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.82.29.03.00.A0.7A.03.00 (22)
[15:36:22][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3E.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.19.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.67.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:22][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.AD (22)
[15:36:22][D][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.76.32.0D.32.0D.33.0D.33.0D.32.0D.32.0D.33.0D.32.0D.33.0D.33.0D.32.0D.32.0D.32.0D.33.0D.32.0D.32.0D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.FF.00.00.32.0D.01.00.02.00.3D.00.3D.00.40.00.40.00.3C.00.3B.00.3C.00.3B.00.3B.00.3C.00.3B.00.3B.00.3B.00.3C.00.3C.00.3B.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.24.D3.00.00.8A.42.04.00.2D.14 (128)
[15:36:22][D][jk_bms_ble:220]: Notification received: 00.00.CD.00.D0.00.E5.00.00.00.00.00.00.5A.83.29.03.00.A0.7A.03.00 (22)
[15:36:22][D][jk_bms_ble:220]: Notification received: 41.00.00.00.9F.9E.E3.00.64.00.A0.02.3E.D6.DC.00.01.01.7A.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.09.04.19.00.00.00.A2.A2.41.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00.6C.1C.F1.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:22][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.CD (22)
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 2': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 3': Sending state 3.37900 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 4': Sending state 3.37900 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 5': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 6': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 7': Sending state 3.37900 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 8': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 9': Sending state 3.37900 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 10': Sending state 3.37900 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 11': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 12': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 13': Sending state 3.37800 V with 3 decimals of accuracy
[15:36:22][D][sensor:127]: 'jk-bms cell voltage 14': Sending state 3.37900 V with 3 decimals of accuracy
[... cut ...] 0.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:36:38][D][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.C0 (22)
[15:36:38][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 3.38000 V with 3 decimals of accuracy
[15:36:38][D][sensor:127]: 'jk-bms cell voltage 2': Sending state 3.38000 V with 3 decimals of accuracy
[15:36:38][D][sensor:127]: 'jk-bms cell voltage 3': Sending state 3.38000 V with 3 decimals of accuracy`
...

syssi, 2022-10-19:
Please decrease the log level by:

```
logger:
  level: INFO
```

syssi, 2022-10-19:
Both messages are fine if you use the log level `DEBUG` because the output should help to improve or debug the implementation. If you want to use the implementation in production the log level `INFO` is preferably.

zbyndatvrd, 2022-10-19:
yeah, that did the trick. 

I've got a bit overwhelmed looking into the default log outputs :)

Thanks!