syssi/esphome-jk-bms #203: Very interesting data from JKBS - over data?

proasnet, 2022-11-04:
Hi, today I found in a history, and in real time for some seconds very interesting data from BMS. I thing, that is bad calculated in esp, not in bms. Please see attached picture. I see it on my both BMS, but on both with issue
https://github.com/syssi/esphome-jk-bms/issues/189
My battery is 280Ah, and my max load currend is 25A, max charging current in this time 30A
[image]

syssi, 2022-11-04:
Could you enable the debug output of the `uart` component and provide some logs? What has happend at 20 o'clock?

proasnet, 2022-11-04:
Well, I checked the values ​​via mobile phone directly from the bms and everything is normal there. I had to unplug the esp from power to connect. After reconnecting the esp, everything works normally from one bms to ha as both, the other one does not give data.
I don't remember any extraordinary event yesterday at 8 pm.

syssi, 2022-11-17:
Let's close the issue for now. Feel free to re-open the issue if you hit the problem again.

closed 2022-11-17 by syssi