syssi/esphome-jk-bms #204: Complete Connection with arduino

Kamalkant1234, 2022-11-04:
Kindly share complete connection and codes sketch with Arduino.

syssi, 2022-11-04:
I don't understand your question. Can you be more specific about what you're missing?

syssi, 2022-11-04:
Did you see this paragraph of the docs? https://github.com/syssi/esphome-jk-bms#schematics

Kamalkant1234, 2022-11-04:
Please Help !!
I am using JK-BD6A20S6P BMS, and I want to view BMS data on serial monitor with the help of Arduino IDE, using two data wires of BMS. Please provide the detail which pins will be use of Arduino pins, and other modules that will be used in this project ??, as a PDF please share mentioned details. Need your help here please support !!
[image]
Have you used any other software or application for these kind of sequence output ?

syssi, 2022-11-04:
What kind of microcontroller do you want to use?

Kamalkant1234, 2022-11-05:
> What kind of microcontroller do you want to use?

Arduino UNO or NodeMcu/esp32.

syssi, 2022-11-05:
Please use an ESP32 and flash the `esp32-example.yaml` by using `esphome run esp32-example.yaml`. You have to install ESPHome first. There is no INO sketch required.

Kamalkant1234, 2022-11-05:
And after that ??

syssi, 2022-11-05:
You are able to monitor the measurements of your BMS via serial, MQTT or Home Assistant API. What's your goal?

Kamalkant1234, 2022-11-05:
> You are able to monitor the measurements of your BMS via serial, MQTT or Home Assistant API. What's your goal?

And sir which connection diagram i have to be follow, share please connection diagram !!
And my goal i want show my BMS Data on Arduino IDE and after that I will share all data on my cloud.

syssi, 2022-11-05:
This is the connection diagram: https://github.com/syssi/esphome-jk-bms#schematics

You have to connect three wires (GND, RX, TX) to your BMS using a 4 Pin, JST 1.25mm pitch connector. This project can share all the measurements directly to a MQTT broker (your cloud). Why are you looking for a Arduino IDE compatible solution? I cannot provide a Arduino package.

Kamalkant1234, 2022-11-05:
> This is the connection diagram: https://github.com/syssi/esphome-jk-bms#schematics
>
> You have to connect three wires (GND, RX, TX) to your BMS using a 4 Pin, JST 1.25mm pitch connector. This project [...]

Esp8266 will support TTL connection without any converter module.

syssi, 2022-11-05:
Yes. You can attach a ESP8266 or ESP32 directly to the BMS because the logic level is 3.3V.

Kamalkant1234, 2022-11-05:
> Yes. You can attach a ESP8266 or ESP32 directly to the BMS because the logic level is 3.3V.

Ya exactly 2.2 to 3.3v data output i received, So can I  connect it to RX,TX pin both wire directly, without any converter.
And here is need any kind of coding in NodeMcu ??

syssi, 2022-11-05:
Yes, you can connect the BMS directly to the ESP/NodeMCU board. No coding required if you want to use this project/implementation.

closed 2022-11-07 by syssi

Kamalkant1234, 2022-11-07:
Ok