syssi/esphome-jk-bms #205: Connection issues

markusgg84, 2022-11-05:
I'm trying to connect to 2 JKBMS but i've connection issues. I've double checked the wifi and ble mac adresses.
I've started from the esp32-ble-example-multiple-devices.yaml

Log Output:

`
INFO Successfully uploaded program.

INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[12:06:46]\xff[I][logger:243]: Log initialized
[12:06:46][C][ota:465]: There have been 0 suspected unsuccessful boot attempts.
[12:06:46][D][esp32.preferences:113]: Saving 1 preferences to flash...
[12:06:46][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[12:06:46][I][app:029]: Running through setup()...
[12:06:47][D][esp32_ble_tracker:264]: Starting scan...
[12:06:47][C][wifi:037]: Setting up WiFi...
[12:06:47][D][esp-idf:000]: E (2055) wifi:
[12:06:47][D][esp-idf:000]: Expected to init 16 rx buffer, actual is 3
[12:06:47][D][esp-idf:000]:
[12:06:47]
[12:06:47][E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 257
[12:06:47][W][wifi_esp32:057]: Setting WiFi mode failed!
[12:06:47][D][wifi:384]: Starting scan...
[12:06:47][D][esp-idf:000]: E (2097) wifi:
[12:06:47][D][esp-idf:000]: Expected to init 16 rx buffer, actual is 0
[12:06:47][D][esp-idf:000]:
[12:06:47]
[12:06:47][E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 257
[12:06:47][W][wifi_esp32:057]: Setting WiFi mode failed!
[12:06:47][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:06:47][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:06:47][D][switch:037]: 'jk-bms-links enable bluetooth connection': Sending state ON
[12:06:47][D][switch:037]: 'jk-bms-rechts enable bluetooth connection': Sending state ON
[12:06:47][D][esp32_ble_client:036]: Found device at MAC address [C8:47:8C:E5:04:89]
[12:06:47][I][esp32_ble_client:058]: Attempting BLE connection to c8:47:8c:e5:04:89
[12:06:50][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:06:51][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:06:55][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:06:56][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:07:00][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:07:01][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:07:05][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:07:06][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:07:10][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:07:11][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:07:15][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
[12:07:16][W][jk_bms_ble:342]: [c8:47:8c:e5:1c:74] Not connected
[12:07:17][E][wifi:392]: Scan timeout!
[12:07:20][W][jk_bms_ble:342]: [c8:47:8c:e5:04:89] Not connected
`
First part of the config file:

`
substitutions:
  name: jk-bms
  bms0: "${name}-links"
  bms1: "${name}-rechts"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:8C:E5:04:89
  bms1_mac_address: C8:47:8C:E5:1C:74
  bms0_protocol_version: JK02
  bms1_protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: ttgo-lora32-v1
  framework:
   type: arduino
   version: latest
`

syssi, 2022-11-05:
It looks like multiple BLE client connection are broken ESPHome 2022.9.3. See https://github.com/syssi/esphome-jk-bms/issues/189

Could you try this workaround: https://github.com/syssi/esphome-jk-bms/issues/189#issuecomment-1274756531

And please try to use the ESP-IDF framework instead of Arduino:

```
  framework:
    type: esp-idf
    version: latest
```

To better isolate the error you could start with a single BLE connection / BMS.

markusgg84, 2022-11-05:
Didn't help, here is the log output:

`

[14:36:23][W][jk_bms_ble:291]: [c8:47:8c:e5:04:89] Not connected
[14:36:23][W][jk_bms_ble:291]: [c8:47:8c:e5:1c:74] Not connected
[14:36:23][D][esp-idf:000]: I (5850) wifi:
[14:36:23][D][esp-idf:000]: state: init -> auth (b0)
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5861) wifi:
[14:36:23][D][esp-idf:000]: state: auth -> assoc (0)
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5875) wifi:
[14:36:23][D][esp-idf:000]: state: assoc -> run (10)
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5905) wifi:
[14:36:23][D][esp-idf:000]: connected with MGG, aid = 2, channel 11, BW20, bssid = 44:4e:6d:6c:f7:70
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5906) wifi:
[14:36:23][D][esp-idf:000]: security: WPA2-PSK, phy: bgn, rssi: -28
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5909) wifi:
[14:36:23][D][esp-idf:000]: pm start, type: 1
[14:36:23]
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: I (5924) wifi:
[14:36:23][D][esp-idf:000]: AP's beacon interval = 102400 us, DTIM period = 1
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp-idf:000]: W (5942) wifi:
[14:36:23][D][esp-idf:000]: <ba-add>idx:0 (ifx:0, 44:4e:6d:6c:f7:70), tid:6, ssn:2, winSize:64
[14:36:23][D][esp-idf:000]: 
[14:36:23]
[14:36:23][D][esp32_ble_tracker:731]: Found device 4D:4D:4D:05:5A:BD RSSI=-82
[14:36:23][D][esp32_ble_tracker:752]:   Address Type: RANDOM
[14:36:24][D][esp-idf:000]: I (6605) esp_netif_handlers: sta ip: 192.168.188.163, mask: 255.255.255.0, gw: 192.168.188.1
[14:36:24]
[14:36:24][C][wifi:353]:   Local MAC: 50:02:91:97:A2:40
[14:36:24][C][wifi:354]:   SSID: 'MGG'
[14:36:24][C][wifi:355]:   IP Address: 192.168.188.163
[14:36:24][C][wifi:357]:   BSSID: 44:4E:6D:6C:F7:70
[14:36:24][C][wifi:358]:   Hostname: 'jk-bms'
[14:36:24][C][wifi:360]:   Signal strength: -28 dB ▂▄▆█
[14:36:24][C][wifi:364]:   Channel: 11
[14:36:24][C][wifi:365]:   Subnet: 255.255.255.0
[14:36:24][C][wifi:366]:   Gateway: 192.168.188.1
[14:36:24][C][wifi:367]:   DNS1: 192.168.188.1
[14:36:24][C][wifi:368]:   DNS2: 0.0.0.0
[14:36:24][C][ota:085]: Over-The-Air Updates:
[14:36:24][C][ota:086]:   Address: jk-bms.local:3232
[14:36:24][W][ota:095]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[14:36:24][C][api:025]: Setting up Home Assistant API server...
[14:36:24][C][mqtt:029]: Setting up MQTT...
[14:36:24][I][mqtt:176]: Connecting to MQTT...
[14:36:24][D][esp-idf:000]: W (6689) wifi:
[14:36:24][D][esp-idf:000]: <ba-add>idx:1 (ifx:0, 44:4e:6d:6c:f7:70), tid:0, ssn:0, winSize:64
[14:36:24][D][esp-idf:000]: 
[14:36:24]
[14:36:24][I][mqtt:216]: MQTT Connected!
[14:36:24]
[14:36:24]abort() was called at PC 0x401b2c86 on core 1
[14:36:24]
[14:36:24]Backtrace:0x40081d4e:0x3ffcdf80 0x40090d89:0x3ffcdfa0 0x40096e32:0x3ffcdfc0 0x401b2c86:0x3ffce030 0x401b1b3b:0x3ffce050 0x401b1aa2:0x3ffce070 0x401b1cf1:0x3ffce090 0x401b1e34:0x3ffce0b0 0x401b1fb1:0x3ffce0f0 0x400e07b7:0x3ffce110 0x400e07f7:0x3ffce130 0x400e0a4a:0x3ffce150 0x400e0a26:0x3ffce170 0x400e0a26:0x3ffce190 0x400e0b5d:0x3ffce1b0 0x400e32bd:0x3ffce230 0x400e3a42:0x3ffce270 0x400e3a87:0x3ffce2c0 0x401c9c25:0x3ffce2e0 0x400ea080:0x3ffce300 0x400f6ef1:0x3ffce330 0x400d840b:0x3ffce400
WARNING Found stack trace! Trying to decode it
[... 104 lines cut ...]
[14:36:26][D][esp32.preferences:113]: Saving 1 preferences to flash...
[14:36:26][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[14:36:26][I][app:029]: Running through setup()...
[14:36:26][D][esp-idf:000]: I (94) BTDM_INIT: BT controller compile version [6a07b06]
[14:36:26]
[14:36:26][D][esp-idf:000]: I (973) system_api: Base MAC address is not set
[14:36:26]
[14:36:26][D][esp-idf:000]: I (974) system_api: read default base MAC address from EFUSE
[14:36:26]
[14:36:26][D][esp-idf:000]: I (977) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
[14:36:26]
[14:36:26][D][esp32_ble_tracker:222]: Starting scan...
[14:36:26][C][wifi:037]: Setting up WiFi...
[14:36:26][D][esp-idf:000]: I (1594) wifi:
[14:36:26][D][esp-idf:000]: wifi driver task: 3fff0498, prio:23, stack:6656, core=0
[14:36:26][D][esp-idf:000]: 
[14:36:26]
[14:36:26][D][esp-idf:000]: I (1614) wifi:
[14:36:26][D][esp-idf:000]: wifi firmware version: eb52264
[14:36:26][D][esp-idf:000]: 
[14:36:26]
[14:36:26][D][esp-idf:000]: I (1615) wifi:
[14:36:26][D][esp-idf:000]: wifi certification version: v7.0
[14:36:26][D][esp-idf:000]: 
`

markusgg84, 2022-11-05:
I've got the same issue with only one device.

syssi, 2022-11-05:
It looks like you hit a resource limit. Please remove most of the sensors (just for testing). If it's working you can decrease the log level to free some resources and try to increase the number of sensors step by step.

markusgg84, 2022-11-06:
That work, thank you!

closed 2022-11-06 by markusgg84

syssi, 2022-11-06:
It looks like the BLE client component is broken at 2022.10.1 Please be careful!

https://github.com/esphome/issues/issues/3729