syssi/esphome-jk-bms #210: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'

rauchbergerk, 2022-11-08:
Hi, I tried to flash the esp32 with esphome, unfortunately i got this message

Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'

I connected the esp to the windows and tried also via raspberry which is running homeassistant with esphome.

any ideas, have seen comments 4 days ago with other problems. Seems they got it to work...
With arduino I got only one reading and then it stucked.

syssi, 2022-11-08:
1. You have to use ESPHome 2022.11.2
2. You have to stick to ESP-IDF if you want to use BLE
3. ESP-IDF doesn't build on aarch64 processors (Raspberry Pi). Please use a x86 (windows or linux) system as build host

rauchbergerk, 2022-11-08:
Thank you for fast reply. I am new in esphome flashing. 
I have the current version: 2022.10.2 . 2022.11.2 ? 
I will set up a linux usb stick to run my windows computer with any linux and try again... and again... and again...
thank you for your help. Hope I have it understood and can get it running.

syssi, 2022-11-08:
Sorry, my fault. ESPHome 2022.10.2 is required. ;-)

TheSmartGerman, 2022-11-10:
You have to compile on a x86 pc.

Install ESPHome on windows. Compile and run it from there.

I installed the Samba Add-on

I did a Script to copy from my Homeassistant to local machine, compile & uplade the jk-bms on esp32.

TheSmartGerman, 2022-11-10:
```
xcopy /e /k /h /i \\192.168.178.100\config\esphome\*.* %TMP%\esphome\
cd %TMP%\esphome\
esphome run --no-logs %~nx1
del /f /s /q %TMP%\esphome\
```

rauchbergerk, 2022-11-10:
Thank you, I taught my question is so stupid and I am earning only head nodding…
Thank you. Will try more things. I keep you updated in every case. :-)
I seek for a old pc with 32bit or try a vmware or something like this.

syssi, 2022-11-11:
The toolchain is 64bit compatible. Only the processor architecture of the Raspberry Pi isn't supported. So any other processor architecture should work.

closed 2022-12-17 by syssi