syssi/esphome-jk-bms #211: MQTT Crash

infester86, 2022-11-11:
Hi,

my ESP only reboots and has Problems with the MQTT. I dont have Homeassistant and need the MQTT. The esphome version is the newest. 

This is the crash:

`[12:11:00][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms001 cell voltage overvoltage recovery'
[12:11:00][C][jk_bms_ble.number:009]:   Unit of Measurement: 'V'
[12:11:00][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms001 cell voltage undervoltage protection'
[12:11:00][C][jk_bms_ble.number:009]:   Unit of Measurement: 'V'
[12:11:00][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms001 cell voltage undervoltage recovery'
[12:11:00][C][jk_bms_ble.number:009]:   Unit of Measurement: 'V'
[12:11:00][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms001 balance starting voltage'
[12:11:00][C][jk_bms_ble.number:009]:   Unit of Measurement: 'V'
[12:11:00][C][jk_bms_ble.number:009]: JkBmsBle Number 'jkbms001 voltage calibration'Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[12:11:00]
[12:11:00]Core  1 register dump:
[12:11:00]PC      : 0x400014fd  PS      : 0x00060e30  A0      : 0x800dd764  A1      : 0x3ffcdfa0  
[12:11:00]A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
[12:11:00]A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcdf60  
[12:11:00]A10     : 0x3ffcdfb8  A11     : 0x3ffcdfb8  A12     : 0x3ffc6138  A13     : 0x00000000  
[12:11:00]A14     : 0x3ffc8000  A15     : 0x3ffc01f4  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
[12:11:00]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
[12:11:00]
[12:11:00]Backtrace:0x400014fa:0x3ffcdfa0 0x400dd761:0x3ffcdfb0 0x400dd83a:0x3ffce010 0x400dec88:0x3ffce030 0x401be109:0x3ffce050 0x401be1f5:0x3ffce070 0x400e5124:0x3ffce090 0x400e72e2:0x3ffce0c0 0x400d33ca:0x3ffce0e0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400dd761: esphome::mqtt::MQTTBackendIDF::mqtt_event_handler_(esp_mqtt_event_t const&) at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:114
WARNING Decoded 0x400dd83a: esphome::mqtt::MQTTBackendIDF::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:67
 (inlined by) esphome::mqtt::MQTTBackendIDF::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:62
WARNING Decoded 0x400dec88: esphome::mqtt::MQTTClientComponent::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_client.cpp:228
WARNING Decoded 0x401be109: esphome::Component::call_loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/component.cpp:75
WARNING Decoded 0x401be1f5: esphome::Component::call() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/component.cpp:97
WARNING Decoded 0x400e5124: esphome::Application::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/application.cpp:74 (discriminator 2)
WARNING Decoded 0x400e72e2: loop() at /home/michael/config/.esphome/build/jkbms001/src/main.cpp:2870
WARNING Decoded 0x400d33ca: esphome::loop_task(void*) at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/esp32/core.cpp:74 (discriminator 1)
[12:11:00]
[12:11:00]
[12:11:00]ELF file SHA256: fd85411865b2ef7f
[12:11:00]
[12:11:00]Rebooting...
[12:11:00]ets Jun  8 2016 00:22:57
[12:11:00]
[12:11:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:11:00]configsip: 0, SPIWP:0xee
[12:11:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:11:00]mode:DIO, clock div:2
[12:11:00]load:0x3fff0030,len:7572
[12:11:00]load:0x40078000,len:14912
[12:11:00]ho 0 tail 12 room 4
[12:11:00]load:0x40080400,len:3688
[12:11:00]entry 0x4008067c
[12:11:00]I (29) boot: ESP-IDF 4.3.2 2nd stage bootloader`

and this is my config:

`substitutions:
  name: jkbms001
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E8:1C:3D
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

[... 230 lines cut ...]
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"`

syssi, 2022-11-11:
Please try to decrease the number of sensors (temporary) and decrease the log level to free some resources just for testing.

infester86, 2022-11-14:
I did like you said. This is my config:

> substitutions:
>   name: jkbms001
>   device_description: "Monitor and control a JK-BMS via bluetooth"
>   external_components_source: github://syssi/esphome-jk-bms@main
>   mac_address: C8:47:8C:E8:1C:3D
>   # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
>   protocol_version: JK02
> 
> esphome:
>   name: ${name}
>   comment: ${device_description}
>   project:
>     name: "syssi.esphome-jk-bms"
>     version: 1.3.0
> 
> esp32:
>   board: esp32dev
>   framework:
>     type: esp-idf
>     version: latest
> 
> external_components:
>   - source: ${external_components_source}
>     refresh: 0s
> 
> wifi:
>   ssid: "xxx"
>   password: "xxx"
> 
> # Enable ESPHome API
> #api:
> 
> ota:
>   password: "ab30e14015896e9d5e36c4e474cb87d6"
> 
> logger:
>   level: WARN
>   
> #api:
> 
> # The MQTT component is ESP-IDF compatible since ESPHome version 2022.4.0. If
> # ESPHome suggests to use the arduino framework instead because of missing ESP-IDF
> # framework support you should update your setup.
> mqtt:
>   broker: "xxx"
>   username: "xxx"
>   password: "xxx"
> 
> esp32_ble_tracker:
> 
> ble_client:
>   - mac_address: ${mac_address}
>     id: client0
> 
> jk_bms_ble:
>   - ble_client_id: client0
>     protocol_version: ${protocol_version}
>     throttle: 5s
>     id: bms0
> 
> binary_sensor:
>   - platform: jk_bms_ble
>     balancing:
>       name: "${name} balancing"
>     charging:
>       name: "${name} charging"
>     discharging:
>       name: "${name} discharging"
> 
> button:
>   - platform: jk_bms_ble
>     retrieve_settings:
>       name: "${name} retrieve settings"
>     retrieve_device_info:
>       name: "${name} retrieve device info"

and this is my mqtt crash:

> [11:24:14]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> [11:24:14]
> [11:24:14]Core  1 register dump:
> [11:24:14]PC      : 0x400014fd  PS      : 0x00060030  A0      : 0x800d8694  A1      : 0x3ffcdc50  
> [11:24:14]A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
> [11:24:14]A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcdc10  
> [11:24:14]A10     : 0x3ffcdc68  A11     : 0x3ffcdc68  A12     : 0x3ffc6130  A13     : 0x00000000  
> [11:24:14]A14     : 0x3ffc7d28  A15     : 0x3ffc01f0  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
> [11:24:14]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> [11:24:14]
> [11:24:14]Backtrace:0x400014fa:0x3ffcdc50 0x400d8691:0x3ffcdc60 0x400d876a:0x3ffcdcc0 0x400d99c8:0x3ffcdce0 0x401b0651:0x3ffcdd00 0x401b073d:0x3ffcdd20 0x400de434:0x3ffcdd40 0x400e0272:0x3ffcdd70 0x400d2e96:0x3ffcdd90
> WARNING Found stack trace! Trying to decode it
> WARNING Decoded 0x400d8691: esphome::mqtt::MQTTBackendIDF::mqtt_event_handler_(esp_mqtt_event_t const&) at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:114
> WARNING Decoded 0x400d876a: esphome::mqtt::MQTTBackendIDF::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:67
>  (inlined by) esphome::mqtt::MQTTBackendIDF::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_backend_idf.cpp:62
> WARNING Decoded 0x400d99c8: esphome::mqtt::MQTTClientComponent::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/mqtt/mqtt_client.cpp:228
> WARNING Decoded 0x401b0651: esphome::Component::call_loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/component.cpp:75
> WARNING Decoded 0x401b073d: esphome::Component::call() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/component.cpp:97
> WARNING Decoded 0x400de434: esphome::Application::loop() at /home/michael/config/.esphome/build/jkbms001/src/esphome/core/application.cpp:74 (discriminator 2)
> WARNING Decoded 0x400e0272: loop() at /home/michael/config/.esphome/build/jkbms001/src/main.cpp:357
> WARNING Decoded 0x400d2e96: esphome::loop_task(void*) at /home/michael/config/.esphome/build/jkbms001/src/esphome/components/esp32/core.cpp:74 (discriminator 1)
> [11:24:14]
> [11:24:14]
> [11:24:14]ELF file SHA256: 6c68d4d54f220434
> [11:24:14]
> [11:24:14]Rebooting...
> [11:24:14]ets Jun  8 2016 00:22:57
> [11:24:14]
> [11:24:14]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> [11:24:14]configsip: 0, SPIWP:0xee
> [11:24:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> [11:24:14]mode:DIO, clock div:2
> [11:24:14]load:0x3fff0030,len:7572
> [11:24:14]load:0x40078000,len:14912
> [11:24:14]ho 0 tail 12 room 4
> [11:24:14]load:0x40080400,len:3688
> [11:24:14]entry 0x4008067c

syssi, 2022-11-14:
Thats sad. Which ESPHome version do you use?

infester86, 2022-11-14:
the newest esphome version prints: 2022.10.2

syssi, 2022-11-14:
I will try to reproduce the issue later the week. Do you have access to another ESP32 board to make sure it's no hardware issue?

infester86, 2022-11-14:
I tried 3 different board (but every board from the same manufacturer) and every board has the same result. I will try to get my hands on another one. Thanks!

infester86, 2022-11-14:
It might be a Problem with the MQTT Server. Switchted to a Public Server and it seems to work. Will try some more and report back.

syssi, 2022-11-14:
Good catch!

infester86, 2022-11-14:
Well i have a NUC with ioBroker installed. As MQTT Server i used the one from a ioBroker Plugin and it always crashed. I switched to the offical eclipse-mosquitto docker container and now it works. So nothing wrong here just some crazy stuff happening.