syssi/esphome-jk-bms #212: B2A20S20P wrong 0xb9 (capacity remaining Ah)

sigxcpu76, 2022-11-13:
This is not directly related to `esphome-jk-bms` project but it is related to reading data from it (I used it as a model to read data over modbus).

I have 2 version of `B2A20S20P`, one is software `v10.08` the other one is software `v10.10`.
I read them both over BLE or over modbus.
The older version never reports over 277Ah capacity remaining over BLE, but it goes down when pulling from battery. This is ok-ish. What is not ok is that It returns a constant 256Ah regardless (bytes `B900000100`) over modbus, regardless of the actual charge (the SOC percent is fine).
The newer version (v10.10) reports correctly this value. It goes to 280Ah over BLE and over modbus.

Has anyone encountered this behavior? Is there any way to upgrade the JK BMS firmware?

sigxcpu76, 2022-11-13:
What is funny is that the SOC percent is working correctly, therefore I compute total_capacity * SOC to get the value of the wrong field. It looks like they are fiddling with bits on each minor release.

syssi, 2022-11-18:
In my case the BMS (JK-BD6A17S6P-7.1.0H-UART-TTL) returns always `0x00000000` here (`0xb9`). The same workaround is in place here: https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L304-L305

commit 584435f9 (2022-12-18): Add a comment regarding capacity remaining (`0xb9`) measurements

Closes: #212
closed 2022-12-18 by syssi