syssi/esphome-jk-bms #213: Using on JK-B5A24S Balancer Only unit, mostly works but some issues

JSladen, 2022-11-16:
Hi,

I am trying to use this on a Balancer Only unit (JK-B5A24S), and while most of it works there are a couple of bits which don't, and I was wondering if they could be investigated.

I've disabled all the sensors/switches which I don't think apply to the Balancer, which has left the following sections visible, but empty and non functional.
[Configuration] Balance Trigger voltage
[Configuration] Cell count
[Configuration] Max Balance Current
[Configuration] Power Off Voltage
[Configuration] Voltage Calibration
[Sensors] Temperature Sensor 1

I can see most of these values are correctly extracted from the settings frame, but are not displayed in HA, and attempting to send them isn't allowed. I can set them absolutely fine over bluetooth.

[08:51:03][I][jk_bms_ble:1116]: Device info frame (300 bytes):
[08:51:03][I][jk_bms_ble:1168]:   Vendor ID: JK-B5A24S
[08:51:03][I][jk_bms_ble:1169]:   Hardware version: 8.X
[08:51:03][I][jk_bms_ble:1170]:   Software version: 8.0.3M

[08:51:04][I][jk_bms_ble:1025]: Settings frame (300 bytes):
[08:51:04][D][jk_bms_ble:1052]:   Unknown6: 00 00 80 3F (1.000000)
[08:51:04][I][jk_bms_ble:1061]:   Cell count: 16
[08:51:04][I][jk_bms_ble:1064]:   Power off voltage: 3.200000 V
[08:51:04][D][jk_bms_ble:1075]:   Unknown74: 3.033129
[08:51:04][I][jk_bms_ble:1083]:   Start balance voltage: 4.250000 V
[08:51:04][D][jk_bms_ble:1086]:   Unknown102: 3.400000
[08:51:04][I][jk_bms_ble:1089]:   Trigger Delta Voltage: 0.008000 V
[08:51:04][I][jk_bms_ble:1092]:   Max. balance current: 4.500000 A
[08:51:04][D][switch:037]: 'esphome-jkbalancer balancer': Sending state OFF
[08:51:04][I][jk_bms_ble:1096]:   ADC Vref: unknown V

[09:01:29][D][number:054]: 'esphome-jkbalancer power off voltage' - Setting number value
[09:01:29][D][number:113]:   New number value: 3.100000
[09:01:29][E][jk_bms_ble.number:019]: This number entity isn't supported by the selected protocol version

The following is a list of the unknown parts when decoding the data, if any of that is of use?

[08:58:51][D][jk_bms_ble:788]: Unknown210: 00 00 00 00 (always 0x00 0x00 0x00 0x00)
[08:58:51][D][jk_bms_ble:791]: Unknown214: FF FF 00 00 (0xFF 0xFF: 24 cells?)
[08:58:51][D][jk_bms_ble:794]: Unknown218: 02
[08:58:51][D][jk_bms_ble:797]: Unknown219: 0C
[08:58:51][D][jk_bms_ble:805]: Unknown221: 01
[08:58:51][D][jk_bms_ble:812]: Unknown226: 00 00 00 00 00 00 00 (always 0x00...0x00?)
[08:58:51][D][jk_bms_ble:816]: Unknown233: 64 0D 9D 4A (5146290.000000)
[08:58:51][D][jk_bms_ble:820]: Unknown237: 40 00 00 00 (always 0x40 0x00 0x00 0x00?)
[08:58:51][D][jk_bms_ble:834]: Unknown290: 00 E0 82 3F (always 0x00 0x00 0x00 0x00?)
[08:58:51][D][jk_bms_ble:837]: Unknown294: 00 90 D5 3F
[08:58:51][D][jk_bms_ble:840]: Unknown298: 00

If there is anything else I can provide then please let me know.
Thank you for any assistance you can provide.

syssi, 2022-11-16:
You are right. The `JK04` support isn't complete yet. Could you increase the log level to `VERY_VERBOSE`? I'm interested in some raw data (notification frames) of your device:

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L219-L220

JSladen, 2022-11-16:
Is this what you are after?

```
[10:03:34][VV][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.AC.FF.A9.54.40.9D.95.54.40.D9.6C.54.40.FF.A9.54.40.3B.81.54.40.9D.95.54.40.D9.6C.54.40.FF.A9.54.40.3B.81.54.40.D9.6C.54.40.3B.81.54.40.FF.A9.54.40.3B.81.54.40.9D.95.54.40.3B.81.54.40.3B.81.54.40.00.00.00.00.00.00.00.00.00.00 (80)
[10:03:34][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.F7.32.99.3D.E9.3E.A1.3D.E4.7C.9D.3D.8A.09.9A.3D.70.90.9A.3D.02.B2.92.3D.5B.82.94.3D.04.2D.93.3D.FA.3C.92.3D.66.EB.93.3D.88.3A.93.3D.52.24.97.3D.E6.53.98.3D.03.9A.95.3D.D9.65 (80)
[10:03:34][VV][jk_bms_ble:220]: Notification received: 92.3D.11.F4.9A.3D.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.CB.91.54.40.00.36.60.3C.00.00.00.00.FF.FF.00.00.0D.01.01.01.3C.F8.8C.40.00.00.00.00.00.00.00.54.39.92.4A.40.00.00 (80)
[10:03:34][VV][jk_bms_ble:220]: Notification received: 00.00.56.3C.5A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.D0.02.0D.02.00.00.00.00.00.00.00.00.00.00.03.00.00.00.00.00.00.00.00.00.08.06.09.00.00.58.24.40.00.A8.D5.3F.00.C6 (60)
```

I Think I remember reading somewhere else that it uses a 320 byte frame rather than a 300, so I don't know if there is any information missing from the end of it?

syssi, 2022-11-16:
This frame looks good & complete (80+80+80+60 = 300, Preamble `55.AA.EB`, CRC at the EOF). You have provided the raw data of a cell info frame:

``` 
55.AA.EB.90.02
            ^^ cell info frame
```

Could you capture a `settings frame` and `device info` frame?

```
55.AA.EB.90.01 # frame_type: settings
55.AA.EB.90.03 # frame_type: device info
```

The frames are requested right after the established BLE connection. You could use the `BLE switch` to disconnect + resume the BLE connection:

```
switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"
```

syssi, 2022-11-16:
As next step I would be happy about a BLE traffic capture (btsnoop). Don't worry, it's quite simple as long as you have an Android device. The btsnoop capture is part of the bug report bundle (if enabled beforehand):

https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

I try to summarize the goal for a brief overview:

1. Turn on the developer mode of your android device + enable the btsnoop capturing
2. Turn off bluetooth, turn on bluetooth again
3. Open the Jikong Balancer App
4. Go to the settings and note down the current values and set some new values.
5. Use `adb` to retrieve the bug report bundle ZIP.
6. Extract the bundle und provide the `btsnoop_hci.log`

I will extract the write commands / instructions from the capture so we can guess the proper payload of the JK04 protocol. Without having this capture there is a risk to write garbage to your device.

JSladen, 2022-11-16:
[bt-snoop-JK-B5A24S.log](https://github.com/syssi/esphome-jk-bms/files/10026454/bt-snoop-JK-B5A24S.log)

Hopefully this includes what you are after. I've removed the before and after as it was quite a large file, but I don't think anything should be missing.

I set each of the following values and then reverted them in this order
			Initial	Changed	Return
Cells Count		16	17	16
Trigger Delta Vol.(V)	0.008	0.108	0.008
Max Balance Cur.(A)	4.500	3.500	4.500
Power Of Vol(V)		3.200	2.200	3.200
ADC Vref		52.xx	51.xx	52.xx

The ADC Ref figures I'm not 100% certain on, as they change, and it came up with the wrong figure next time I opened it.

Please let me know if anything else can be of use.

syssi, 2022-11-17:
Thanks for preparing such a well structured and documented capture. I will try to go the next steps later the week and will report back if something is missing.

syssi, 2022-11-17:
Extracted write commands:

```
aa5590eb9700b44cb1791a3fa86121c638de6f09 # Request device info
aa5590eb96003dd6b2ad5d1a244642b3f70b136d # Request cell info
aa5590eb96004a93c058a9fe5358b092ca4add8a # Request cell info

aa5590eb1c02110004d2e36fb8c5d4912ec07e1f # Set cell count
aa5590eb1c021000378e33f7ada8dbf65ea26926 # Set cell count

aa5590eb64041b2fdd3d22b35cce3623a75a4eed
aa5590eb64046f12033cba4c253e8976b0f6ef9f

aa5590eb6804000060408790eaa88382efbc3413
aa5590eb6804000090409da3c98226fbf6539a45

aa5590eb2004cdcc0c408aac6f3768b50a286b19
aa5590eb2004cdcc4c402c0eaedfa413b82a4366

aa5590eb4404717a3e406e7d4a145009cd1a30e4
aa5590eb4404178e4540ede90757c043ca4d2862

aa5590eb9600a141ed784905039bca92f3db47b4 # Request cell info
```

syssi, 2023-02-05:
See also https://github.com/syssi/esphome-jk-bms/issues/94

iovcharyk, 2023-02-12:
@syssi thanks a lot for your work done, just to confirm it's working fine via BLE and JK-B2A20S20P HW V11.XW, SW V11.25H

syssi, 2023-02-12:
@iovcharyk Thanks for your feedback! Could you do me a favor and create a new issue + report your device as supported? I don't want to reference this issue because your report is off-topic here.

JSladen, 2024-02-22:
Hi,

Please can I check if there was any more progress on fully supporting JK04/balancer units?

I am happy to try and provide further scans or test things out if that's of any assistance?

Many Thanks