syssi/esphome-jk-bms #214: Unhandled C++ exception: OOM on Wemos D1 mini

prezes210, 2022-11-21:
After flashing `Wemos D1 mini` iam getting OOM error during startup.
I can observe this error when iam connected "locally" to wemos using serial port.

Please see captured log:
```
NW           h  )
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: sys
sp: 3fffeda0 end: 3fffffb0 offset: 0000
3fffeda0:  00000005 00000000 00000020 4010075c  
3fffedb0:  000000fe 00000000 00000000 00000000  
3fffedc0:  00000000 00000000 00000000 00000001  
3fffedd0:  00007fff 00000000 60000600 00000000  
3fffede0:  3fff1618 402119e3 00000028 40223f76  
3fffedf0:  3fff1618 402119e3 00000020 40223f95  
3fffee00:  ffffffd3 00000000 00000028 402237b0  
3fffee10:  00000001 3fffee60 00000001 402119e3  
3fffee20:  00000000 00000000 00000000 40210d90  
3fffee30:  0000000b 3ffec9d6 ffffffa5 3fffee90  
3fffee40:  3fff1604 00000000 3fffbe94 40211b74  
3fffee50:  00000000 3fffbacc 3ffec9cc 00000000  
3fffee60:  2b645000 00cc30c6 3fffee70 0000000b <
3fffee70:  53545349 3437365f 00363333 3ffec9e4  
3fffee80:  0001a601 00000000 3fffee90 00000000 <
3fffee90:  53545300 3437365f 00363333 3ffec9e4  
3fffeea0:  c62b6450 3ffecc30 3ffec9fe 3ffeca31  
3fffeeb0:  3fff1618 4023f1f6 3ffef178 00000000  
3fffeec0:  40211ba0 00000000 4bc6a7f0 3fffbacc  
3fffeed0:  40211ba0 00000000 3ffef178 40211bb4  
3fffeee0:  4023f251 3ffef178 3fffbacc 4023f306  
3fffeef0:  0000010f 4023f00d 3ffef190 3ffef9b0  
3fffef00:  005c2091 60000600 00000003 40230cd0  
3fffef10:  4023f571 001c9e36 3ffef9b0 60000600  
3fffef20:  4025809d 3ffef190 3ffef9b0 005c0d05  
3fffef30:  402580e2 3fffdab0 00000000 3fffdcb0  
3fffef40:  3ffef9c8 3fffdad0 3fff04fc 40223906  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 0006a4a1 00000000 00000005  
3fffef70:  00000000 aa55aa55 000000ed 40105a15  
3fffef80:  40105a1b 00000000 00000005 40101150  
3fffef90:  4010000d 257b727e 0006a4a1 401000ac  
3fffefa0:  4024833c 3fffef3c 402482ed 3ffffe88  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
[... 865 lines cut ...]
3fffff60:  00000016 feefef00 feefeffe feefeffe  
3fffff70:  3fffb0b4 3ffe9787 3ffe9851 3ffe9815  
3fffff80:  3ffe971f 3ffe97ea feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff04fc  
3fffffa0:  3fffdad0 00000000 3fff04e8 40223ab0  
<<<stack<<<

last failed alloc call: 402119E3(640)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x402119e3

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006a4b0
~ld
```

syssi, 2022-11-21:
This looks like an out of memory exception. Please remove most of the sensors and try again!

prezes210, 2022-11-21:
Thanks @syssi for super quick response.
I've followed your advice but still no luck.  I've removed all sensors and iam still getting OOM.

One more thing, I am running homeasstiance `raspberry pi 4`. Maybe that is an issue.

My esphome config below.
```
substitutions:
  name: jkbms-280-1
  device_description: "Monitor and control a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:
  baud_rate: 0
  level: DEBUG

api:

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH

jk_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 50ms

jk_bms:
  id: bms0
  jk_modbus_id: modbus0
```

syssi, 2022-11-21:
Hmm... this YAML should just work. Could you try another D1 mini to rule out a faulty ESP?

prezes210, 2022-11-22:
@syssi you were right. 
I used a brand new D1 mini and installation went without any issues.
Thank you for support.

closed 2022-11-22 by prezes210