syssi/esphome-jk-bms #217: Any solution to separe gnd from D1mini

gepetto666, 2022-11-25:
Hi.
i have powered the d1mini from thebateri with a DC DC converter. 24v to 5v.but there is a problem.
 ground from rs485 connector provide power to the esp even if i turnoff discharge from bms.
Is possible to use some optocupler to separate the ground?
Or other solution to separate data gnd from power gnd.
thnks
[image]

syssi, 2022-11-25:
We depend on the community's experience here. I'm a software developer and cannot provide an informed/qualified answer here.

gepetto666, 2022-11-25:
> We depend on the community's experience here. I'm a software developer and cannot provide an informed/qualified answer here.

my mistake , i post in wrong section, is not a issue. can you move it on discussion.
regards