syssi/esphome-jk-bms #221: JK-BD4A17S4P ESP32 Keeps restarting

Condor-XYZ, 2022-11-28:
Hello Unfortunately, my ESP32 keeps restarting. Apparently the software is not entirely compatible with my JK-BD4A17S4P BMS.
can i do something there?

[logs_jk-bms_logs.txt](https://github.com/syssi/esphome-jk-bms/files/10104100/logs_jk-bms_logs.txt)

syssi, 2022-11-28:
Could you provide some details (hardware & software version) about your BMS model? Do you miss some important measurements?

Condor-XYZ, 2022-11-28:
measured values are all there that i need
[image]

syssi, 2022-11-28:
These log messages are fine:

```
[18:38:16][D][jk_bms_ble:626]: Unknown168: 9E.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.0B.04.00.00.00.00.B1.28.41.40.00.00.00.00.E2.04.36.E7.00.00.00.01.00.05.00.00 (46)
[18:38:16][D][jk_bms_ble:640]: Unknown189: 04 00
[18:38:16][D][jk_bms_ble:644]: Unknown192: 00
[18:38:16][D][jk_bms_ble:646]: Unknown193: 00 B1 (0x00 0x8D)
[18:38:16][D][jk_bms_ble:648]: Unknown195: 28 41 (0x21 0x40)
```

There are some bytes at the response payload of which the purpose is not known. You could decrease the log level of the `jk_bms_ble` component from `DEBUG` to `INFO` to get rid of the noise.

Did you build the project with the recommended `ESP-IDF` framework or Arduino?

Condor-XYZ, 2022-11-28:
I used the ESP-IDF framework.
If I let it run like this for longer, the BMS has probably crashed. definitely no longer accessible with BLE (oh not on the smartphone)

syssi, 2022-11-28:
Could you check the red status LED? The BMS is able to handle a single BLE connection. If your ESP32 is connected to the BMS you cannot establish a connection with the Android app anymore and vice-versa. An active connection is indicated by a solid red LED. If the LED blinks the BMS is able to pick up a new connection. If the LED is off the BMS did an emergency shutdown.

syssi, 2022-11-28:
Keep me updated. :-) I keep my fingers crossed that it's no general issue of your hardware version. A assume a new / another BLE module is used now.

Condor-XYZ, 2022-11-28:
Was aber noch zu klären wäre wieso der ESP immer wieder einen neustart macht

syssi, 2022-11-28:
Could you update to the most recent ESPHome version and rebuild + flash the device? The BLE client implementation is changed.

Condor-XYZ, 2022-11-28:
the same
[Log.txt](https://github.com/syssi/esphome-jk-bms/files/10105748/Log.txt)

Condor-XYZ, 2022-11-29:
Yes, without MQTT it works.
my MQTT server is from Iobroker and runs with the rest of the smart home devices without any problems... is there anything to watch out for with ESPHome?

syssi, 2022-11-29:
This is another user having the same issue which could be solved by using mosquitto as MQTT broker: https://github.com/syssi/esphome-jk-bms/issues/211

Could you try to use another broker? Another idea is to capture the traffic between the ESPHome node and your MQTT broker to get an idea what's the different (too many messages? different handshake? protocol version?).

Condor-XYZ, 2022-11-30:
I changed the settings of the MQTT client in Iobroker and that's it. Thanks for your help

before:
[image]
after:
[image]

syssi, 2022-11-30:
Good job! Feel free to close the issue if it's solved!

closed 2022-11-30 by Condor-XYZ

syssi, 2023-04-18:
@Condor-XYZ Could you flash the most recent release of this component? I would like to make sure you should use `protocol_version: JK02_32S` in future or not.

syssi, 2023-04-21:
Thanks for your feedback! Good to know the frame structure of your BMS at software version 11 wasn't changed.