syssi/esphome-jk-bms #225: Build error of the example

AlexThePotter, 2022-12-20:
See log below.

Tried to instal the example on espdev board (esp32).
Install through hass esphome. Doesn't build from cmake  - any ideas?

```
INFO Reading configuration /config/esphome/bms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/jk-bms/.pioenvs/jk-bms/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:509 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:3 (project)

```

syssi, 2022-12-20:
Please make sure your home directory and build folder is on the same filesystem / mountpoint. Cross filesystem builds aren't supported out of the box.

AlexThePotter, 2022-12-20:
That's all hidden behind the Esphome addon from HASS? Not sure I can control that? or even check?

syssi, 2022-12-21:
Which ESPHome version / addon version do you use? I would like to make sure your setup is up-to-date before creating an issue upstream.

AlexThePotter, 2022-12-21:
The latest 2022.12.3

syssi, 2022-12-21:
Could you tell me the CPU architecture of your HASS OS setup?

AlexThePotter, 2022-12-21:
Pi 4

syssi, 2022-12-21:
In this case the issue is related to https://github.com/esphome/issues/issues/3076 and pretty hard to solve. You should pick a x86 host instead and install ESPHome manually here (`pip3 install esphome`). The command line tool (`esphome`) can be used to build, compile and flash the YAML configuration by:

```
esphome run example.yaml
```

AlexThePotter, 2022-12-21:
> In this case the issue is related to esphome/issues#3076

Thanks,

Will give that a go.

AlexThePotter, 2022-12-23:
Nope this doesn't fix this issue.

Ran on windows 10. Fresh latest python, latest esphome. On intel x86.

`fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/Alex/.platformio/packages/framework-espidf/tools/cmake/build.cmake:454 (message):
  C:\Users\Alex\AppData\Local\Programs\Python\Python311\python.exe: Error
  while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  C:/Users/Alex/.platformio/packages/framework-espidf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:3 (project)
`

syssi, 2022-12-23:
Sorry for the inconvenience. Usually ESPHome builds fine on any CPU arch expect for `arm64` (your Raspi 4). My windows knowledge is very limited and I cannot provide much support here. Do you have access to a Ubuntu host? The project setup will look very similar to this screencast:

[image]

closed 2023-01-28 by syssi