syssi/esphome-jk-bms #228: ESP32C3 linking error

Nellor, 2022-12-29:
I successfully installed it already at a standard az-delivery-devkit-v4 board and have a working connection to my balancer.
Im' very thankfully for your work! 
But now I tried to use the tiny seeed_xiao_esp32c3 that I can even integrate at the Heltec/Neey 4A balancer housing.

At final linking I get these errors:

```
Creating esp32c3 image...
Successfully created esp32c3 image.
Compiling /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/libb28/noise-c/crypto/aes/rijndael-alg-fst.o
Linking /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/src/esphome/components/esp32_ble_client/ble_client_base.o: in function `esphome::esp32_ble_client::BLEClientBase::connect()':
/config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_client/ble_client_base.cpp:65: undefined reference to `esp_ble_gattc_open'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::stop_scan()':
/config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:236: undefined reference to `esp_ble_gap_stop_scanning'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::start_scan_(bool)':
/config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:337: undefined reference to `esp_ble_gap_set_scan_params'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:340: undefined reference to `esp_ble_gap_start_scanning'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o: in function `esphome::esp32_ble_tracker::ESP32BLETracker::loop()':
/config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:173: undefined reference to `esp_ble_gap_stop_scanning'
/data/cache/platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /config/esphome/.esphome/build/bms-addon-batterie1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:186: undefined reference to `esp_ble_gap_stop_scanning'
collect2: error: ld returned 1 exit status
*** [/data/bms-addon-batterie1/.pioenvs/bms-addon-batterie1/firmware.elf] Error 1
```

[ESPHOME-Heltec BMS BLE - ESP32c3.txt](https://github.com/syssi/esphome-jk-bms/files/10319861/ESPHOME-Heltec.BMS.BLE.-.ESP32c3.txt)

syssi, 2022-12-29:
Did you try a clean build already?

```
esphome clean yourconfig.yaml
esphome run yourconfig.yaml
```

syssi, 2022-12-29:
Hmmm.. a clean build doesn't solve the linker issue. I'm able to reproduce the issue locally.

Nellor, 2022-12-29:
Yes I did. The board was brand new and it appeared on the first time to install it.
I found some similar issues when searching for "esp32_ble_tracker" for example at the esp32c3 but have absolutely no knowledge if this belongs to it as well...

[https://github.com/esphome/issues/issues/2648](url)

I added now:
sdkconfig_options:
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y

It now works without any errors (compiling, linking, flashing) but doesn't connect to the balancer...

syssi, 2022-12-29:
Good catch! You are on the right track! Please take a look at the log output of ESPHome (using the serial line) to get an better idea what's going on. Do you see any warnings or boot loops?

Nellor, 2022-12-29:
Im already looking into this... but just compile again... I'll keep you posted, thank you!

Nellor, 2022-12-29:
I'm a little bit further... as any reason no connection with the above sdkconfig options....

The log is attached..

[logs_heltec-balancer2_upload (1).txt](https://github.com/syssi/esphome-jk-bms/files/10320851/logs_heltec-balancer2_upload.1.txt)

Nellor, 2022-12-29:
Next step... with the arduino platform it works basically without any modification, with the drawback it might crash?.

esp32:
  board: seeed_xiao_esp32c3
  framework:
#  The arduino platform crashs
    type: arduino
    version: latest
#    type: esp-idf
#    version: latest

syssi, 2022-12-29:
If the device doesn't crash in the first BLE notification frame/data feel free to stick to the Arduino framework. In the past it wasn't possible to receive more than 1 frame without crashing afterwards.

Nellor, 2023-01-06:
I used the Arduino framework as I couldn't get it run with esp-idf...

closed 2023-01-06 by Nellor