syssi/esphome-jk-bms #231: Add a switch to file JK_BMS

firstlovez, 2023-01-09:
Hi Bro 
Please brother, add switch control function to the library to connect with RS485.
this code :
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via UART-TTL"
  external_components_source: github://syssi/esphome-jk-bms@main
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

# Please use the native `api` component instead of the `mqtt` section
# if you use Home Assistant. The native API is more lightweight.
#
# api:
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}

jk_modbus:
  id: modbus0
  uart_id: uart0
  rx_timeout: 50ms

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "${name} balancing"
    balancing_switch:
      name: "${name} balancing switch"
    charging:
      name: "${name} charging"
    charging_switch:
      name: "${name} charging switch"
    discharging:
      name: "${name} discharging"
    discharging_switch:
      name: "${name} discharging switch"
    dedicated_charger_switch:
      name: "${name} dedicated charger switch"

sensor:
  - platform: jk_bms
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
    cell_voltage_15:
      name: "${name} cell voltage 15"
    cell_voltage_16:
      name: "${name} cell voltage 16"
    cell_voltage_17:
      name: "${name} cell voltage 17"
    cell_voltage_18:
      name: "${name} cell voltage 18"
    cell_voltage_19:
      name: "${name} cell voltage 19"
    cell_voltage_20:
      name: "${name} cell voltage 20"
    cell_voltage_21:
      name: "${name} cell voltage 21"
    cell_voltage_22:
      name: "${name} cell voltage 22"
    cell_voltage_23:
      name: "${name} cell voltage 23"
    cell_voltage_24:
      name: "${name} cell voltage 24"
    power_tube_temperature:
      name: "${name} power tube temperature"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    capacity_remaining:
      name: "${name} capacity remaining"
    capacity_remaining_derived:
      name: "${name} capacity remaining derived"
    temperature_sensors:
      name: "${name} temperature sensors"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    battery_strings:
      name: "${name} battery strings"
    errors_bitmask:
      name: "${name} errors bitmask"
    operation_mode_bitmask:
      name: "${name} operation mode bitmask"
    total_voltage_overvoltage_protection:
      name: "${name} total voltage overvoltage protection"
    total_voltage_undervoltage_protection:
      name: "${name} total voltage undervoltage protection"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_overvoltage_delay:
      name: "${name} cell voltage overvoltage delay"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    cell_voltage_undervoltage_delay:
      name: "${name} cell voltage undervoltage delay"
    cell_pressure_difference_protection:
      name: "${name} cell pressure difference protection"
    discharging_overcurrent_protection:
      name: "${name} discharging overcurrent protection"
    discharging_overcurrent_delay:
      name: "${name} discharging overcurrent delay"
    charging_overcurrent_protection:
      name: "${name} charging overcurrent protection"
    charging_overcurrent_delay:
      name: "${name} charging overcurrent delay"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    balance_opening_pressure_difference:
      name: "${name} balance opening pressure difference"
    power_tube_temperature_protection:
      name: "${name} power tube temperature protection"
    power_tube_temperature_recovery:
      name: "${name} power tube temperature recovery"
    temperature_sensor_temperature_protection:
      name: "${name} temperature sensor temperature protection"
    temperature_sensor_temperature_recovery:
      name: "${name} temperature sensor temperature recovery"
    temperature_sensor_temperature_difference_protection:
      name: "${name} temperature sensor temperature difference protection"
    charging_high_temperature_protection:
      name: "${name} charging high temperature protection"
    discharging_high_temperature_protection:
      name: "${name} discharging high temperature protection"
    charging_low_temperature_protection:
      name: "${name} charging low temperature protection"
    charging_low_temperature_recovery:
      name: "${name} charging low temperature recovery"
    discharging_low_temperature_protection:
      name: "${name} discharging low temperature protection"
    discharging_low_temperature_recovery:
      name: "${name} discharging low temperature recovery"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    current_calibration:
      name: "${name} current calibration"
    device_address:
      name: "${name} device address"
    sleep_wait_time:
      name: "${name} sleep wait time"
    alarm_low_volume:
      name: "${name} alarm low volume"
    manufacturing_date:
      name: "${name} manufacturing date"
    total_runtime:
      name: "${name} total runtime"
#    start_current_calibration:
#      name: "${name} start current calibration"
    actual_battery_capacity:
      name: "${name} actual battery capacity"
#    protocol_version:
#      name: "${name} protocol version"

text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    battery_type:
      name: "${name} battery type"
    password:
      name: "${name} password"
    device_type:
      name: "${name} device type"
    software_version:
      name: "${name} software version"
    manufacturer:
      name: "${name} manufacturer"
    total_runtime_formatted:
      name: "${name} total runtime formatted"

[image]

Because Now it not display function switch on the Home assistance

syssi, 2023-01-09:
The RS485 communication protocol is read-only even if the specifications claim otherwise. The BMS cannot be controlled/configured via RS485 / UART-TTL.

syssi, 2023-01-09:
If you use BLE you can control the BMS using this implementation. The BMS supports two different protocols. The BLE communication supports write instructions the UART-TTL protocol doesn't accept write instructions.

The screenshot is a little bit outdated. It added some R/W registers as switch entities in the past because I wasn't aware of the "read-only limitation". Later on I migrated the entities to binary sensors (because they are read-only / not controllable).

closed 2023-02-05 by syssi