syssi/esphome-jk-bms #237: allow anonymous mqtt, eg. without username/passwort

abc2006, 2023-01-20:


syssi, 2023-01-20:
It looks like the `username` and `passwort` parameter is optional: https://esphome.io/components/mqtt.html

Feel free to remove these lines.

abc2006, 2023-01-20:
Hi, 
thanks for your answer.
Unfortunately, this does not work. Should have mentioned i tried that before: 

```
stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms$ esphome run esp32-example.yaml
INFO Reading configuration esp32-example.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Secret 'mqtt_username' not defined
  in "esp32-example.yaml", line 38, column 13:
      username: !secret mqtt_username
                ^
stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms$ cat secrets.yaml 
wifi_ssid: SNSH
wifi_password: thisisascecretpassword 
mqtt_host: 192.168.0.25

stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms$ 
```

abc2006, 2023-01-20:
leaving empty also does not work: 

`stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms$ cat secrets.yaml 
wifi_ssid: SNSH
wifi_password: thisisascecretpassword 
mqtt_host: 192.168.0.25
mqtt_username:
mqtt_password:

stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms$ esphome run esp32-example.yaml
INFO Reading configuration esp32-example.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
Failed config

mqtt: [source esp32-example.yaml:37]
  broker: !secret mqtt_host
  
  string value is None.
  username: 
  
  string value is None.
  password: 
  id: mqtt_client
stephan@stephan-desktop-neu:~/esp32/esphome-jk-bms`

syssi, 2023-01-20:
You have to reduce the `mqtt` component section from:

```
mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client
```

to

```
mqtt:
  broker: !secret mqtt_host
  id: mqtt_client
```

This builds fine.

closed 2023-01-20 by syssi