syssi/esphome-jk-bms #240: E (66371) BT_BTC: btc_gattc_cback transfer failed

austin202220, 2023-01-25:
**Issues:**

I am getting some errors in my logs and I suspect my esp32 keeps rebooting all the time. 

I have also noticed that my bms's report back as 'unavailable' status quite often, 3-4 times per hour or more. My bms, router, and esp32 are all within close proximity of less than 10 feet apart. I have tried several different esp32 chips and all are behaving the same way. I get same behavior when I run one single bms on one single esp32, and same behavior trying to run two bms's through one esp32. I believe each time they go to unavailable status the esp32 is rebooting. Also I notice by BMS's beep quite often on their own, but again, I think this is happening because the esp32 chip keeps rebooting.

**My Setup:**

I am monitoring 2 BMS's for this error report, both are connected to one ESP32-WROOM-32U
Inside esphome my esp32 chip/node is named 'jk-bms-esp32-01'
BMS #1: JK-B2A20S20P (in 16s config, device is named jk-bms-01 in home assistant and in the logs)
BMS #2: JK-BD6A20S10P (in 16s config, device is named jk-bms-02 in home assistant and in the logs)

Here are a few highlights, full logs posted seperately below: 

[13:35:34][D][esp-idf:000]: E (66371) BT_BTC: btc_gattc_cback transfer failed

[13:35:53][D][jk_bms_ble:630]: Unknown158: 64 00 (always 0x64 0x00?)
[13:35:53][D][jk_bms_ble:633]: Unknown160: 5E 06 (always 0xC5 0x09?)
[13:35:56][D][jk_bms_ble:899]:   Unknown6: 0.600000

[13:37:43][D][jk_bms_ble:630]: Unknown158: 64 00 (always 0x64 0x00?)
[13:37:43][D][jk_bms_ble:633]: Unknown160: E5 04 (always 0xC5 0x09?)
[13:37:43][D][sensor:127]: 'jk-bms-02 total runtime': Sending state 318313.00000 s with 0 decimals of accuracy
[13:37:43][D][text_sensor:067]: 'jk-bms-02 total runtime formatted': Sending state '3d 16h'
[13:37:43][D][jk_bms_ble:646]: Unknown168: 61.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.C3.03.34.00.00.00.A7.C9.3F.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00 (46)
[13:37:43][D][jk_bms_ble:660]: Unknown189: 03 34
[13:37:43][D][jk_bms_ble:664]: Unknown192: 00
[13:37:43][D][jk_bms_ble:666]: Unknown193: 00 A7 (0x00 0x8D)
[13:37:43][D][jk_bms_ble:668]: Unknown195: C9 3F (0x21 0x40)

I have these running separately, can be used for testing too if needed:
JK-B1A20S15P (pending install, will be 16s config)
JK-BD4A8S4P (two of these in 4s config, running on two different esp32's
[logs_jk-bms-esp32-01_logs (7).txt](https://github.com/syssi/esphome-jk-bms/files/10503227/logs_jk-bms-esp32-01_logs.7.txt)
)

[jk-bms-esp32-01 (2).yaml.txt](https://github.com/syssi/esphome-jk-bms/files/10503257/jk-bms-esp32-01.2.yaml.txt)

syssi, 2023-01-25:
Please try the following experiment:

1. Talk to a single BMS
2. Reduce the number of entities to f.e. 3 sensors
3. Decrease the log level from `DEBUG` to `INFO`
4. Use the native `api` instead of `mqtt` if possible

This will free a lot of resources / reduce the load. Every beep of the BMS indicates a "reconnect" it doesn't need to be a reboot of the ESP. In best case you can consume the log of ESPHome using the serial interface instead of using WiFi. In this case you should see every crash or BLE connection retry. Does this experiment stabilize the BLE connection / reduce the number of BMS beeps per hour?

austin202220, 2023-01-25:
> 4\. Use the native `api` instead of `mqtt` if possible

How do I make the change to use native apt instead of mqtt? From reading my yaml code I dont see any references to mqtt?

syssi, 2023-01-25:
If you use the native `api` already there should be a `api:` section at your yaml. If you are using MQTT there should be an `mqtt:` section plus some parameters.

austin202220, 2023-01-25:
I posted my full yaml config as an attachment in the original post, I am already using the native api and not mqtt

```
substitutions:
  name: jk-bms
  bms0: "${name}-01"
  bms1: "${name}-02"
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  bms0_mac_address: C8:47:8C:xx:xx:xx
  bms1_mac_address: C8:47:8C:xx:xx:xx
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  bms0_protocol_version: JK02
  bms1_protocol_version: JK02

esphome:
  name: ${name}-esp32-01
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:
    level: debug

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:

ble_client:
  - mac_address: ${bms0_mac_address}
    id: client0
  - mac_address: ${bms1_mac_address}
    id: client1

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${bms0_protocol_version}
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    protocol_version: ${bms1_protocol_version}
    throttle: 5s
    id: bms1

binary_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balancing:
[... 332 lines cut ...]
    balancer:
      name: "${bms1} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${bms0} enable bluetooth connection"
  - platform: ble_client
    ble_client_id: client1
    name: "${bms1} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    errors:
      name: "${bms0} errors"
    total_runtime_formatted:
      name: "${bms0} total runtime formatted"

  - platform: jk_bms_ble
    jk_bms_ble_id: bms1
    errors:
      name: "${bms1} errors"
    total_runtime_formatted:
      name: "${bms1} total runtime formatted"
```

syssi, 2023-01-25:
Sorry. I've missed the second attachment.

austin202220, 2023-01-25:
I will try changing to one bms only and reducing the sensors next but that will require additional time and testing. I'll work on those changes a little later.

For now, I have removed all of the cell resistance entities from the yaml code to try to reduce the number of sensors some, and changed log level to info. Still getting one error is red text right away:

[15:30:49][I][esp-idf:000]: E (152153) BT_BTC: btc_gattc_cback transfer failed

```
INFO Reading configuration /config/esphome/jk-bms-esp32-01.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms-esp32-01.local using esphome API
INFO Successfully connected to jk-bms-esp32-01.local
[15:29:18][I][app:102]: ESPHome version 2022.12.6 compiled on Jan 25 2023, 15:14:45
[15:29:18][I][app:104]: Project syssi.esphome-jk-bms version 1.3.0
[15:29:19][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:19][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:19][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[15:29:21][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:21][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:24][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:24][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:27][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:27][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:29][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:29][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:32][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:32][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:34][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:34][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:38][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:38][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:40][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:40][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:43][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:43][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:45][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:45][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:49][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:49][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:50][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:50][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:54][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:54][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:29:56][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:29:56][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:00][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:00][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:01][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:01][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:05][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:05][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:06][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:06][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:11][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:11][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:11][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:11][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:16][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:16][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:17][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:17][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:21][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:21][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:22][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:22][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:27][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:27][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:27][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:27][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:30:32][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:30:32][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[... 131 lines cut ...]
[15:33:26][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:26][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:27][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:27][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:31][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:31][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:32][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:32][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:37][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:37][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:37][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:37][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:42][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:42][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:42][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[... cut ...]
[15:33:48][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:48][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:48][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:48][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:53][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:53][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[15:33:53][I][jk_bms_ble:516]: Enabled cells: FF FF 00 00
[15:33:53][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
```

syssi, 2023-01-25:
Another idea: You could increase the `throttle` option to `10s` or `30s` to publish state updates every 30 seconds and skip/discard the incoming data in between.

syssi, 2023-02-20:
Can we close this issue? If have no idea how to fix the ESP-IDF issue:

```
[15:30:49][I][esp-idf:000]: E (152153) BT_BTC: btc_gattc_cback transfer failed
```

but it looks like the communication is able to recover and is working in general. I assume the issue will disappear with one of the next ESP-IDF/ESPHome releases.