syssi/esphome-jk-bms #243: JK_B2A8S20P (SW V11.14H)   current sensor not working

starfish1107, 2023-02-02:
the current sensor shows 0  despite app shows some current.

syssi, 2023-02-03:
Could yout provide some more details? Do use use UARR-TTL or BLE? Could you increase the log level (see the "Debug" section of the logs) and provide some logs? I'm interested in some raw frames of your BMS which is decoded to 0 amps but should decoded to something else.

starfish1107, 2023-02-03:
found the reason. It has nothing to do with esphome-jk-bms fortunately. my jk-bms unit shows 0  below 0,7A and above it shows something but not the correct value. This number is shown in the app and in esphome-jk-bms as well - so everything ok on esphome side. this issue could be deleted or closed.

closed 2023-02-03 by syssi