syssi/esphome-jk-bms #247: not work jk bms bd6a20s10p

amgstone, 2023-02-12:
```
INFO Reading configuration /config/esphome/jk-bms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms.local using esphome API
INFO Successfully connected to jk-bms.local
[12:14:06][I][app:102]: ESPHome version 2022.12.8 compiled on Feb 11 2023, 12:56:38
[12:14:06][I][app:104]: Project syssi.esphome-jk-bms version 1.3.0
[12:14:06][C][wifi:504]: WiFi:
[12:14:06][C][wifi:362]:   Local MAC: C4:5B:BE:48:9F:CB
[12:14:06][C][wifi:363]:   SSID: [redacted]
[12:14:06][C][wifi:364]:   IP Address: 192.168.31.154
[12:14:06][C][wifi:365]:   BSSID: [redacted]
[12:14:06][C][wifi:367]:   Hostname: 'jk-bms'
[12:14:06][C][wifi:369]:   Signal strength: -78 dB ▂▄▆█
[12:14:06][C][wifi:373]:   Channel: 1
[12:14:06][C][wifi:374]:   Subnet: 255.255.255.0
[12:14:06][C][wifi:375]:   Gateway: 192.168.31.1
[12:14:06][C][wifi:376]:   DNS1: 192.168.31.1
[12:14:06][C][wifi:377]:   DNS2: 8.8.8.8
[12:14:06][C][logger:293]: Logger:
[12:14:06][C][logger:294]:   Level: DEBUG
[12:14:06][C][logger:295]:   Log Baud Rate: 115200
[12:14:06][C][logger:296]:   Hardware UART: UART0
[12:14:06][C][uart.arduino_esp8266:102]: UART Bus:
[12:14:06][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[12:14:06][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[12:14:06][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[12:14:06][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[12:14:06][C][uart.arduino_esp8266:109]:   Data Bits: 8
[12:14:06][C][uart.arduino_esp8266:110]:   Parity: NONE
[12:14:06][C][uart.arduino_esp8266:111]:   Stop bits: 1
[12:14:06][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[12:14:06][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
[12:14:06][C][jk_modbus:097]: JkModbus:
[12:14:06][C][jk_modbus:098]:  RX timeout: 50 ms
[12:14:06][C][jk_bms:487]: JkBms:
[12:14:06][C][jk_bms:488]:   Address: 0x4E
[12:14:06][C][jk_bms:489]:   Fake traffic enabled: NO
[12:14:06][C][jk_bms:490]: Minimum Cell Voltage 'jk-bms min cell voltage'
[12:14:06][C][jk_bms:490]:   Device Class: 'voltage'
[12:14:06][C][jk_bms:490]:   State Class: 'measurement'
[12:14:06][C][jk_bms:490]:   Unit of Measurement: 'V'
[12:14:06][C][jk_bms:490]:   Accuracy Decimals: 3
[12:14:06][C][jk_bms:491]: Maximum Cell Voltage 'jk-bms max cell voltage'
[12:14:06][C][jk_bms:491]:   Device Class: 'voltage'
[12:14:06][C][jk_bms:491]:   State Class: 'measurement'
[12:14:06][C][jk_bms:491]:   Unit of Measurement: 'V'
[12:14:06][C][jk_bms:491]:   Accuracy Decimals: 3
[12:14:06][C][jk_bms:492]: Minimum Voltage Cell 'jk-bms min voltage cell'
[12:14:06][C][jk_bms:492]:   State Class: 'measurement'
[12:14:06][C][jk_bms:492]:   Unit of Measurement: ''
[12:14:06][C][jk_bms:492]:   Accuracy Decimals: 0
[12:14:06][C][jk_bms:492]:   Icon: 'mdi:battery-minus-outline'
[12:14:06][C][jk_bms:493]: Maximum Voltage Cell 'jk-bms max voltage cell'
[12:14:06][C][jk_bms:493]:   State Class: 'measurement'
[12:14:06][C][jk_bms:493]:   Unit of Measurement: ''
[12:14:06][C][jk_bms:493]:   Accuracy Decimals: 0
[12:14:06][C][jk_bms:493]:   Icon: 'mdi:battery-plus-outline'
[12:14:06][C][jk_bms:494]: Delta Cell Voltage 'jk-bms delta cell voltage'
[12:14:06][C][jk_bms:494]:   Device Class: 'voltage'
[12:14:06][C][jk_bms:494]:   State Class: 'measurement'
[12:14:06][C][jk_bms:494]:   Unit of Measurement: 'V'
[12:14:06][C][jk_bms:494]:   Accuracy Decimals: 3
[12:14:06][C][jk_bms:495]: Average Cell Voltage 'jk-bms average cell voltage'
[12:14:06][C][jk_bms:495]:   Device Class: 'voltage'
[12:14:06][C][jk_bms:495]:   State Class: 'measurement'
[12:14:06][C][jk_bms:495]:   Unit of Measurement: 'V'
[12:14:06][C][jk_bms:495]:   Accuracy Decimals: 3
[12:14:06][C][jk_bms:496]: Cell Voltage 1 'jk-bms cell voltage 1'
[12:14:06][C][jk_bms:496]:   Device Class: 'voltage'
[... 620 lines cut ...]
#    start_current_calibration:
#      name: "${name} start current calibration"
    actual_battery_capacity:
      name: "${name} actual battery capacity"
#    protocol_version:
#      name: "${name} protocol version"

text_sensor:
  - platform: jk_bms
    errors:
      name: "${name} errors"
    operation_mode:
      name: "${name} operation mode"
    battery_type:
      name: "${name} battery type"
    password:
      name: "${name} password"
    device_type:
      name: "${name} device type"
    software_version:
      name: "${name} software version"
    manufacturer:
      name: "${name} manufacturer"
    total_runtime_formatted:
      name: "${name} total runtime formatted"`

syssi, 2023-02-12:
Did you see this warning:

```
[12:14:06][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[12:14:06][W][uart.arduino_esp8266:127]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
```

At the moment the `logger` component and the `jk_bms` components reads & writes to GPIO1/3. Please disable the serial logging using the `baud_rate: 0` option or use another GPIO set for the `uart` component.

amgstone, 2023-02-12:
```
INFO Reading configuration /config/esphome/jk-bms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms.local using esphome API
INFO Successfully connected to jk-bms.local
[13:48:28][I][app:102]: ESPHome version 2022.12.8 compiled on Feb 12 2023, 13:41:31
[13:48:28][I][app:104]: Project syssi.esphome-jk-bms version 1.3.0
[13:48:28][C][wifi:504]: WiFi:
[13:48:28][C][wifi:362]:   Local MAC: C4:5B:BE:48:9F:CB
[13:48:28][C][wifi:363]:   SSID: [redacted]
[13:48:28][C][wifi:364]:   IP Address: 192.168.31.154
[13:48:28][C][wifi:365]:   BSSID: [redacted]
[13:48:28][C][wifi:367]:   Hostname: 'jk-bms'
[13:48:28][C][wifi:369]:   Signal strength: -72 dB ▂▄▆█
[13:48:28][C][wifi:373]:   Channel: 1
[13:48:28][C][wifi:374]:   Subnet: 255.255.255.0
[13:48:28][C][wifi:375]:   Gateway: 192.168.31.1
[13:48:28][C][wifi:376]:   DNS1: 192.168.31.1
[13:48:28][C][wifi:377]:   DNS2: 8.8.8.8
[13:48:28][C][logger:293]: Logger:
[13:48:28][C][logger:294]:   Level: DEBUG
[13:48:28][C][logger:295]:   Log Baud Rate: 0
[13:48:28][C][logger:296]:   Hardware UART: UART0
[13:48:28][C][uart.arduino_esp8266:102]: UART Bus:
[13:48:28][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[13:48:28][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[13:48:28][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[13:48:28][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[13:48:28][C][uart.arduino_esp8266:109]:   Data Bits: 8
[13:48:28][C][uart.arduino_esp8266:110]:   Parity: NONE
[13:48:28][C][uart.arduino_esp8266:111]:   Stop bits: 1
[13:48:28][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[13:48:28][C][jk_modbus:097]: JkModbus:
[13:48:28][C][jk_modbus:098]:  RX timeout: 50 ms
[13:48:28][C][jk_bms:487]: JkBms:
[13:48:28][C][jk_bms:488]:   Address: 0x4E
[13:48:28][C][jk_bms:489]:   Fake traffic enabled: NO
[13:48:28][C][jk_bms:490]: Minimum Cell Voltage 'jk-bms min cell voltage'
[13:48:28][C][jk_bms:490]:   Device Class: 'voltage'
[13:48:28][C][jk_bms:490]:   State Class: 'measurement'
[13:48:28][C][jk_bms:490]:   Unit of Measurement: 'V'
[13:48:28][C][jk_bms:490]:   Accuracy Decimals: 3
[13:48:28][C][jk_bms:491]: Maximum Cell Voltage 'jk-bms max cell voltage'
[13:48:28][C][jk_bms:491]:   Device Class: 'voltage'
[13:48:28][C][jk_bms:491]:   State Class: 'measurement'
[13:48:28][C][jk_bms:491]:   Unit of Measurement: 'V'
[13:48:28][C][jk_bms:491]:   Accuracy Decimals: 3
[13:48:28][C][jk_bms:492]: Minimum Voltage Cell 'jk-bms min voltage cell'
[13:48:28][C][jk_bms:492]:   State Class: 'measurement'
[13:48:28][C][jk_bms:492]:   Unit of Measurement: ''
[13:48:28][C][jk_bms:492]:   Accuracy Decimals: 0
[13:48:28][C][jk_bms:492]:   Icon: 'mdi:battery-minus-outline'
[13:48:28][C][jk_bms:493]: Maximum Voltage Cell 'jk-bms max voltage cell'
[13:48:28][C][jk_bms:493]:   State Class: 'measurement'
[13:48:28][C][jk_bms:493]:   Unit of Measurement: ''
[13:48:28][C][jk_bms:493]:   Accuracy Decimals: 0
[13:48:28][C][jk_bms:493]:   Icon: 'mdi:battery-plus-outline'
[13:48:28][C][jk_bms:494]: Delta Cell Voltage 'jk-bms delta cell voltage'
[13:48:28][C][jk_bms:498]:   Accuracy Decimals: 3
[13:48:28][C][jk_bms:499]: Cell Voltage 4 'jk-bms cell voltage 4'
[13:48:28][C][jk_bms:499]:   Device Class: 'voltage'
[13:48:28][C][jk_bms:499]:   State Class: 'measurement'
[13:48:28][C][jk_bms:499]:   Unit of Measurement: 'V'
[13:48:28][C][jk_bms:499]:   Accuracy Decimals: 3
[13:48:28][C][jk_bms:500]: Cell Voltage 5 'jk-bms cell voltage 5'
[13:48:28][C][jk_bms:500]:   Device Class: 'voltage'
[13:48:28][C][jk_bms:500]:   State Class: 'measurement'
[13:48:28][C][jk_bms:500]:   Unit of Measurement: 'V'
[13:48:28][C][jk_bms:500]:   Accuracy Decimals: 3
[13:48:28][C][jk_bms:501]: Cell Voltage 6 'jk-bms cell voltage 6'
[... 267 lines cut ...]
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
[13:48:51][W][jk_modbus:050]: Invalid header
```

not work http://joxi.ru/YmEpbEWFwEdGor

syssi, 2023-02-12:
Please enable the debug mode of the `uart` component by updating the `uart` section to:

```
uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
```

Please flash again and provide another log. This time the raw traffic should be part of the log.

amgstone, 2023-02-12:
install
```
INFO Successfully compiled program.
INFO Resolving IP address of jk-bms.local
INFO  -> 192.168.31.154
INFO Uploading /data/jk-bms/.pioenvs/jk-bms/firmware.bin (422976 bytes)
INFO Compressed to 284926 bytes
Uploading: [=====================================================       ] 88% 
ERROR Error sending data: timed out
```

syssi, 2023-02-12:
If the ESP it pretty busy OTA can fail every now and then. You could add a safe mode switch to your setup to boot the device into the safe mode before doing an OTA: https://esphome.io/components/switch/safe_mode.html

amgstone, 2023-02-12:
```
INFO Reading configuration /config/esphome/jk-bms.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Starting log output from jk-bms.local using esphome API
INFO Successfully connected to jk-bms.local
[18:45:49][I][app:102]: ESPHome version 2022.12.8 compiled on Feb 12 2023, 18:36:27
[18:45:49][I][app:104]: Project syssi.esphome-jk-bms version 1.3.0
[18:45:49][C][wifi:504]: WiFi:
[18:45:49][C][wifi:362]:   Local MAC: C4:5B:BE:48:9F:CB
[18:45:49][C][wifi:363]:   SSID: [redacted]
[18:45:49][C][wifi:364]:   IP Address: 192.168.31.154
[18:45:49][C][wifi:365]:   BSSID: [redacted]
[18:45:49][C][wifi:367]:   Hostname: 'jk-bms'
[18:45:49][C][wifi:369]:   Signal strength: -70 dB ▂▄▆█
[18:45:49][C][wifi:373]:   Channel: 1
[18:45:49][C][wifi:374]:   Subnet: 255.255.255.0
[18:45:49][C][wifi:375]:   Gateway: 192.168.31.1
[18:45:49][C][wifi:376]:   DNS1: 192.168.31.1
[18:45:49][C][wifi:377]:   DNS2: 8.8.8.8
[18:45:49][C][logger:293]: Logger:
[18:45:49][C][logger:294]:   Level: DEBUG
[18:45:49][C][logger:295]:   Log Baud Rate: 0
[18:45:49][C][logger:296]:   Hardware UART: UART0
[18:45:49][C][uart.arduino_esp8266:102]: UART Bus:
[18:45:49][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[18:45:49][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[18:45:49][C][uart.arduino_esp8266:106]:   RX Buffer Size: 384
[18:45:49][C][uart.arduino_esp8266:108]:   Baud Rate: 115200 baud
[18:45:49][C][uart.arduino_esp8266:109]:   Data Bits: 8
[18:45:49][C][uart.arduino_esp8266:110]:   Parity: NONE
[18:45:49][C][uart.arduino_esp8266:111]:   Stop bits: 1
[18:45:49][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[18:45:49][C][jk_modbus:097]: JkModbus:
[18:45:49][C][jk_modbus:098]:  RX timeout: 50 ms
[18:45:49][C][jk_bms:487]: JkBms:
[18:45:49][C][jk_bms:488]:   Address: 0x4E
[18:45:49][C][jk_bms:489]:   Fake traffic enabled: NO
[18:45:49][C][jk_bms:490]: Minimum Cell Voltage 'jk-bms min cell voltage'
[18:45:49][C][jk_bms:490]:   Device Class: 'voltage'
[18:45:49][C][jk_bms:490]:   State Class: 'measurement'
[18:45:49][C][jk_bms:490]:   Unit of Measurement: 'V'
[18:45:49][C][jk_bms:490]:   Accuracy Decimals: 3
[18:45:49][C][jk_bms:491]: Maximum Cell Voltage 'jk-bms max cell voltage'
[18:45:49][C][jk_bms:491]:   Device Class: 'voltage'
[18:45:49][C][jk_bms:491]:   State Class: 'measurement'
[18:45:49][C][jk_bms:491]:   Unit of Measurement: 'V'
[18:45:49][C][jk_bms:491]:   Accuracy Decimals: 3
[18:45:49][C][jk_bms:492]: Minimum Voltage Cell 'jk-bms min voltage cell'
[18:45:49][C][jk_bms:492]:   State Class: 'measurement'
[18:45:49][C][jk_bms:492]:   Unit of Measurement: ''
[18:45:49][C][jk_bms:492]:   Accuracy Decimals: 0
[18:45:49][C][jk_bms:492]:   Icon: 'mdi:battery-minus-outline'
[18:45:49][C][jk_bms:493]: Maximum Voltage Cell 'jk-bms max voltage cell'
[18:45:49][C][jk_bms:493]:   State Class: 'measurement'
[18:45:49][C][jk_bms:493]:   Unit of Measurement: ''
[18:45:49][C][jk_bms:493]:   Accuracy Decimals: 0
[18:45:49][C][jk_bms:493]:   Icon: 'mdi:battery-plus-outline'
[18:45:49][C][jk_bms:494]: Delta Cell Voltage 'jk-bms delta cell voltage'
[18:45:49][C][jk_bms:494]:   Device Class: 'voltage'
[18:45:49][C][jk_bms:494]:   State Class: 'measurement'
[18:45:49][C][jk_bms:494]:   Unit of Measurement: 'V'
[18:45:49][C][jk_bms:494]:   Accuracy Decimals: 3
[18:45:49][C][jk_bms:495]: Average Cell Voltage 'jk-bms average cell voltage'
[18:45:49][C][jk_bms:495]:   Device Class: 'voltage'
[18:45:49][C][jk_bms:495]:   State Class: 'measurement'
[18:45:49][C][jk_bms:495]:   Unit of Measurement: 'V'
[18:45:49][C][jk_bms:495]:   Accuracy Decimals: 3
[18:45:49][C][jk_bms:496]: Cell Voltage 1 'jk-bms cell voltage 1'
[18:45:49][C][jk_bms:496]:   Device Class: 'voltage'
[18:45:49][C][jk_bms:496]:   State Class: 'measurement'
[... 125 lines cut ...]
INFO Successfully connected to jk-bms.local
[18:46:08][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[18:46:08][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:80:00:06:00:01:79:2A:01:0F:E5:02:0F:E4:03:0F:E3:04:0F:E4:05:0F:E3:06:0F:E2:07:0F:E2:08:0F:E5:09:0F:E4:0A:0F:E3:0B:0F:E7:0C:0F:E5:0D:0F:E4:0E:0F:E5:80:00:16:81:00:12:82:00:12:83:16:3F:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:F9:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[18:46:08][W][jk_modbus:075]: JkModbus CRC Check failed! 5968!=58E4
[18:46:08][D][uart_debug:114]: <<< A2:00:14:A3:00:46:A4:04:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:2D:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E4
[18:46:13][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
INFO jk-bms.local: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO jk-bms.local: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for jk-bms.local
WARNING Disconnected from API
INFO Successfully connected to jk-bms.local
[18:46:27][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[... cut ...]
[18:46:43][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
INFO jk-bms.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO jk-bms.local: Ping Failed: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO Disconnected from ESPHome API for jk-bms.local
WARNING Disconnected from API
INFO Successfully connected to jk-bms.local
[18:46:58][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
```

syssi, 2023-02-12:
This is the important part of the `uart` traffic:

```
[18:46:08][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[18:46:08][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:80:00:06:00:01:79:2A:01:0F:E5:02:0F:E4:03:0F:E3:04:0F:E4:05:0F:E3:06:0F:E2:07:0F:E2:08:0F:E5:09:0F:E4:0A:0F:E3:0B:0F:E7:0C:0F:E5:0D:0F:E4:0E:0F:E5:80:00:16:81:00:12:82:00:12:83:16:3F:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:F9:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[18:46:08][W][jk_modbus:075]: JkModbus CRC Check failed! 5968!=58E4
[18:46:08][D][uart_debug:114]: <<< A2:00:14:A3:00:46:A4:04:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:2D:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E4
[18:46:13][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[18:46:27][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[18:46:58][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
```

The BMS did respond to **one** of the requests but in 3 of 4 times there was no response. This is strange. The received response had a frame length of 285 bytes. Could you tell me the number of cells of your setup?

amgstone, 2023-02-12:
14

syssi, 2023-02-12:
The frame length fits perfectly for a 14S system. It's strange the CRC doesn't match. How long is the cable between your BMS and the ESP?

amgstone, 2023-02-12:
1 meter

syssi, 2023-02-12:
Could you provide some more response frames / log messages? I would like to know the CRC value is always off by the same offset (84).

amgstone, 2023-02-12:
`[22:05:55][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
INFO jk-bms.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO jk-bms.local: Ping Failed: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO Disconnected from ESPHome API for jk-bms.local
WARNING Disconnected from API
INFO Successfully connected to jk-bms.local
[22:06:10][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
INFO jk-bms.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO jk-bms.local: Ping Failed: Error while reading data: 0 bytes read on a total of 3 expected bytes
INFO Disconnected from ESPHome API for jk-bms.local
WARNING Disconnected from API
INFO Successfully connected to jk-bms.local
[22:06:25][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[22:06:25][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D8:03:0F:D7:04:0F:D7:05:0F:D7:06:0F:D5:07:0F:D5:08:0F:D8:09:0F:D9:0A:0F:D8:0B:0F:D8:0C:0F:D8:0D:0F:D9:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:10:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[22:06:25][W][jk_modbus:075]: JkModbus CRC Check failed! 58FE!=58ED
[22:06:25][D][uart_debug:114]: <<< A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:15:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F4:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:ED
[22:06:30][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[22:06:30][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D5:03:0F:D7:04:0F:D8:05:0F:D4:06:0F:D5:87:0F:D5:08:0F:D7:09:0F:D9:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D8:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2E:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:40:4B:19:44:61:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:02:64:A2
[22:06:30][D][uart_debug:114]: <<< 00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:73:30:31:B6:00:00:BC:F4:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:04:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E7
[22:06:35][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[22:06:35][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D7:04:0F:D8:05:0F:D5:06:0F:D7:07:0F:D5:08:0F:D9:09:0F:D9:0A:0F:D7:0B:0F:D8:0C:0F:D9:0D:0F:D7:0E:0F:D8:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E5:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[22:06:35][W][jk_modbus:075]: JkModbus CRC Check failed! 5912!=58EC
[22:06:35][D][uart_debug:114]: <<< A2:00:14:A3:00:46:B4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:59:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:4D:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:69:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:EC
[22:06:40][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[22:06:40][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D4:04:0F:D5:05:0F:D8:06:0F:D7:07:0F:D5:08:0F:D7:09:0F:D8:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D7:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9E:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64
[22:06:40][W][jk_modbus:075]: JkModbus CRC Check failed! 58EF!=58E7
[22:06:40][D][uart_debug:114]: <<< A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:04:00:00:00:68:00:00:58:E7
[22:06:45][D][uart_debug:114]: >>> 4E:57:00:13:00:00:00:00:06:03:00:00:00:00:00:00:68:00:00:01:29
[22:06:45][D][uart_debug:114]: <<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D7:04:0F:D7:05:0F:D5:06:0F:D4:07:0F:D4:08:0F:D8:09:0F:D9:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D8:0E:0F:D8: [... cut ...]
[22:09:10][W][jk_modbus:075]: JkModbus CRC Check failed! 58F7!=58ED
[22:09:10][D][uart_debug:114]: <<< A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F7:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:ED
`

syssi, 2023-02-12:
Summary:

```
<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D8:03:0F:D7:04:0F:D7:05:0F:D7:06:0F:D5:07:0F:D5:08:0F:D8:09:0F:D9:0A:0F:D8:0B:0F:D8:0C:0F:D8:0D:0F:D9:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:10:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:15:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F4:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:ED
CRC Check failed! 58FE!=58ED

<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D5:03:0F:D7:04:0F:D8:05:0F:D4:06:0F:D5:87:0F:D5:08:0F:D7:09:0F:D9:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D8:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2E:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:40:4B:19:44:61:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:02:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:73:30:31:B6:00:00:BC:F4:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:04:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E7
CRC OK?

<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D7:04:0F:D8:05:0F:D5:06:0F:D7:07:0F:D5:08:0F:D9:09:0F:D9:0A:0F:D7:0B:0F:D8:0C:0F:D9:0D:0F:D7:0E:0F:D8:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E5:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64:A2:00:14:A3:00:46:B4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:59:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:4D:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:69:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:EC
CRC Check failed! 5912!=58EC

<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D4:04:0F:D5:05:0F:D8:06:0F:D7:07:0F:D5:08:0F:D7:09:0F:D8:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D7:0E:0F:D9:80:00:16:81:00:12:82:00:12:83:16:2D:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9E:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:04:00:00:00:68:00:00:58:E7
CRC Check failed! 58EF!=58E7

<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D8:02:0F:D7:03:0F:D7:04:0F:D7:05:0F:D5:06:0F:D4:07:0F:D4:08:0F:D8:09:0F:D9:0A:0F:D7:0B:0F:D9:0C:0F:D8:0D:0F:D8:0E:0F:D8:80:00:16:81:10:12:82:00:12:83:16:2C:84:00:38:85:5B:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:91:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:00:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:80:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:FC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:10:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E6
CRC Check failed! 59C7!=58E6

<<< 4E:57:01:1B:00:00:00:00:06:00:01:79:2A:01:0F:D7:02:0F:D7:03:0F:D5:04:0F:D7:05:0F:D7:06:0F:D5:07:0F:D4:08:0F:D9:09:0F:D8:0A:0F:D5:0B:0F:D9:0C:0F:D7:0D:0F:D8:0E:0F:D8:80:00:16:81:00:12:82:00:12:83:16:2C:84:00:3C:85:5A:86:02:87:00:07:89:00:00:09:FB:8A:00:0E:8B:00:00:8C:00:03:8E:16:F8:8F:0F:6C:90:10:68:91:10:54:92:00:05:93:0B:04:94:0B:22:95:00:05:96:01:2C:97:00:64:98:01:2C:99:00:50:9A:00:1E:9B:0C:E4:9C:00:14:9D:01:9E:00:5A:9F:00:46:A0:00:64:A1:01:64:A2:00:14:A3:00:46:A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:EE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F5:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:10:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:6C:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:E2
CRC Check failed! 593B!=58E2

[... cut ...] :A4:00:46:A5:FF:EC:A6:FF:F6:A7:FF:EC:A8:FF:F6:A9:0E:AA:00:00:01:54:AB:01:AC:01:AD:03:A6:AE:01:AF:01:B0:00:0A:B1:14:B2:31:32:33:34:35:36:00:00:00:00:B3:00:B4:49:72:65:6E:61:5F:68:6F:B5:32:33:30:31:B6:00:00:BC:F7:B7:31:30:2E:58:57:5F:53:31:30:2E:31:30:5F:5F:5F:B8:00:B9:00:00:01:54:BA:49:72:65:6E:61:5F:68:6F:75:73:65:00:49:72:65:6E:61:68:6F:75:73:65:5F:34:C0:01:00:00:00:00:68:00:00:58:ED
CRC Check failed! 58F7!=58ED
```

It will take some time to analyze the traffic and get an idea what's going on.

syssi, 2023-02-16:
May be you could try to replace and reduce the length of the data cable between the ESP and the BMS in the meantime.

amgstone, 2023-02-16:
put the bluetooth version, everything works

syssi, 2023-02-20:
Could you try to reduce the cable length? I'm pretty sure the transmission errors can be solved by reducing the cable length to 10cm.

syssi, 2023-02-21:
I'm pretty sure this communication can be fixed by reducing the length of the wire.

closed 2023-02-21 by syssi

amgstone, 2023-02-25:
after disconnecting bluetooth and then connecting bms does not reconnect to esp.

syssi, 2023-02-25:
The BMS only accepts new connections if the LED is blinking. If the LED is solid there is an active connection already. If you are sure there is no other BLE client talking to the BMS just wait some time. As soon the LED blinks again the ESP32 should be able to connect.

amgstone, 2023-02-25:
after restarting the bms, the connection was immediately restored

DragotaDarius, 2023-04-29:
I have the same problem as amgstone regarding the crc fail and invalid headers warnings.
I have a direct connection to the bms with the GPS port, aka the ttl, wired to my esp8266 nodemcu main uart.
The wire is around 20cm from the nodemcu to the port on the bms so I would exclude the length issue.
[image]
[logs_esphome-web-0bbb4b_logs.txt](https://github.com/syssi/esphome-jk-bms/files/11359224/logs_esphome-web-0bbb4b_logs.txt)

syssi, 2023-04-29:
Did you buy your BMS with or without CAN bus support?

DragotaDarius, 2023-05-03:
I think it is with both, hence in the menu I can select both.
I may have found the issue, but I haven't investigated with the raw data.
As you can see in the image attached, when the bms is in discharge mode, the data gets read via the ttl port(I have disconnected the rs485 adapter for the jk with my rs485 to ttl and connected the esp8266 directly to the GPS port).
When the battery is charging, it seems that the esp cannot read properly the values and you see the straight line to the upper SoC.
[image]
The logs that I sent the last time seem to match exactly the time the battery was charging, as shown underneath, around 4pm local time in the image:
[image]
I assume that something does not match the type of expected data length and thus failing to read what comes during the charging period.