syssi/esphome-jk-bms #249: JK-B2A20S20P HW V11.XW, SW V11.25H working well via ESP32 BLE

iovcharyk, 2023-02-12:
I would like to say big thank you to Sebastian for providing the solution. 

Here is my code in HA:
`substitutions:
  name: jk-bms
  
esphome:
  name: jk-bms  
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: nodemcu-32s  
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

esp32_ble_tracker:

ble_client:
  - mac_address: C8:47:8C:E9:49:4E
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: JK02
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_voltage_9:
      name: "${name} cell voltage 9"
    cell_voltage_10:
      name: "${name} cell voltage 10"
    cell_voltage_11:
      name: "${name} cell voltage 11"
    cell_voltage_12:
      name: "${name} cell voltage 12"
    cell_voltage_13:
      name: "${name} cell voltage 13"
    cell_voltage_14:
      name: "${name} cell voltage 14"
      
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"    
           

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"
  - platform: restart
    name: "Jk-Esp32-Switch"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
    `

syssi, 2023-02-12:
Thanks for your feedback! I will extend the docs.

closed 2023-02-12 by syssi

syssi, 2023-04-18:
@iovcharyk Please use protocol version `JK02_32S` and avoid the auto-detection of the new (JK02_32S) frames because it's unstable.

iovcharyk, 2023-07-06:
Hi @syssi, thanks, now I'm using protocol version `JK02_32S`
But in order to compile it I have to use framework: type: arduino instead of "esp-idf"

commit 87c95d3d (2023-02-12): Add JK-B2A20S20P v11 as supported (Closes: #249) (#248)