syssi/esphome-jk-bms #252: add a utility meter to the firmware

amgstone, 2023-02-18:
add a utility meter to the firmware to connect to the "Battery" dashboard. So that you can add and see how much went into the accumulator and how much came out.

syssi, 2023-02-18:
You can find the solution here: https://github.com/syssi/esphome-jk-bms/issues/80

amgstone, 2023-02-18:
I can insert a utility meter into this firmware

```
substitutions:
  name: ble
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E8:FF:73
  protocol_version: JK02

esphome:
  name: ${name}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

logger:

api:
  encryption:
    key: "eqVtRZEcrMMZUMkO7GvrSCyCe//cZuQnIsvLbIJ2iZQ="

ota:
  password: "e1d447bf8bc832253849bbb3867e6e86"

wifi:
  ssid: "Irena_house"
  password: "1212121212"

  ap:
    ssid: "Ble Fallback Hotspot"
    password: "6T8A5xN6qtU3"

esp32_ble_tracker:

time:
  - platform: sntp

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_count:
      name: "${name} cell count"
    total_battery_capacity:
      name: "${name} total battery capacity"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    voltage_calibration:
      name: "${name} voltage calibration"
    current_calibration:
      name: "${name} current calibration"
    power_off_voltage:
      name: "${name} power off voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```

amgstone, 2023-02-19:
utility meter dont work

```
substitutions:
  name: ble
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E8:FF:73
  protocol_version: JK02

esphome:
  name: ${name}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

logger:

api:
  encryption:
    key: "eqVtRZEcrMMZUMkO7GvrSCyCe//cZuQnIsvLbIJ2iZQ="

ota:
  password: "e1d447bf8bc832253849bbb3867e6e86"

wifi:
  ssid: "Irena_house"
  password: "1212121212"

  ap:
    ssid: "Ble Fallback Hotspot"
    password: "6T8A5xN6qtU3"

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_count:
      name: "${name} cell count"
    total_battery_capacity:
      name: "${name} total battery capacity"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    voltage_calibration:
      name: "${name} voltage calibration"
    current_calibration:
      name: "${name} current calibration"
    power_off_voltage:
      name: "${name} power off voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

time:
  - platform: sntp

sensor:
  - platform: total_daily_energy
    name: "BMS 1 Total Charging Energy Today"
    power_id: ble_charging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001 
  - platform: total_daily_energy
    name: "BMS 1 Total Discharging Energy Today"
    power_id: ble_discharging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001      

  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
      id: ble_charging_power
    discharging_power:
      name: "${name} discharging power"
      id: ble_discharging_power
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```

syssi, 2023-02-19:
The configuration looks good to me. Could you explain what isn't working in detail?

amgstone, 2023-02-19:
http://joxi.ru/MAjegKyTj5VyX2 battery dont work

amgstone, 2023-02-19:
How can I get access to HA so you know?

amgstone, 2023-02-20:
I'm sorry, it's being updated, but the utility meter somehow works strangely, the information is not updated once an hour
[image]

syssi, 2023-02-20:
Please monitor the behavior over a few days.

closed 2023-02-20 by syssi

amgstone, 2023-02-26:
what am i doing wrong? my utility meter does not work.

sensor: 
  - platform: total_daily_energy
    name: "BMS 1 Total Charging Energy Today"
    power_id: ble_charging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001    

  - platform: total_daily_energy
    name: "BMS 1 Total Discharging Energy Today"
    power_id: ble_discharging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001

  - platform: jk_bms_ble
    charging_power:
      id: ble_charging_power
      name: "ble charging power"
      unit_of_measurement: W
      device_class: energy
      state_class: measurement
      entity_category: diagnostic
    discharging_power:
      id: ble_discharging_power
      name: "ble discharging power"
      unit_of_measurement: W
      device_class: energy
      state_class: measurement
      entity_category: diagnostic
----------
charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"

syssi, 2023-02-27:
Could you describe what's not working?

amgstone, 2023-02-27:
utility meter inverter does not transmit values ​​to HA. there is a mistake somewhere.

syssi, 2023-02-27:
Does the ESPHome node report a value greater zero? Could you provide some logs?

amgstone, 2023-02-27:
can you add utility meter in yaml?

syssi, 2023-02-27:
[esp32-ble-example-plus-energy-meter.yaml.txt](https://github.com/syssi/esphome-jk-bms/files/10837202/esp32-ble-example-plus-energy-meter.yaml.txt)

robio339, 2023-03-23:
Hi, any idea why this counters doesn't reset at midnight anymore. It was working fine and the stopped to work. I have two bms. They are connected via ble to two esp32. The other one works fine. Is this home assistant problem? What should i check?
Thank you for answer/help
Robert

amgstone, 2023-03-24:
It works well for me. looked at 12 o'clock resets

syssi, 2023-03-24:
If the ESHome node looses the time source it's unable to reset once a day.

amgstone, 2023-03-24:
flash and the time will be updated

robio339, 2023-03-24:
I thought that i deleted my post fast enough but forgot on email notifications. 
I did flash this esp multiple times since this happened. I also changed name manually (from ${name} total ... to bms total....) and it went to 0, but after midnight again didn't reset.
Yesterday before posting I changed it back to ${name} total .... and after changing entity's in entity card to new ones values stayed the same as with previous entity. I remembered than that the same entity is used in energy dashboard but there the values worked fine all the time. So i deleted entity card, created new one and everything was fine. Values were both on 0 and time on clock was also 00:00 ;). I will see tonight at midnight what will happen.
Thanks for replying and keep on good work.
Best regards
Robert