syssi/esphome-jk-bms #256: xtensa-esp32-elf-g++: fatal error: Killed signal terminated program as compilation terminated.

JohnGalt1717, 2023-02-20:
Compiling .pioenvs/bmses/driver/periph_ctrl.o
xtensa-esp32-elf-g++: fatal error: Killed signal terminated program as
compilation terminated.
*** [.pioenvs/bmses/src/main.o] Error 1

I get this as soon as I add any of the - number: components. Doesn't matter which.

I've done a clean.

Everything else works perfectly.

syssi, 2023-02-20:
Interesting. Which ESPHome version do you use and what's the CPU architecture of the build host?

JohnGalt1717, 2023-02-20:
Here's the board/cpu:

```
esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf
    version: latest

```

And it's using the latest esphome in docker.

syssi, 2023-02-20:
I asked about the processor architecture of your docker host/machine. ;-)

JohnGalt1717, 2023-02-20:
Ah. Sorry. x64

syssi, 2023-02-20:
What's the memory limit of the docker container and your host? 

See https://github.com/esphome/issues/issues/3867

JohnGalt1717, 2023-02-20:
1G for container and 64G for the host.

syssi, 2023-02-20:
Please increase the memory limit of the container. I assume the out of memory killer stops the compiler.

JohnGalt1717, 2023-02-20:
That worked! Thanks!

closed 2023-02-20 by JohnGalt1717