syssi/esphome-jk-bms #264: esphome complie error with exampe

riker65, 2023-02-24:
Hi

just diving into this.

did run
after doing `git clone https://github.com/syssi/esphome-jk-bms.git`

running on `Version: 2022.12.4`
`esphome run esp32-example.yaml`

got error : 

`fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/riker/.platformio/packages/framework-espidf/tools/cmake/build.cmake:454 (message):

  C:\Users\riker\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\python.exe:
  Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')
`

Unfortunaltey I am quite new to esphome  (normally using tasmota) 
I do not see the issue in the log 

any advice maybe?

Thanks for supporting.

BR Thomas

syssi, 2023-02-24:
This issues looks like this: https://github.com/esphome/issues/issues/4084 

Could you try the workaround/fix from the last comment?

riker65, 2023-02-24:
will check thanks

riker65, 2023-02-24:
it worked thanks

closed 2023-02-24 by riker65