syssi/esphome-jk-bms #269: bluetooth connection attempt loops

Septain21, 2023-03-05:
Hello

my esp32 is looping on bluetooth connection attempts with my JKBMS. Once out of 100 he connects to it then it's gone again...

What to do ?

Thx

syssi, 2023-03-05:
What's the distance between the ESP and the BMS?

Septain21, 2023-03-05:
2 meters

syssi, 2023-03-05:
Please check the dmesg output of the host. It looks like the out of memory killer did stop the compiler.

Septain21, 2023-03-05:
I don't understand how to do.
I'm on an RPI 4 with 4Gb of memory. Do I have to stop integrations in order to leave power for the process to be finalized?

Septain21, 2023-03-05:
Or Do I check sudo dmesg on my rpi ?

syssi, 2023-03-06:
The error at your terminal isn't very verbose. It just says "The compiler was killed by something". If it's the out of memory killer of your Linux kernel you should find something about it in the output of `sudo dmesg`. Is you ESPHome build setup dockerized? Is the ESPHome container probably limited to 1GB memory? The build process requires around 2GB.

Septain21, 2023-03-06:
Hello

Can I show sudo demsg log here?
I am in supervised version with HA

syssi, 2023-03-06:
Yes. Feel free to post some logs. This will help a lot! :-)

Septain21, 2023-03-06:
OK this is very long 

`core-ssh ~]$ dmesg
[ 2093.926146] [   9719]     0  9719       51        1    28672        4           200 s6-supervise
[ 2093.926154] [   9720]     0  9720       51        1    28672        4           200 s6-supervise
[ 2093.926161] [   9721]     0  9721       51        1    28672        3           200 s6-supervise
[ 2093.926169] [   9735]     0  9735       45        0    24576        3           200 s6-ipcserverd
[ 2093.926177] [  10210]     0 10210    84224     6255   147456      925           200 esphome
[ 2093.926184] [  10295]     0 10295     2030       21    45056      167           200 nginx
[ 2093.926192] [  10360]     0 10360    81125       66   307200     5516           200 npm start --set
[ 2093.926201] [  10365]     0 10365     4460        6    49152      309           200 nginx
[ 2093.926209] [  10417]     0 10417     2146      114    49152      222           200 nginx
[ 2093.926216] [  10502]     0 10502   102969    19842   512000     4793           200 node-red
[ 2093.926224] [  11383]     0 11383     4518        0    45056      367           200 nginx
[ 2093.926232] [  11894]     0 11894     2504       35    49152      266           200 nginx
[ 2093.926240] [  12631]     0 12631    16054     5073   163840     4061           200 appdaemon
[ 2093.926249] [  18190]     0 18190     7927     4186    86016      449           200 esphome
[ 2093.926257] [  18448]     0 18448    48070     7543   131072        0           200 esphome
[ 2093.926264] [  18463]     0 18463    14354    10721   143360        0           200 python3
[ 2093.926272] [  18504]     0 18504      415        1    32768        0          -300 sleep
[ 2093.926280] [  18632]     0 18632      411        1    36864        0           200 sleep
[ 2093.926288] [  19078]     0 19078      520       23    32768        0           200 sh
[ 2093.926296] [  19083]     0 19083     1450      124    36864        0           200 xtensa-esp32-el
[ 2093.926303] [  19109]     0 19109    97236    96405   802816        0           200 as
[ 2093.926310] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=docker-5f1eab52a999cbd5399bbcad80a04a7af35a44f7219a2fae6f2def4f2451f7ce.scope,mems_allowed=0,global_oom,task_memcg=/system.slice/docker-edd7c8a36bacf70407b0d20e476355a5838324423d1a1980d710ca706bcbe5fa.scope,task=as,pid=19109,uid=0
[ 2093.926376] Out of memory: Killed process 19109 (as) total-vm:388944kB, anon-rss:385320kB, file-rss:300kB, shmem-rss:0kB, UID:0 pgtables:784kB oom_score_adj:200
[ 4178.480208] python3 invoked oom-killer: gfp_mask=0x1100cca(GFP_HIGHUSER_MOVABLE), order=0, oom_score_adj=200
[ 4178.480248] CPU: 0 PID: 7637 Comm: python3 Tainted: G         C        5.15.84-v8 #1
[ 4178.480256] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[ 4178.480260] Call trace:
[ 4178.480262]  dump_backtrace+0x0/0x1b0
[ 4178.480280]  show_stack+0x20/0x30
[ 4178.480287]  dump_stack_lvl+0x8c/0xb8
[ 4178.480295]  dump_stack+0x18/0x34
[ 4178.480299]  dump_header+0x50/0x20c
[ 4178.480306]  oom_kill_process+0x210/0x220
[ 4178.480314]  out_of_memory+0xec/0x330
[ 4178.480319]  __alloc_pages_slowpath.constprop.0+0x814/0xb90
[ 4178.480329]  __alloc_pages+0x29c/0x330
[ 4178.480336]  pagecache_get_page+0x1cc/0x440
[ 4178.480343]  filemap_fault+0x52c/0x940
[ 4178.480349]  __do_fault+0x44/0x170
[ 4178.480355]  __handle_mm_fault+0x53c/0xb20
[ 4178.480361]  handle_mm_fault+0x1b4/0x290
[ 4178.480366]  do_page_fault+0x170/0x470
[ 4178.480373]  do_translation_fault+0x60/0x80
[ 4178.480379]  do_mem_abort+0x48/0xc0
[ 4178.480386]  el0_ia+0x70/0x100
[ 4178.480392]  el0t_64_sync_handler+0x98/0xc0
[ 4178.480398]  el0t_64_sync+0x1a0/0x1a4
[ 4178.480404] Mem-Info:
[ 4178.480410] active_anon:120744 inactive_anon:245023 isolated_anon:0
[ 4178.480410]  active_file:1896 inactive_file:1858 isolated_file:38
[ 4178.480410]  unevictable:0 dirty:0 writeback:0
[ 4178.480410]  slab_reclaimable:17802 slab_unreclaimable:16846
[ 4178.480410]  mapped:3281 shmem:20 pagetables:5929 bounce:0
[ 4178.480410]  kernel_misc_reclaimable:0
[ 4178.480410]  free:8928 free_pcp:919 free_cma:79
[ 4178.480423] Node 0 active_anon:482976kB inactive_anon:980092kB active_file:7584kB inactive_file:7532kB unevictable:0kB isolated(anon):0kB isolated(file):160kB mapped:13124kB dirty:0kB writeback:0kB shmem:80kB writeback_tmp:0kB kernel_stack:17360kB pagetables:23716kB all_unreclaimable? no
[ 4178.480434] DMA free:19340kB min:24876kB low:26788kB high:28700kB reserved_highatomic:0KB active_anon:230380kB inactive_anon:494084kB active_file:1440kB inactive_file:2560kB unevictable:0kB writepending:0kB present:969728kB managed:883980kB mlocked:0kB bounce:0kB free_pcp:840kB local_pcp:88kB free_cma:316kB
[ 4178.480445] lowmem_reserve[]: 0 983 983 983
[ 4178.480467] DMA32 free:16372kB min:28348kB low:30528kB high:32708kB reserved_highatomic:0KB active_anon:252596kB inactive_anon:485588kB active_file:5944kB inactive_file:5232kB unevictable:0kB writepending:0kB present:1048576kB managed:1007212kB mlocked:0kB bounce:0kB free_pcp:2836kB local_pcp:716kB free_cma:0kB
[ 4178.480478] lowmem_reserve[]: 0 0 0 0
[ 4178.480497] DMA: 1226*4kB (UMEC) 537*8kB (UEC) 280*16kB (UE) 133*32kB (UE) 35*64kB (UE) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 20176kB
[ 4178.480564] DMA32: 1433*4kB (UME) 96*8kB (UME) 260*16kB (UME) 133*32kB (UME) 38*64kB (UE) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 17348kB
[ 4178.480630] 4783 total pagecache pages
[ 4178.480634] 1005 pages in swap cache
[... cut ...]
[15330.266882] audit: type=1334 audit(1678048402.172:716): prog-id=181 op=LOAD
[15360.397717] audit: type=1334 audit(1678048432.303:717): prog-id=0 op=UNLOAD
[15360.397764] audit: type=1334 audit(1678048432.303:718): prog-id=0 op=UNLOAD
[15360.397796] audit: type=1334 audit(1678048432.303:719): prog-id=0 op=UNLOAD
[15360.983079] audit: type=1334 audit(1678048432.887:720): prog-id=0 op=UNLOAD
[15360.983137] audit: type=1334 audit(1678048432.887:721): prog-id=0 op=UNLOAD
[15360.983171] audit: type=1334 audit(1678048432.887:722): prog-id=0 op=UNLOAD
[core-ssh ~]$ 
`

syssi, 2023-03-07:
These lines confirm my assumption:

```
[14017.113342] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=docker-edd7c8a36bacf70407b0d20e476355a5838324423d1a1980d710ca706bcbe5fa.scope,mems_allowed=0,global_oom,task_memcg=/system.slice/docker-edd7c8a36bacf70407b0d20e476355a5838324423d1a1980d710ca706bcbe5fa.scope,task=cc1plus,pid=39459,uid=0
[14017.113406] Out of memory: Killed process 39459 (cc1plus) total-vm:369848kB, anon-rss:342560kB, file-rss:516kB, shmem-rss:0kB, UID:0 pgtables:744kB oom_score_adj:200
```

The `cc1plus` process was killed by the out of memory killer.

Septain21, 2023-03-07:
oops what to do?

syssi, 2023-03-07:
Yes. The BMS supports just **one** simultaneous BLE connection. If the Android App is running the ESP32 won't connect anymore and vice versa.

austin202220, 2023-03-28:
Clean the build files and also update your config: change the framework version to 'recommended' instead of latest

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: recommended

closed 2023-04-16 by syssi