syssi/esphome-jk-bms #273: When uart cable is disconnected previous values are kept and not set to null

jrventer, 2023-03-12:
When the uart cable to the BMS is disconnected for whatever reason the previous values are kept. This can lead to issue not discovered for a long time until it is too late.

 I suggest either setting the values to null when there is no response from the BMS or have a last updated variable that you can track to see when last the value was updated. Both options might also be option.

syssi, 2023-03-12:
Good point! I will implement the first suggestion. The second is already available if you use Home Assistant or another Home Automation which keeps track of the latest update per sensor.

syssi, 2023-03-20:
There is a new binary sensor called `online_status` now:

```
binary_sensor:
  - platform: jk_bms
    online_status:
      name: "${name} online status"
```

If the BMS doesn't respond (on 5 consecutive updates/requests) all sensor values are changed to `Unknown` (NaN), the `online_status` resports `Disconnected` (false) and the text sensor `errors` will show `Offline`.

closed 2023-03-20 by syssi

jrventer, 2023-03-22:
Great thanks I will recompile and test this.

syssi, 2023-03-31:
I'm implementing a pretty similar setup here: https://github.com/syssi/esphome-virtual-can-bms

Could you share some details about your inverter (model) and the canbus id / registers you are pushing to the inverter via canbus?

jrventer, 2023-06-07:
> I'm implementing a pretty similar setup here: https://github.com/syssi/esphome-virtual-can-bms
> 
> Could you share some details about your inverter (model) and the canbus id / registers you are pushing to the inverter via canbus?

Hi sorry missed this message. The 1st integration I am testing is with a Voltronix Axpert King II using the Soltaro CAN Protocol. I will soon start also to test with a Sunsynk/Deye Inverter using the Pylontech CAN protocol.

Below is the can bus specific config I am busy using - reworked some of the config from [https://github.com/Uksa007/esphome-jk-bms-can](url) to work with Soltaro protocol supported by the Axpert inverter.

`spi:
  id: McpSpi
  clk_pin: ${spican_clk_pin}
  mosi_pin: ${spican_mosi_pin}
  miso_pin: ${spican_miso_pin}

canbus:
  - platform: mcp2515
    spi_id: McpSpi
    cs_pin: ${spican_cs_pin}
    #mode: LOOPBACK
    #clock: 8MHZ
    can_id: 4
    bit_rate: 500kbps
    on_frame:
    - can_id: 0x305
      then:
      - lambda: |-
          std::string b(x.begin(), x.end());
          ESP_LOGI("main", "received can id: 0x305 ascii: %s", &b[0] );
          id(can_305_rx) = 0;
interval:
  - interval: 1000ms
    then:
      - if:
          condition:
            - binary_sensor.is_on: online_status
          then:
          - canbus.send: # BMS instructs inverter: Charge Volts, Charge Amps, Discharge Amps, Min voltage
              use_extended_id: True
              can_id: 0x1872
              data: !lambda |-
                uint8_t can_mesg[7];
                if (((!id(charging).state) | (!id(inverter_charging).state)) & (id(charge_status) != "Disabled")) {
                  id(charge_status) = "Disabled";
                  id(can_disable_charge_script).execute();
                } else if (id(inverter_chg_on).state) {
                    id(charge_status) = "Bulk Manually";
                } else if ((id(charging).state) & (id(inverter_charging).state) & (id(charge_status) == "Disabled")) {
                  id(charge_status) = "Wait";
                  id(can_enable_charge_script).execute();
                } else if ((!id(inverter_chg_on).state) & (id(charge_status) == "Bulk Manually")) {
                  id(charge_status) = "Wait";
                } else if (id(total_voltage).state <= (id(charging_voltage).state - ${rebulk_offset})) { ///Bulk Charge eg 53.6v 10% 
                    if (id(absorption_script).is_running()) id(absorption_script).stop();
                    id(charge_status) = "Bulk";
                } else if ((id(total_voltage).state > (id(charging_voltage).state - ${rebulk_offset})) & (id(total_voltage).state < (id(charging_voltage).state - 0.05))) { ///If in startup rebulk
                        if (id(charge_status) == "Startup") {
                            id(charge_status) = "Bulk"; /// If in startup, 10% low rebulk
                        }
                } else if (id(total_voltage).state >= (id(charging_voltage).state  - 0.05)) { ///10 % from top start absorption timer
                        if (id(charge_status) == "Bulk") {
                            id(charge_status) = "Absorption";
                            if (!id(absorption_script).is_running()) id(absorption_script).execute();
                        }
                } else {
                    id(charge_status) = "Wait";
                }
                if ((id(charge_status) == "Bulk") | (id(charge_status) == "Absorption") | (id(inverter_chg_on).state)) {
                  can_mesg[0] = uint16_t(id(charging_voltage).state * 10) & 0xff;
                  can_mesg[1] = uint16_t(id(charging_voltage).state * 10) >> 8 & 0xff;
                  can_mesg[4] = uint16_t(id(charging_current).state * 10) & 0xff;
[... 100 lines cut ...]
            can_mesg[2] = 0x00;
            can_mesg[3] = 0x00; 
            can_mesg[4] = 0x00;
            can_mesg[5] = 0x00;
            can_mesg[6] = 0x00;
            can_mesg[7] = 0x00;
            ESP_LOGI("main", "end can id: 0x1876 hex: %x %x %x %x %x %x %x %x", can_mesg[0], can_mesg[1], can_mesg[2], can_mesg[3], can_mesg[4], can_mesg[5], can_mesg[6], can_mesg[7]);
            return {can_mesg[0], can_mesg[1], can_mesg[2], can_mesg[3], can_mesg[4], can_mesg[5], can_mesg[6], can_mesg[7]};
      - delay: 10ms
      - canbus.send: # Heartbeat 1
          use_extended_id: True
          can_id: 0x187b
          data: !lambda |-
            uint8_t can_mesg[7];
            can_mesg[0] = 0x00;
            can_mesg[1] = 0x00;
            can_mesg[2] = 0x00;
            can_mesg[3] = 0x00; 
            can_mesg[4] = 0x00;
            can_mesg[5] = 0x00;
            can_mesg[6] = 0x00;
            can_mesg[7] = 0x00;
            ESP_LOGI("main", "end can id: 0x187b hex: %x %x %x %x %x %x %x %x", can_mesg[0], can_mesg[1], can_mesg[2], can_mesg[3], can_mesg[4], can_mesg[5], can_mesg[6], can_mesg[7]);
            return {can_mesg[0], can_mesg[1], can_mesg[2], can_mesg[3], can_mesg[4], can_mesg[5], can_mesg[6], can_mesg[7]};
      - delay: 10ms`

commit c8c0af40 (2023-03-20): Track online status of the BMS (Closes: #273) (#293)