syssi/esphome-jk-bms #274: JK_B2A24S20P show some wrong values

ayasystems, 2023-03-12:


ayasystems, 2023-03-12:
Thanks for your work 

[image]

`substitutions:
  name: bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E9:2F:E7
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: wemos_d1_mini32
  framework:
#  The arduino platform crashs
#    type: arduino
#    version: latest
    type: esp-idf
    version: latest

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: wifi
  password: wifi

ota:
  password: "92629564668e34fc31334bb6e16ad854"
logger:
  level: NONE

uart:
  id: uart0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: D1
  rx_pin: D3
  debug:
    direction: BOTH
 

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
#api:

# The MQTT component is ESP-IDF compatible since ESPHome version 2022.4.0. If
# ESPHome suggests to use the arduino framework instead because of missing ESP-IDF
# framework support you should update your setup.
mqtt:
  broker: mqtt.local
 
  id: mqtt_jk_bms

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
[... 174 lines cut ...]
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
    
    
    `

I think that some values are not correct:
 
[image]

syssi, 2023-03-13:
Could you increase the log level and provide some logs? I'm interested in the raw cell info frame of your device version:

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: VERY_VERBOSE
    jk_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

Thanks in advance!

ayasystems, 2023-03-13:
Yes, thanks for your quick response

OMG, I saw that i have enable fake traffic

[08:41:39][C][jk_bms_ble:054]:   Fake traffic enabled: YES

I will test again withuot it and check again ..... Done.... Now all looks fine, i'm sorry

Here the logs if you want to check anything 

[logs_jk-ea4gkq_upload (1).txt](https://github.com/syssi/esphome-jk-bms/files/10954835/logs_jk-ea4gkq_upload.1.txt)

syssi, 2023-03-13:
Looks good to me! No action needed. ;-) 73 de DO2KSM

closed 2023-03-13 by syssi