syssi/esphome-jk-bms #275: Some values are 0 / wrong after many months of perfect running. (JKB2A8S20P)

bassmaster187, 2023-03-13:
For any reason some values are 0 after 6th March. 
- I didn't change anything
- I tried to turn off / on the ESP32
- I tried to turn off / on the JK-BMS

Still working:
- cell voltage 1-8
- min cell voltage
- max cell coltage

Only 0 Values:
- delta cell voltage
- average cell voltage
- total voltage
- current
- power
- charging_power
- discharging_power
- temperature sensor 1 & 2
- power tube temperature
- state of charge

# Wrong Values:
- charging cycles. It's actually like 26742 and goes up and down. The Battery is like 8 Months old so maybe this is causing the issue?
[image]

within seconds it reports 50 additional charging cycles
[image]

- charging cycle capacity goes up and down:
[image]

# Debug
shows only Wire resistance warning & Enable cells, but i turned off verbose logging

Here is my yaml file:
[JKBMS - Kopie.txt](https://github.com/syssi/esphome-jk-bms/files/10956010/JKBMS.-.Kopie.txt)

App seams to work fine:
[image]

[image]

bassmaster187, 2023-03-13:
I hope that helps:
[logfile.zip](https://github.com/syssi/esphome-jk-bms/files/10957433/logfile.zip)

syssi, 2023-03-13:
Test frame:

```
0x55, 0xAA, 0xEB, 0x90, 0x02, 0xBA, 0x48, 0x0D, 0x4F, 0x0D, 0x4F, 0x0D, 0x53, 0x0D, 0x54, 0x0D, 0x46, 0x0D,
0x4B, 0x0D, 0x4A, 0x0D, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0x00,
0x00, 0x00, 0x4D, 0x0D, 0x0E, 0x00, 0x04, 0x05, 0x34, 0x00, 0x34, 0x00, 0x34, 0x00, 0x34, 0x00, 0x33, 0x00,
0x33, 0x00, 0x32, 0x00, 0x34, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x63, 0x01, 0x00, 0x00, 0x00, 0x00, 0x69, 0x6A, 0x00, 0x00, 0x47, 0x8D, 0x21, 0x00, 0x4F, 0x3B, 0x01, 0x00,
0x18, 0x01, 0x28, 0x01, 0x00, 0x00, 0x00, 0x00, 0x6B, 0xF8, 0x02, 0x4B, 0x19, 0xA1, 0x05, 0x00, 0x70, 0x6A,
0x07, 0x00, 0x25, 0x00, 0x00, 0x00, 0x69, 0xFF, 0x14, 0x01, 0x64, 0x00, 0x00, 0x00, 0x8F, 0x9D, 0xBA, 0x00,
0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00,
0x01, 0x00, 0x00, 0x00, 0x8D, 0x03, 0xBC, 0x01, 0x00, 0x00, 0xE9, 0x10, 0x40, 0x40, 0xAC, 0x00, 0x00, 0x00,
0xA4, 0x0A, 0x0E, 0xE2, 0x00, 0x01, 0x00, 0x01, 0x00, 0x05, 0x00, 0x00, 0x9A, 0x24, 0x36, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFE,
0xFF, 0x7F, 0xDC, 0x3F, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0xF6
```

syssi, 2023-03-13:
Could you tell me the hardware and software version of your BMS?

bassmaster187, 2023-03-13:
[image]

> If I force the implementation to use protocol version 3 again the decoded data looks good again:

Is this something I can do as well or do I have to wait for an update?

syssi, 2023-03-13:
I've prepared a workaround in a feature. Please update your YAML configuration to:

```
substitutions:
  external_components_source: github://syssi/esphome-jk-bms@force-frame-version-3
```

and update your ESP32. To fix this issue / auto-detect the frame version properly I need some more time. I will ask you about some more `cell info frame` examples probably. Let's see! :-)

bassmaster187, 2023-03-13:
It seems like it works. Thank you very much. Please let me know how I can help to solve this problem.

syssi, 2023-03-14:
Please update your configuration YAML to:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  protocol_version: JK02_32S
```

bassmaster187, 2023-03-14:
Works perfekt

closed 2023-03-14 by bassmaster187

syssi, 2023-03-14:
@bassmaster187 Do you know the meaning of "Heat current" at your screenshot above? Do you make use of the heating function?

bassmaster187, 2023-03-15:
I think it's for a heat pad used for outdoor batteries. Below 5°C you should preheat your cells to avoid degradation. But I'm not sure.

https://diysolarforum.com/threads/jk-bms-heating-port.44994/

syssi, 2023-03-15:
Awesome! I would like to add the solution to the "Goodies section".