syssi/esphome-jk-bms #276: mqtt command failed for number

riker65, 2023-03-13:
Hi
using jk-bms B2A8S20P

hardwware v11.XW
Software V11.26

Tried to change overvoltage recovery via MQTT

jk-bms/number/jk-bms_cell_voltage_overvoltage_recovery/command -m 3.9

looks like new number was recognized but not stored?
see log

[14:45:41][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[14:45:41][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[14:45:41][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.C6 (22)
**[14:45:41][D][number:054]: 'jk-bms cell voltage overvoltage recovery' - Setting number value**
**[14:45:41][D][number:113]:   New number value: 3.900000**
[14:45:41][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.05.04.3C.0F.00.00.00.00.00.00.00.00.00.00.00.CE (20)
[14:45:41][D][number:012]: 'jk-bms cell voltage overvoltage recovery': Sending state 4.180000
[14:45:41][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[14:45:41][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=4 gattc_if=3
[14:45:41][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10

syssi, 2023-03-13:
Could you provide a ESPHome log? I'm interested in a `cell info frame` and a `settings frame`. The retrieval of the settings frame must be triggered using the `retrieve_settings` button or gets retrieved once on every BLE re-connect.

riker65, 2023-03-13:
Hi

[14:45:43][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[14:45:43][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[14:45:43][VV][jk_bms_ble:220]: Notification received: E9.3B.00.00.00.00.00.00.00.00.00.00.C3.00.C2.00.00.00.08.00.00.00.00.00.00.00.00.00.C8.32.00.00.00.00.00.00.7E.04.00.00.64.00.00.00.87.79.02.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.AD.03.00.00.00.00.58.72.3F.40.00.00.00.00.FD.05.ED.1B.00.01.00.01.00.05.00.00.2F.50.18.00.00.00.00.00.EA.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[14:45:43][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[14:45:43][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[14:45:43][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.E5 (22)
[14:45:43][VV][jk_bms_ble:450]: Cell info frame (version 3, 300 bytes):
[14:45:43][VV][jk_bms_ble:451]:   55.AA.EB.90.02.77.FE.0E.FD.0E.06.0F.FA.0E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0F.00.00.00.FA.0E.0C.00.02.01.91.00.7D.00.A9.00.70.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.EA.00.00.00.00.00 (150)
[14:45:43][VV][jk_bms_ble:452]:   E9.3B.00.00.00.00.00.00.00.00.00.00.C3.00.C2.00.00.00.08.00.00.00.00.00.00.00.00.00.C8.32.00.00.00.00.00.00.7E.04.00.00.64.00.00.00.87.79.02.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.AD.03.00.00.00.00.58.72.3F.40.00.00.00.00.FD.05.ED.1B.00.01.00.01.00.05.00.00.2F.50.18.00.00.00.00.00.EA.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.E5 (150)
[14:45:43][D][sensor:127]: 'jk-bms cell voltage 1': Sending state 3.83800 V with 3 decimals of accuracy

syssi, 2023-03-13:
Alright. It's the newest generation (`frame version 3`) of the BMS. It looks like the manufacturer has changed something (registers or instruction set) so you cannot and shouldn't control the BMS using this implementation at the moment.

Do you own an Android device? I would be happy about a capture of the BLE traffic between the Android App and your BMS. I would provide some guidance how to record such a capture.

syssi, 2023-03-13:
Let's ignore the settings frame for now.

Could you try to follow this tutorial: https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa

This is a short summary to get an idea about the progress:

1. Enable the developer mode of your android phone
2. Enable Bluetooth HCI snoop log at the developer settings
3. Disable bluetooth, enable bluetooth
4. Open the JK-BMS app go the settings page and change the overvoltage recovery voltage
5. Close the app
6. Use `adb` to create and download a bug report. This is a big ZIP archive which should include the btsnoop capture (if we are lucky and your android version supports this feature)

The btsnoop_hci.log can be opened in wireshark. I would be happy about a btsnoop_hci.log, so I can extract and implement the command support for your BMS version.

riker65, 2023-03-13:
found some working commands for overvoltage recovery

`[15:41:46][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[15:41:46][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[15:41:46][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.61 (22)
[15:41:46][D][number:054]: 'jk-bms cell voltage overvoltage recovery' - Setting number value
[15:41:46][D][number:113]:   New number value: 3.900000
[15:41:46][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.05.04.3C.0F.00.00.00.00.00.00.00.00.00.00.00.CE (20)
[15:41:46][D][number:012]: 'jk-bms cell voltage overvoltage recovery': Sending state 3.900000
[15:41:46][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[15:41:46][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=4 gattc_if=3
[15:41:46][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[15:41:46][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[15:41:46][VV][jk_bms_ble:220]: Notification received: AA.55.90.EB.C8.01.01.00.00.00.00.00.00.00.00.00.00.00.00.44 (20)
[15:41:47][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[15:41:47][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[15:41:47][VV][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.C1.FD.0E.FB.0E.01.0F.FB.0E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0F.00.00.00.FD.0E.10.00.02.00.91.00.7D.00.A9.00.70.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[15:41:47][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10`

riker65, 2023-03-13:
similar to setting start balance.
new number value 3.88 but state send old value 4.1

`[16:21:52][VV][jk_bms_ble:220]: Notification received: EA.3B.00.00.00.00.00.00.00.00.00.00.C5.00.C3.00.00.00.08.00.00.00.00.00.00.00.00.00.C8.32.00.00.00.00.00.00.A8.04.00.00.64.00.00.00.0F.90.02.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.AD.03.00.00.00.00.58.72.3F.40.00.00.00.00.FD.05.08.1C.00.01.00.01.00.05.00.00.84.31.19.00.00.00.00.00.EB.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:52][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:52][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.24 (22)
[16:21:52][D][number:054]: 'jk-bms balance starting voltage' - Setting number value
[16:21:52][D][number:113]:   New number value: 3.880000
[16:21:52][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.26.04.28.0F.00.00.00.00.00.00.00.00.00.00.00.DB (20)
[16:21:52][D][number:012]: 'jk-bms balance starting voltage': Sending state 4.100000
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[16:21:52][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=4 gattc_if=3
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:52][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:52][VV][jk_bms_ble:220]: Notification received: AA.55.90.EB.C8.01.02.00.00.00.00.00.00.00.00.00.00.00.00.45 (20)
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:52][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:52][VV][jk_bms_ble:220]: Notification received: 55.AA.EB.90.02.A3.FB.0E.F9.0E.FF.0E.F9.0E.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0F.00.00.00.FB.0E.05.00.02.00.91.00.7D.00.A9.00.70.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:52][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:52][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.EB.00.00.00.00.00 (22)
[16:21:52][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:53][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:53][VV][jk_bms_ble:220]: Notification received: EC.3B.00.00.00.00.00.00.00.00.00.00.C5.00.C3.00.00.00.08.00.00.00.00.00.00.00.00.00.C8.32.00.00.00.00.00.00.A8.04.00.00.64.00.00.00.10.90.02.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.AD.03.00.00.00.00.58.72.3F.40.00.00.00.00.FE.05.E2.1B.00.01.00.01.00.05.00.00.8E.31.19.00.00.00.00.00.EB.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[16:21:53][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[16:21:53][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[16:21:53][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.03 (22)
[16:21:53][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[16:21:53][VV][esp32_ble_tracker:361]: Parse Result:
[16:21:53][VV][esp32_ble_tracker:378]:   Address: 73:F3:FB:3B:BB:A0 (RANDOM)
[16:21:53][VV][esp32_ble_tracker:380]:   RSSI: -95
[16:21:53][VV][esp32_ble_tracker:381]:   Name: ''
[16:21:53][VV][esp32_ble_tracker:383]:   TX Power: 2
[16:21:53][VV][esp32_ble_tracker:389]:   Ad Flag: 26
[16:21:53][VV][esp32_ble_tracker:395]:   Manufacturer data: 10.07.4D.1F.68.BE.E0.BB.08 (9)
[16:21:53][VV][esp32_ble_tracker:411]: Adv data: 02.01.1A.02.0A.0C.0C.FF.4C.00.10.07.4D.1F.68.BE.E0.BB.08 (19)
[16:21:53][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10`

syssi, 2023-03-13:
Good job! I did forget one important piece of the puzzle. :-) Sorry for the inconvenience.

The write instruction (`Write register: AA.55.90.EB.26.04.28.0F.00.00.00.00.00.00.00.00.00.00.00.DB`) updates a single register of the `settings frame`. The BMS configuration entities (the published values via MQTT) get's updated only if the `settings frame` gets received. The default configuration doesn't request the settings frame. You have to press the `retrieve_settings` button (using mosquitto_pub in your case).

Alternatively you could add this `interval` to your YAML:

```
button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button

interval:
  - interval: 1min
    then:
      - button.press: retrieve_settings_button
```

This will press the button periodically but it will also let the BMS beep once every minute.

riker65, 2023-03-13:
do you still need the bug report? have a big one now

riker65, 2023-03-13:
how does the full publish topic look like for 

`You have to press the retrieve_settings button (using mosquitto_pub in your case).`

syssi, 2023-03-13:
This frame format and register does match with the command from the log above:

```
[15:41:46][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.05.04.3C.0F.00.00.00.00.00.00.00.00.00.00.00.CE (20)
                                                            register 0x05, len 0x04, data 0x0F3C = 3900 = 3.9V
```

riker65, 2023-03-13:
as mentioned earlier, working now for overvoltage recovery

tried blanace start voltage

not working I assume

no change after button press and refresh.
Having the automated 1  minute refresh in place.

`17:18:05][VV][jk_bms_ble:220]: Notification received: DC.3B.00.00.00.00.00.00.00.00.00.00.C5.00.C4.00.00.00.08.00.00.00.00.00.00.00.00.00.C8.32.00.00.00.00.00.00.C0.04.00.00.64.00.00.00.3D.9D.02.00.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.FF.00.01.00.00.00.AD.03.00.00.00.00.58.72.3F.40.00.00.00.00.FC.05.FA.1B.00.01.00.01.00.05.00.00.4B.B5.19.00.00.00.00.00.EB.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[17:18:05][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:05][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:05][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.FE.FF.7F.DC.0F.01.00.00.00.00.00.00.2B (22)
[17:18:05][D][number:054]: 'jk-bms balance starting voltage' - Setting number value
[17:18:05][D][number:113]:   New number value: 3.950000
[17:18:05][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.26.04.6E.0F.00.00.00.00.00.00.00.00.00.00.00.21 (20)
[17:18:05][D][number:012]: 'jk-bms balance starting voltage': Sending state 4.100000
[17:18:05][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[17:18:05][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=4 gattc_if=3
[17:18:06][D][button:013]: 'jk-bms retrieve settings' Pressed.
[17:18:06][D][jk_bms_ble:1225]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:06][VV][jk_bms_ble:220]: Notification received: AA.55.90.EB.C8.01.02.00.00.00.00.00.00.00.00.00.00.00.00.45 (20)
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 4
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=4 gattc_if=3
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:06][VV][jk_bms_ble:220]: Notification received: 55.AA.EB.90.01.45.58.02.00.00.24.0E.00.00.74.0E.00.00.68.10.00.00.D8.0E.00.00.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10.0E.00.00.D0.07.00.00.1E.00.00.00.3C.00.00.00.D0.07.00.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.2C.01.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.38.FF.FF.FF.9C.FF.FF.FF.E8.03.00.00.20.03.00.00.04.00.00.00.01.00.00.00.00.00.00.00.01.00 (128)
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:06][VV][jk_bms_ble:220]: Notification received: 00.00.C8.32.00.00.DC.05.00.00.04.10.00.00.00.00.00.00.00.00.00.00 (22)
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:06][VV][jk_bms_ble:220]: Notification received: 00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (128)
[17:18:06][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 3] - 10
[17:18:06][V][esp32_ble_client:111]: [0] [C8:47:8C:EC:77:7D] gattc_event_handler: event=10 gattc_if=3
[17:18:06][VV][jk_bms_ble:220]: Notification received: B0.71.0B.00.08.00.00.00.00.7C.F8.FF.FF.1F.0D.00.00.00.00.00.00.01 (22)
[17:18:06][I][jk_bms_ble:873]: Settings frame (300 bytes):
[17:18:06][VV][jk_bms_ble:874]:   55.AA.EB.90.01.45.58.02.00.00.24.0E.00.00.74.0E.00.00.68.10.00.00.D8.0E.00.00.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10.0E.00.00.D0.07.00.00.1E.00.00.00.3C.00.00.00.D0.07.00.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.2C.01.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.38.FF.FF.FF.9C.FF.FF.FF.E8.03.00.00.20.03.00.00.04.00.00.00.01.00.00.00.00.00.00.00.01.00.00.00.C8.32.00.00.DC.05.00.00.04.10.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 
[17:18:06][VV][jk_bms_ble:875]:   00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.71.0B.00.08.00.00.00.00.7C.F8.FF.FF.1F.0D.00.00.00.00.00.00.01 (140)
[17:18:06][D][jk_bms_ble:899]:   Unknown6: 0.600000
[17:18:06][I][jk_bms_ble:901]:   Cell UVP: 3.620000 V`

syssi, 2023-03-13:
Could you provide a full settings frame (including the decoded values):

```
[17:18:06][I][jk_bms_ble:873]: Settings frame (300 bytes):
[17:18:06][VV][jk_bms_ble:874]:   55.AA.EB.90.01.45.58.02.00.00.24.0E.00.00.74.0E.00.00.68.10.00.00.D8.0E.00.00.0A.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10.0E.00.00.D0.07.00.00.1E.00.00.00.3C.00.00.00.D0.07.00.00.2C.01.00.00.3C.00.00.00.3C.00.00.00.2C.01.00.00.BC.02.00.00.58.02.00.00.BC.02.00.00.58.02.00.00.38.FF.FF.FF.9C.FF.FF.FF.E8.03.00.00.20.03.00.00.04.00.00.00.01.00.00.00.00.00.00.00.01.00.00.00.C8.32.00.00.DC.05.00.00.04.10.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 
[17:18:06][VV][jk_bms_ble:875]:   00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B0.71.0B.00.08.00.00.00.00.7C.F8.FF.FF.1F.0D.00.00.00.00.00.00.01 (140)
[17:18:06][D][jk_bms_ble:899]:   Unknown6: 0.600000
[17:18:06][I][jk_bms_ble:901]:   Cell UVP: 3.620000 V
...
the following lines here would be nice
....
```

syssi, 2023-03-13:
To get an idea why updating the `Start balance voltage` doesn't work another btsnoop capture would be nice. These is the payload the BMS doesn't want to accept:

```
# 3.88V
AA.55.90.EB.26.04.28.0F.00.00.00.00.00.00.00.00.00.00.00.DB

# 3.95V
AA.55.90.EB.26.04.6E.0F.00.00.00.00.00.00.00.00.00.00.00.21
```

riker65, 2023-03-13:
ok let me check

syssi, 2023-03-13:
```
aa5590eb220404100000bc5ba14a8f60e08f4559
        |  |^^^^^^^^0x1004 = 4100 = 4.1V
        |  |
        |  data_len 4
        address 0x22
```

It looks like 0x22 is the "start balance voltage" register of your BMS. Did you apply 4.1V here?

riker65, 2023-03-13:
I think 4.1 was the setting before I changed it, have to check

riker65, 2023-03-13:
I am unsure, it used to be 4.1 I thought I changed it , maybe I reverted it. 
Can do another trace.

did you find another change of Balance start voltage change in the file?

syssi, 2023-03-13:
This is the single command at the second capture. Do you like to provide another one?

syssi, 2023-03-13:
```
aa5590eb2204280f0000c0bbb38a723196e73eed
            ^^^^ 0x0f28 = 3880 = 3.88V

aa5590eb2204a00f00009ec074687be46b634d03
            ^^^^ 0x0fa0 = 4000 = 4.00V
```

syssi, 2023-03-13:
So we know for sure register 0x22 is the start balancing register in your case. I will verify it on my side asap. May be it has always been wrong.

syssi, 2023-03-14:
Start balance voltage of a JK-BD6A17S6P: `aa5590eb26049c090000d327863be2124db0b6ab`

The registers has changed. :-(

riker65, 2023-03-14:
ok thanks for checking

riker65, 2023-03-14:
Hi, can try, 
which settings can be adjusted and need to be checked?

all which are available in the app I assume.

can try which are working and which not and capture those.
Any docu available for tracing changes?

will take some time, have to run now.

riker65, 2023-03-14:
did send a file with controls, maybe it can help

syssi, 2023-03-14:
```
Command: aa5590eb1d0400000000d2a8888b2a3cd4360098 # Charge OFF
Command: aa5590eb1d04010000000cc404c33747565e3da2 # Charge ON

Command: aa5590eb1e0401000000808685a3e9760be7c4e0 # Discharge ON
Command: aa5590eb1e0400000000b2430b4e95b4d6215e88 # Discharge OFF

Command: aa5590eb1f040000000093f6149382d818c59094 # Balance OFF
Command: aa5590eb1f0401000000de9bee1c9bf4a1be5e6d # Balance ON

Command: aa5590eb6b040100000019ca465d7d09e74c9ec7 # Emergency ON
Command: aa5590eb6b04000000001ca240f936b60fca39de # Emergency OFF

Command: aa5590eb280401000000ab3f0ae522bb935b024d # Disable temp. sensor ON
Command: aa5590eb28040000000026441e01ce8ea4ac2a05 # Disable temp. sensor OFF

Command: aa5590eb2a0400000000373787c259dc922cbc0e # Port Switch CAN
Command: aa5590eb2a0401000000f4c37ea5ee3d2fd45f10 # Port Switch RS485?

Command: aa5590eb2b0401000000ebe73aed0f7f8b10cd99 # Display Always On ON
Command: aa5590eb2b040000000099347bc4ea32cb2e8b55 # Display Always On OFF
```

[image]

riker65, 2023-03-14:
[jk-bms-11-RS485 protocol.pdf](https://github.com/syssi/esphome-jk-bms/files/10968243/jk-bms-11-RS485.protocol.pdf)
Hi
does this help?

RS485 protocol reference? you probably have this already

syssi, 2023-03-14:
This document describes the UART-TTL protocol. It's implemented already and it's different to the BLE communication.

syssi, 2023-03-14:
```
Command: aa5590eb 04 04 68100000 05aa400f6aa466909d99 # OVP 4200
Command: aa5590eb 05 04 a00f0000 14876e9f40e22802aad0 # OVPR 4000
Command: aa5590eb 03 04 100e0000 0eb48b0f6edf31c4a6e3 # UVPR 3600
Command: aa5590eb 02 04 ac0d0000 1da58b588e4fd90db95a # UVP 3500
Command: aa5590eb 0b 04 480d0000 c0d1aaa4906c3a33f016 # Power Off Vol. 3400
Command: aa5590eb 22 04 04100000 abd4dc520aa300253a6d # Start Balance Volt. 4100
```

syssi, 2023-03-14:
```
Command: aa5590eb 13 04 2c010000 e577a998fab6c3b535b8 Max Balance Current 300 (0.3 A)
Command: aa5590eb 0c 04 b80b0000 4582fe93a41562eade88 Continued Charge Current 3000
Command: aa5590eb 0d 04 28000000 d3e1d60edce047a8140a Charge OCP Delay 40
Command: aa5590eb 0e 04 46000000 ed585e91333a0c8a1821 Charge OCPR Time 70
Command: aa5590eb 0f 04 e8030000 7b2a25b144905226c807 Continued Discharge Current 1000
Command: aa5590eb 10 04 2c010000 e8a617872fc4cbc6f964 Discharge OCP Delay 300
Command: aa5590eb 10 04 c8000000 be4c472710566be5e86c Discharge OCP Delay 200
Command: aa5590eb 11 04 3c000000 c4fb0290996c038c9c4c Discharge OCPR Time 60
Command: aa5590eb 11 04 46000000 b19a992622f78bcfae00 Discharge OCPR Time 70
Command: aa5590eb 21 04 d0070000 884ca9d793b551a5747c SCP Delay 2000
Command: aa5590eb 12 04 46000000 9431c6e7fdd89f62bfdd SCPR Time 70

Command: aa5590eb 14 04 94020000 67b6b6012690b4d0b5eb Charge OTP 660
Command: aa5590eb 15 04 62020000 283f85d2c8517da0644f Charge OTPR 610
Command: aa5590eb 16 04 02030000 1fc8321a48ca575626b1 Discharge OTP 770
Command: aa5590eb 17 04 8a020000 017c0893b33b1687b276 Discharge OTPR 650
Command: aa5590eb 18 04 24ffffff 77307fd8686f0227b66b Charge UTP -22
Command: aa5590eb 19 04 a6ffffff 15b39a96891db7da90f9 Charge UTPR -9
Command: aa5590eb 1a 04 de030000 c74eea81f7cf73b5e1c8 MOS OTP 990
Command: aa5590eb 1b 04 0c030000 21329850c768cf89d842 MOS OTPR 780
```

syssi, 2023-03-14:
```
Command: aa5590eb 23 04 640000000 02c914c08024b41d175 # Con Wire Res 01: 100
Command: aa5590eb 23 04 c80000000 13f881a899fc07218bd # Con Wire Res 02: 200
Command: aa5590eb 23 04 2c0100000 2f9eea6d367348f3f99 # Con Wire Res 03: 300
Command: aa5590eb 23 04 900100000 32d35263e01e3830c6e # Con Wire Res 04: 400
Command: aa5590eb 23 04 f40100000 4fe7c79c2da55a9234a # Con Wire Res 05: 500
Command: aa5590eb 23 04 580200000 51c1729305a39d006f5 # Con Wire Res 06: 600
Command: aa5590eb 23 04 bc0200000 61ec3097a36e45277ac # Con Wire Res 07: 700
Command: aa5590eb 23 04 200300000 70fb3a82ff5d6125596 # Con Wire Res 08: 800
```

syssi, 2023-03-14:
I've merged the new settings registers into `main`. All available number entities should work now if you reflash your ESP.

riker65, 2023-03-15:
Hi 
very fast development thanks a lot

so I just need to change protocol version JK02_32S right?

syssi, 2023-03-15:
Correct!

riker65, 2023-03-15:
the mac by the way is 
  mac_address: C8:47:8C:EC:77:7D
not sure if this same  for all B2A8S20P

riker65, 2023-03-15:
please check new sensors mqtt topics

jk-bms and sometime jk_bms  . Underscore versus minus

[image]

syssi, 2023-03-17:
```
RS485/CAN
    -1  01 01 00 01 01
   282  08 08 00 08 08 <---

Display always on
    -1  00 00 00 00 01 01 00 00
   282  08 08 08 08 18 18 08 08 <---
   
Temperature sensors disabled
    -1  00 00 00 01 01 01 00 00
   282  08 08 08 0a 0a 0a 08 08 <-- temp sensors disabled bit
```

syssi, 2023-03-18:
I've prepared a new release. Please give it a try! Many thanks for your great support!

closed 2023-03-18 by syssi

riker65, 2023-03-19:
thanks for fast implementing.
no errors found so far.

syssi, 2023-03-19:
I thank you too! :-)