syssi/esphome-jk-bms #287: 3.3V or 5V tap from JK BMS?

JohnGalt1717, 2023-03-16:
Anyone figure out how to get 3.3V or 5V from the JK BMS itself? I'm not adverse to opening it up to get it.

It would be nice to be able to just power the ESP32/8266 directly from the BMS instead of having to run additional power.

It has to have it somewhere, it's a matter of if it's exposed.

hotovson, 2023-03-16:
an 80V step-down DC-DC module on GPS port, i.e. https://www.aliexpress.com/item/1005001281280904.html?spm=a2g0o.order_list.order_list_main.22.4be21802K959mS

JohnGalt1717, 2023-03-16:
The problem is that that's constantly drawing while the JK BMS obviously has a 3.3V buck already onboard that should be tappable so no more draw required.

syssi, 2023-03-16:
The buck converter of the BMS cannot be used because it's too small and doesn't survive the drawn current of an ESP.

JohnGalt1717, 2023-03-24:
I ended up using this: https://www.mornsun-power.com/html/pdf/K78U03-500R3.html

Which worked great.

closed 2023-03-24 by JohnGalt1717

JohnGalt1717, 2023-03-24:
On the RS485 port there is ground and full battery voltage. I took pin 2 and hooked ground to the BMS and to the esp8266.  Then the VIN to the pin on the RS485 port and VOUT to the V3.3 on the ESP8266.

Then of course TX/RX per your diagram.  I also added a logic level fet on D1 for controlling a 5A external balancer that I have on the run pins for it.  And then in Home Assistant automations if the voltage of a single cell goes above 3.42V or below 3V it turns on the fet and thus the active balancer to balance the LiFEPO4 pack.

Works pretty well.

syssi, 2023-03-24:
Which ground (B- or P-) of the BMS did you use?

syssi, 2023-03-24:
You could move the automation from Home Assistant to the ESPHome node to be independent.

JohnGalt1717, 2023-03-24:
I just used the ground on the rs485 port. I presume that's battery ground so that it will always work.

Thanks for the suggestion on the automation. Will have to look into esphome automations.

syssi, 2023-03-24:
There is a component called `safe mode switch`. This can be used to boot the ESP into the safe mode to free resources / stop the UART traffic. Decoding the serial data at 115k2 is pretty heavy for the ESP8266.

JohnGalt1717, 2023-03-24:
Just did up a board design: https://oshwlab.com/johngalt1717/jkbms-rs485-board

Let me know what you think. I'm not a big PCB design guy, but I think I got it all correct.  Should be a drop in with population from JLCPCB if you want or you can solder yourself. Figure it would just go in heat shrink stuck to the side of the BMS case.

Designed for either an ESP8266 or an ESPC3-12 (same pin out)  I'd recommend the ESPC3 it's way faster.

syssi, 2023-03-25:
In the past (ESP01) some pins of the ESP must be pulled to boot into the flash and to avoid a floating reset pin if I remember correctly. Plus a capacitor was recommended. I don't know your ESP package in detail. If there are any recommendations at the specifications I would add these to the board too.

JohnGalt1717, 2023-03-25:
Here's the one I'm using. It's identical to the standard ESP8266-12F but has the C3.

https://www.farnell.com/datasheets/3621259.pdf

Looks like EN just needs to be at 3.3V for operation and then it can just program from TX/RX as normal. Do you see anything else that I'm missing?

Thinking about switching the power supply to the 5V version and adding a 5V to 3.3V LDO for stability and then a BJT on the mosfet trigger and using a different fet that can handle up to 100V with a 5V trigger so that 24S X 4.2V won't freak out.