syssi/esphome-jk-bms #291: no vales over MQTT and endless tree structure

eMindFfM, 2023-03-17:
Dear all, I just updated to the latest version an I do only get the configuration listed for all Items in the MQTT structure.
In addtion there is a endless tree structure in the MQTT path

[image]

[image]

syssi, 2023-03-18:
Could you provide your configuration YAML and the version number of the used ESPHome release?

eMindFfM, 2023-03-18:
esphome version
Version: 2022.12.3
(venv) torsten@debian:~$ python --version
Python 3.9.2

eMindFfM, 2023-03-18:
[DIY_Powerwall_2.txt](https://github.com/syssi/esphome-jk-bms/files/11007910/DIY_Powerwall_2.txt)
renamed the file to .txt instead of .yaml as it is not supported.

eMindFfM, 2023-03-18:
I remarked that I'm using both sections 'api:' and 'mqtt:' . Might this be the problem? I'll give it a go :-)

eMindFfM, 2023-03-18:
If I'm removing the section 'api:' it does not bild anymore. see the trace:
(venv) torsten@debian:~/esphome-jk-bms$ esphome run DIY_Powerwall_2.yaml
INFO Reading configuration DIY_Powerwall_2.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing jk-bms (board: az-delivery-devkit-v4; framework: espidf; platform: platformio/espressif32 @ 5.2.0)
-------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ArduinoJson @ 6.18.5
*** [.pioenvs/jk-bms/src/esphome/components/api/api_connection.o] Source `src/esphome/components/api/api_connection.cpp' not found, needed by target `.pioenvs/jk-bms/src/esphome/components/api/api_connection.o'.
========================================================== [FAILED] Took 2.29 seconds ==========================================================
(venv) torsten@debian:~/esphome-jk-bms$

syssi, 2023-03-18:
If you use `api` and `mqtt` in parallel please disable the `reboot_timeout` of the native API component:

```
api:
  reboot_timeout: 0s
```

Without disabling the timeout the Node will reboot every 15 minutes if there is no API connection.

syssi, 2023-03-18:
Please try a clean build:

```
esphome clean DIY_Powerwall_2.yaml
esphome run DIY_Powerwall_2.yaml
```

eMindFfM, 2023-03-18:
I was able to compile and build a version only using mqtt and 'api:' was disabled.
[image]
The tree structure is still present and no values are send via mqtt
Any other suggestions?

eMindFfM, 2023-03-18:
on the IOBroker Lebel ( mqtt subscriber ) I get the following messages

mqtt.0 | 2023-03-18 21:18:23.060 | warn | State "mqtt.0.jk-bms.debug" has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.060 | warn | State "mqtt.0.jk-bms.debug" has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.060 | warn | State "mqtt.0.jk-bms.debug" has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.060 | warn | State   "mqtt.0.homeassistant.sensor.jk-bms.jk-bms_total_runtime_formatted.config"  has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.059 | warn | State  "mqtt.0.homeassistant.sensor.jk-bms.jk-bms_errors.config" has no  existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.059 | warn | State  "mqtt.0.jk-bms.switch.jk-bms_enable_bluetooth_connection.state" has no  existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.059 | warn | State   "mqtt.0.homeassistant.switch.jk-bms.jk-bms_enable_bluetooth_connection.config"  has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.059 | warn | State "mqtt.0.jk-bms.switch.jk-bms_balancer.state" has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.059 | warn | State  "mqtt.0.homeassistant.switch.jk-bms.jk-bms_balancer.config" has no  existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.058 | warn | State "mqtt.0.jk-bms.switch.jk-bms_discharging.state" has no existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.056 | warn | State  "mqtt.0.homeassistant.switch.jk-bms.jk-bms_discharging.config" has no  existing object, this might lead to an error in future versions
mqtt.0 | 2023-03-18 21:18:23.056 | warn | State "mqtt.0.jk-bms.switch.jk-bms_charging.state" has no existing object, this might lead to an error in future versions

eMindFfM, 2023-03-19:
Please find attached the result
esphome logs DIY_Powerwall_3.yaml
INFO Reading configuration DIY_Powerwall_3.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller)
  [2] Over The Air (192.168.178.94)
(number): 2
INFO Starting log output from 192.168.178.94 using esphome API
INFO Successfully connected to 192.168.178.94
[11:26:10][I][app:102]: ESPHome version 2022.12.3 compiled on Mar 19 2023, 11:12:34
[11:26:10][I][app:104]: Project syssi.esphome-jk-bms version 1.3.0
[11:26:12][I][jk_bms_ble:464]: Cell info frame (version 2, 300 bytes) received
[11:26:18][I][jk_bms_ble:464]: Cell info frame (version 2, 300 bytes) received
[11:26:23][I][jk_bms_ble:464]: Cell info frame (version 2, 300 bytes) received
[11:26:28][I][jk_bms_ble:464]: Cell info frame (version 2, 300 bytes) received
[11:26:34][I][jk_bms_ble:464]: Cell info frame (version 2, 300 bytes) received

eMindFfM, 2023-03-19:
How can I see the content of the frames? Is there a logfile which is written to the local disc?

syssi, 2023-03-19:
If you increase the log level you will see much more details:

```
logger:
  level: DEBUG
```

If you increase the log level to `VERY_VERBOSE` you can see the raw traffic too:

```
logger:
  level: VERY_VERBOSE
  logs:
    esp32_ble_tracker: VERY_VERBOSE
    jk_bms_ble: VERY_VERBOSE
    scheduler: DEBUG
    component: DEBUG
    sensor: DEBUG
    mqtt: INFO
    mqtt.idf: INFO
    mqtt.component: INFO
    mqtt.sensor: INFO
    mqtt.switch: INFO
    api.service: INFO
    api: INFO
```

Please be careful this requires many resources and isn't recommended in production.

syssi, 2023-03-19:
You are using the `jk_bms_ble` component and you are able to receive cell info frames from your BMS. This is perfect! As soon as you increase the log level to `DEBUG` you will see the decoded values per sensor. The next step is to push this data to your MQTT broker.

The iobroker MQTT broker is a bit wird some times. If you have the chance I recommend to switch to `mosquitto`. If you want to stick to the MQTT broker of iobroker we have to debug the issue in detail.

1. The topics with the prefix `homeassistant/#` are discovery topics. They are used by Home Assistant to auto-discover the `state` and `command` topics of the ESPHome device. iobroker doesn't understand these informations so let's disable the publishing of discovery stuff:

```
mqtt:
   broker: !secret mqtt_host
   id: mqtt_client
   discovery: false
```

The topic structure `mqtt/0/mqtt/0/mqtt/0/mqtt/.../0/jk-bms/status` is new to me but `status` looks like the last will (`will_message`) of the MQTT component. Could you update your ESPHome installation to the latest release?

eMindFfM, 2023-03-19:
Update of esphome - done
esphome version
Version: 2023.3.0

mqtt settings are enabled

logger is to DEGUG

**esphome clean is executed**

INFO Reading configuration DIY_Powerwall_3.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Deleting .esphome/build/jk-bms/.pioenvs
INFO Deleting .esphome/build/jk-bms/.piolibdeps
INFO Done!

**esphome run -> causes an error. Is it realted to the warning of the ESP-IDF Version?**

-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/torsten/esphome-jk-bms/.esphome/build/jk-bms/sdkconfig.jk-bms
-- Configuring incomplete, errors occurred!
See also "/home/torsten/esphome-jk-bms/.esphome/build/jk-bms/.pioenvs/jk-bms/CMakeFiles/CMakeOutput.log".

fatal: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git
CMake Error at /home/torsten/.platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /home/torsten/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /home/torsten/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /home/torsten/.platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /home/torsten/.platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /home/torsten/.platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)

eMindFfM, 2023-03-19:
Where an how can I update the ESP-IDF? Which version shall I use?

syssi, 2023-03-19:
Please don't use the platform version `latest`. Remove the `version` parameter or use the option `recommended`:

```
esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf
    version: recommended
```

eMindFfM, 2023-03-19:
Now it dies build a new version and also loggs the information on DEGUG level. Great :-)
I does not write data to mqtt. Any Ideas?
[DIY_Powerwall_3.txt](https://github.com/syssi/esphome-jk-bms/files/11011422/DIY_Powerwall_3.txt)

[log.txt](https://github.com/syssi/esphome-jk-bms/files/11011383/log.txt)

Please find attached the log and the configuration

eMindFfM, 2023-03-19:
Many Thanks that you take so much time!! Well appreciated !
Could you tell me how I can send a command to the BMS using IOBroker?

syssi, 2023-03-19:
> Many Thanks that you take so much time!! Well appreciated!

You are welcome! The MQTT part still doesn't work, correct?

eMindFfM, 2023-03-19:
> > Many Thanks that you take so much time!! Well appreciated!
> 
> You are welcome! The MQTT part still doesn't work, correct?

Yes there is no data  received by my mqtt broker / client.
With other programs I do have no issues. So in general it works

syssi, 2023-03-19:
Could you provide a screenshot of the iobroker MQTT service? There was another user in the past which changed the settings slightly: https://github.com/syssi/esphome-jk-bms/issues/221#issuecomment-1332452070

eMindFfM, 2023-03-19:
[image]
[image]

eMindFfM, 2023-03-19:
Great! all data is now received :+1: great job

eMindFfM, 2023-03-19:
Could you tell me how I can send a command to the BMS using IOBroker?

syssi, 2023-03-19:
You will find a required details to control your BMS via MQTT here: https://github.com/syssi/esphome-jk-bms/discussions/197

Long story short: Next to each `state` topic there is a `command` topic. Just publish supported values (depending on the entity type here):

```
mosquitto_pub -t 'basement-btle-gateway/button/button_entity_id_example/command' -m 'PRESS'
mosquitto_pub -t 'basement-btle-gateway/switch/switch_entity_id_example/command' -m 'ON'    # Supported values: ON, OFF, TOGGLE
mosquitto_pub -t 'basement-btle-gateway/number/number_entity_id_example/command' -m 10
mosquitto_pub -t 'basement-btle-gateway/select/select_entity_id_example/command' -m 'option1'
```

A `button` entity doesn't have a state (topic) but you can publish a command (`PRESS`) nevertheless.

closed 2023-03-19 by syssi