syssi/esphome-jk-bms #295: Mention required settings of the ioBroker MQTT adapter

syssi, 2023-03-22:
1. https://github.com/syssi/esphome-jk-bms/issues/221#issuecomment-1332452070
2. https://github.com/syssi/esphome-jk-bms/discussions/197#discussioncomment-5378383

@eMindFfM Could you provide a screenshot of your current / working ioBroker MQTT adapter configuration? Thanks in advance!

eMindFfM, 2023-03-22:
[image]
[image]

syssi, 2023-03-22:
The background:

If the checkbox "Eigene States beim Verbinden publizieren" is checked the broker does publish a big bunch of messages as soon the connection is established. The ESP cannot handle this traffic and does crash in most cases. 

If the checkbox "States bei subscribe publizieren" isn't checked ioBroker ignores subscriptions and doesn't pass subscribed topics to the client.