syssi/esphome-jk-bms #297: [jk_bms_ble:454]: You hit the unstable auto detection of the protocol version. This feature will be removed in future!Please update your configuration to protocol version JK02_32S if you are using a JK-B2A8S20P v11+

austin202220, 2023-03-28:
[jk_bms_ble:454]: You hit the unstable auto detection of the protocol version. This feature will be removed in future!Please update your configuration to protocol version JK02_32S if you are using a JK-B2A8S20P v11+

I am getting that message in the logs. I am running a JK_BD4A8S4P, should I be concerned?

syssi, 2023-03-28:
Could you tell me the hardware and software version of your device? I assume your device is a `JK02` device and it shouldn't be "detected" as `JK02_32S`. Please check the `total_voltage` and `power_tube_temperature` if you see such a log message. I assume the sensor measurements are garbled in this case.

austin202220, 2023-03-28:
I have been running this BMS with your software since Jan 2023 or earlier. I dont really check the logs too often, but happened to be updating esphome versions from 2023.3.0 to 2023.3.2 and prior to updating I checked the logs and noticed the message.

Power tube temp and total voltage both are displaying accurate values.

JK_BD4A8S4P
HW: V11.XW
SW: V11.24
Ver: V4.10.1

[image]

[image]

austin202220, 2023-03-28:
I changed log level to DEBUG and here is the output:

[logs_jk-bms-2_run.txt](https://github.com/syssi/esphome-jk-bms/files/11090874/logs_jk-bms-2_run.txt)

austin202220, 2023-03-28:
Just for clarity, I have been running JK02 protocol, and the logs text file linked above is from that protocol version JK02.

I will update to JK02_32S, compile, and provide logs after the update.

syssi, 2023-03-28:
I tried to auto-detect the new frame structure (now called `JK02_32S`) in the past but it has turned out the auto-detection isn't perfect and fails every now and then. This is the cause why I aim to remove the auto-detection in future and let the user choose the correct protocol version.

It looks like your BMS (software version 11+) is auto-detected properly as `JK02_32S`. So it's safe to pin the protocol version to avoid the auto detection.

syssi, 2023-03-28:
The attachments are perfectly fine! Please ignore the `Unknown: ...` bytes. They are printed just for debugging purposes. You could reduce the log level to `INFO` now to get rid of these debug (`[D]`) messages.

syssi, 2023-03-28:
You device is a `JK02_32S` for sure. Your logs are looking good to me.

austin202220, 2023-03-28:
Great thanks! Consider updating the supported / tested models list - I dont think the BD4A8S4P was on there when I checked yesterday.

I have a few other models of BMS's available for testing if interested...
I have B2A20S20P, BD6A20S10P, and B1A20S15P

austin202220, 2023-03-28:
***new question about a different esp board and different bms's different from the above messages***

I have one ESP32-WROOM that is connecting to two JK BMSs, B2A20S20P and BD6A20S10P.

I get the "You hit the unstable auto detection of the protocol version. This feature will be removed in future!Please update your configuration to protocol version JK02_32S if you are using a JK-B2A8S20P v11+" warning in the logs for this one too. Do I need to go ahead and update the protocol on this one?

I am connecting to two different model bms on one esp chip, im not sure which one is causing the warning message, or both. And not sure whats different with jk-2_32s. What do you recommend?

syssi, 2023-03-28:
Please start with a single BMS setup / single BLE client connection and update the MAC address per test. As soon you know which protocol version to use you could flash your multiple BLE client connection configuration.

If the protocol version doesn't fit the temperature sensor will be probably zero and the total voltage won't match.

syssi, 2023-03-28:
The difference between `JK02` and `JK02_32S` is the number of bytes reserved for cell voltages and resistance. The cell info frame of both protocol versions is 300 bytes long / has a fixed size. At the `JK02` protocol version 24 cells are supported because there are 2*24 bytes reserved for cell voltages and 2*24 bytes reserved for resistances. At the `JK02_32S` there are 2*2*32 bytes reserved for cell voltages + resistances. This leads to a different offsets.

austin202220, 2023-03-28:
Got it, I'll do some testing and post the results, thanks!

closed 2023-04-03 by syssi

syssi, 2023-04-16:
> I have a few other models of BMS's available for testing if interested...
> I have B2A20S20P, BD6A20S10P, and B1A20S15P

Is your offer still valid?

austin202220, 2023-04-16:
> 

Yep, what would you like me to do?

syssi, 2023-04-16:
Please collect the MAC addresses of your devices. Flash a ESP32 with this YAML

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

make sure you are able to enabled/disable the discharging mosfet using the switch entity and provide the software and hardware version per device.

austin202220, 2023-04-17:
JK-B1A20S15P works well using jk02 protocol.
Hardware Ver: V10.XW
Software Ver: V10.10
Manuf Date: 2023-04-16
Version: V4.10.1

I just notice the manuf date is today... I wonder if this is because today is the first day its ever been powered on? I bought it a few months back and just wired it up today...

Attached is a log file that you can review. 

[logs_jk-bms-3_logs.txt](https://github.com/syssi/esphome-jk-bms/files/11246234/logs_jk-bms-3_logs.txt)

one bit that may be worth review:

> [22:49:54][D][sensor:127]: 'jk-bms-3 charging cycles': Sending state 0.00000  with 0 decimals of accuracy
[22:49:54][D][sensor:127]: 'jk-bms-3 total charging cycle capacity': Sending state 0.10400 Ah with 3 decimals of accuracy
[22:49:54][D][jk_bms_ble:647]: Unknown158: 0x64 0x00 (always 0x64 0x00?)
[22:49:54][D][jk_bms_ble:650]: Unknown160: 0x9D 0x05 (always 0xC5 0x09?)
[22:49:54][D][sensor:127]: 'jk-bms-3 total runtime': Sending state 19897.00000 s with 0 decimals of accuracy
[22:49:54][D][text_sensor:067]: 'jk-bms-3 total runtime formatted': Sending state '5h'
[22:49:54][D][jk_bms_ble:663]: Unknown168: 5B.06.00.00.00.00.00.00.00.00.00.00.00.00.07.00.01.00.00.00.C7.03.00.00.00.00.A9.76.42.40.00.00.00.00.E2.04.00.00.00.00.00.01.00.05.00.00 (46)
[22:49:54][D][jk_bms_ble:677]: Unknown189: 0x03 0x00
[22:49:54][D][jk_bms_ble:681]: Unknown192: 0x00
[22:49:55][D][api:102]: Accepted 192.168.7.43
[22:49:56][D][api.connection:920]: Home Assistant 2023.4.2 (192.168.7.43): Connected successfully
[22:49:59][I][jk_bms_ble:467]: Cell info frame (version 2, 300 bytes) received
[22:49:59][D][sensor:127]: 'jk-bms-3 cell voltage 1': Sending state 3.32800 V with 3 decimals of accuracy
[22:49:59][D][sensor:127]: 'jk-bms-3 cell resistance 1': Sending state 0.06100 Ω with 3 decimals of accuracy

syssi, 2023-04-17:
> Manuf Date: 2023-04-16

You are right! This is the date of the first power on. I don't know it's a new feature (and written by the Android app) or it has always been like this.