syssi/esphome-jk-bms #298: Increase precision of temperature measurements

octal-ip, 2023-03-28:
In _jk_bms_ble/sensor.py_, **accuracy_decimals** is set to 0 for all temperature sensors (see lines 613, 620, 627).

However the true accuracy can be 1 decimal place, as defined in _jk_bms_ble/jk_bms_ble.cpp_ (see lines 581, 584 and 592).

I recommend altering _CONF_TEMPERATURE_SENSOR_1_, _CONF_TEMPERATURE_SENSOR_2_ and _CONF_POWER_TUBE_TEMPERATURE_ to use **accuracy_decimals=1**.

syssi, 2023-03-28:
Good catch! I will prepare a fix!

closed 2023-03-28 by syssi

octal-ip, 2023-03-28:
Thank you, is there any way I can donate as a token of appreciation for your work?

syssi, 2023-03-29:
There is a [buy me a coffee](https://www.buymeacoffee.com/syssi) button on the top right corner of the README. Thanks in advance!

Sleeper85, 2024-10-01:
Hi @syssi 

The same problem is observed with the `jk_bms` component from `uart`.

The number of decimals is `0`, the BMS measures 17.9°C and the component reports 17°C.

Is it possible to increase it to 1 decimal place as well ?

Thanks

syssi, 2024-10-02:
@Sleeper85 There is no better resolution available at the payload of the `uart` frame. See https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/jk_bms.cpp#L129-L137

commit e9686e9f (2023-03-28): Increase precision of temperature measurements (Closes: #298) (#299)