syssi/esphome-jk-bms #301: Some improvements

preussal, 2023-03-29:
Some improvements

such as Total_charging_Cycle_capacity Ah -> kAh
[image]

```
sensor:
  - platform: jk_bms_ble
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
      # Convert Ah to kAh
      state_class: "total_increasing"
      unit_of_measurement: kAh
      icon: mdi:counter
      accuracy_decimals: 3
      filters:
        - multiply: 0.001
```

For home assistant the determination of how much kWh was charged and discharged
[image]

```
sensor:
  - platform: jk_bms_ble
    charging_power:
      name: "${name} charging power"
      id: bms0_charging_power
      unit_of_measurement: W
      device_class: energy
      state_class: measurement
      entity_category: diagnostic

    discharging_power:
      name: "${name} discharging power"
      id: bms0_discharging_power
      unit_of_measurement: W
      device_class: energy
      state_class: measurement
      entity_category: diagnostic

  - platform: total_daily_energy
    name: "${name} Total Charging Energy Today"
    power_id: bms0_charging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001

  - platform: total_daily_energy
    name: "${name} Total Discharging Energy Today"
    power_id: bms0_discharging_power
    unit_of_measurement: kWh
    icon: mdi:counter
    accuracy_decimals: 3
    restore: true
    filters:
      - multiply: 0.001  
```

Restart Switch to reboot the ESP. And hiding the Bluetooth Switch.
```
switch:
  # Restart the ESP via Rebooting safely
  - platform: restart
    name: "${name} Restart / Reboot Device"
    icon: "mdi:restart"

[... 376 lines cut ...]
    name: "${name} Restart / Reboot Device"
    icon: "mdi:restart"

  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"
    internal: True

text_sensor:
  - platform: jk_bms_ble
    #errors:
    #  name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```

[image]

syssi, 2023-03-29:
Thanks for your support! One minor remark: Feel free to remove the `enable bluetooth connection` switch if you don't need them. Defining the entity as `internal` doesn't mark much sense if you don't want to use the entity by automations.

preussal, 2023-03-30:
I've already tried that. But when building, he said that he couldn't find the component.

syssi, 2023-03-30:
Try a clean build:

```
esphome clean yourconfig.yaml
esphome run yourconfig.yaml
```

strelok, 2023-09-15:
@syssi maybe you know why I don't have balancing current and switch for example? it's available only via BLE? 
I'm using 'esp32-example.yaml' from you repo, UART-TTL (RS485).

syssi, 2023-11-05:
@strelok The balancing current is available via BLE only. The balancer switch should be available via UART-TTL but doesn't work on new devices/hardware version probably.

See https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms/switch/__init__.py#L20-L21