syssi/esphome-jk-bms #304: Sensor "online_status" available from Home Assistant ?

rfenouil, 2023-04-04:
Hi :)

Apologies for opening a second issue in a day, I love how your lib integrates to Home assistant but I think I misunderstand something.

In recent commits, you added a `binary_sensor` named `online_status`, but I cannot find it in home assistant.

I would like to use it for a conditional card in lovelace UI, so I get an alternative presentation when the device is not available (instead of having all values showing 'unavailable').

Is it exported to mqtt or only designed for internal/web interface ?

On the same flavor, should I be able to access to a web interface when connecting to the eps32 ip adress with a browser ?
If so, I probably don't compile the most up to date version because I can't access it... Or maybe it is served on a specific port ?

Thank you again for your help.

syssi, 2023-04-04:
Us the binary sensor `online_status` the entity you are looking for? It is part of your YAML already?

syssi, 2023-04-05:
The ESPHome Integration of Home Assistant is able to talk to the native `api` of the ESPHome node directly. It's the recommended way to talk to ESPHome devices if Home Assistant is in use. In this case no MQTT broker is required anymore.

I assume the MQTT auto-discovery of the exposed entities didn't work well here. If you switch to the native `api` I'm pretty sure you will find the "online status" sensor entities this time.

rfenouil, 2023-04-05:
It does work with api insead of MQTT... 
Still a bit frustrated not to understand what happened but I'll switch to `api`, seems simpler.

Thank you !!

closed 2023-04-05 by rfenouil