syssi/esphome-jk-bms #305: MQTT Issue

andyw73, 2023-04-07:
When only using mqtt protocol and not home assistant api a 'homeassistant' topic also gets sent to the mqtt broker (this doesn't keep updating the sensors and from what i can tell only happens on reboots). How can we stop this behaviour from happening as I dont want to integrate this into Ha. Thanks in advance.

[image]

syssi, 2023-04-07:
The `homeassistant` topic is used for meta informations about the node / exposed sensors to allow Home Assistant to auto-discover the node. Please lookup the parameters of the `mqtt` component of ESPHome. There is an option called `discovery` which can be disabled. This will suppress the auto-discovery messages.

syssi, 2023-04-07:
```
mqtt:
   broker: !secret mqtt_host
   id: mqtt_client
   discovery: false
```

andyw73, 2023-04-07:
Thank you that resolved the issue.

closed 2023-04-07 by syssi