syssi/esphome-jk-bms #308: Lost information

Salve87, 2023-04-12:


Salve87, 2023-04-12:
Hello, 
after months of perfect communication. I now (nothing changed) lost a lot of information.
Cell voltages is still updated. Other information as SOC for example is not updated. 
Using the Bluetooth app everything works fine 

[image]

syssi, 2023-04-12:
Could you provide an ESPHome log, the BMS model name and the software/hardware version? I've have an idea what's going on.

Salve87, 2023-04-12:
> Could you provide an ESPHome log, the BMS model name and the software/hardware version? I've have an idea what's going on.

2023-04-12 06:46:32,361 INFO 304 GET /devices (0.0.0.0) 2.56ms
2023-04-12 06:46:37,367 INFO 304 GET /devices (0.0.0.0) 2.86ms
2023-04-12 06:49:06,189 INFO 200 GET /devices (0.0.0.0) 2.85ms
2023-04-12 06:54:06,173 INFO 200 GET /devices (0.0.0.0) 2.73ms
2023-04-12 06:59:06,148 INFO 200 GET /devices (0.0.0.0) 2.91ms
2023-04-12 07:00:01,451 INFO 304 GET / (0.0.0.0) 2.01ms
2023-04-12 07:00:01,479 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 2.82ms
2023-04-12 07:00:01,495 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 (0.0.0.0) 2.74ms
2023-04-12 07:00:01,514 INFO 200 GET /static/css/esphome-2.css?hash=e81a30ae (0.0.0.0) 2.24ms
2023-04-12 07:00:01,526 INFO 200 GET /static/js/esphome/index-67cb530c.js (0.0.0.0) 8.26ms
2023-04-12 07:00:01,584 INFO 200 GET /static/js/esphome/c.a24dad9a.js (0.0.0.0) 4.43ms
2023-04-12 07:00:01,591 INFO 200 GET /static/js/esphome/c.abbbe9de.js (0.0.0.0) 2.70ms
2023-04-12 07:00:01,598 INFO 200 GET /static/js/esphome/c.bc0ca5c5.js (0.0.0.0) 3.99ms
2023-04-12 07:00:01,640 INFO 200 GET /static/images/logo-text.svg (0.0.0.0) 3.05ms
2023-04-12 07:00:01,710 INFO 200 GET /static/js/esphome/c.e7806b41.js (0.0.0.0) 60.00ms
2023-04-12 07:00:01,725 INFO 304 GET /devices (0.0.0.0) 10.00ms
2023-04-12 07:00:01,729 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 11.66ms
2023-04-12 07:00:02,138 INFO 200 GET /static/js/esphome/c.5cd5d307.js (0.0.0.0) 12.06ms
2023-04-12 07:00:02,147 INFO 200 GET /static/js/esphome/c.2fd5868c.js (0.0.0.0) 4.06ms
2023-04-12 07:00:06,647 INFO 304 GET /devices (0.0.0.0) 2.91ms
2023-04-12 07:00:11,646 INFO 304 GET /devices (0.0.0.0) 2.87ms
2023-04-12 07:00:16,648 INFO 304 GET /devices (0.0.0.0) 2.92ms
2023-04-12 07:04:06,144 INFO 200 GET /devices (0.0.0.0) 2.18ms
2023-04-12 07:09:06,146 INFO 200 GET /devices (0.0.0.0) 2.13ms
2023-04-12 07:14:06,151 INFO 200 GET /devices (0.0.0.0) 2.10ms
2023-04-12 07:19:06,143 INFO 200 GET /devices (0.0.0.0) 2.39ms
2023-04-12 07:24:06,145 INFO 200 GET /devices (0.0.0.0) 2.24ms
2023-04-12 07:29:06,144 INFO 200 GET /devices (0.0.0.0) 2.06ms
2023-04-12 07:32:06,717 INFO 304 GET / (0.0.0.0) 2.53ms
2023-04-12 07:32:07,330 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 2.66ms
2023-04-12 07:32:07,545 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 (0.0.0.0) 3.20ms
2023-04-12 07:32:07,555 INFO 200 GET /static/js/esphome/index-67cb530c.js (0.0.0.0) 3.61ms
2023-04-12 07:32:07,561 INFO 200 GET /static/css/esphome-2.css?hash=e81a30ae (0.0.0.0) 2.86ms
2023-04-12 07:32:08,353 INFO 200 GET /static/js/esphome/c.a24dad9a.js (0.0.0.0) 3.53ms
2023-04-12 07:32:08,367 INFO 200 GET /static/js/esphome/c.abbbe9de.js (0.0.0.0) 2.79ms
2023-04-12 07:32:08,378 INFO 200 GET /static/js/esphome/c.bc0ca5c5.js (0.0.0.0) 3.50ms
2023-04-12 07:32:08,880 INFO 200 GET /static/images/logo-text.svg (0.0.0.0) 2.95ms
2023-04-12 07:32:09,023 INFO 304 GET /devices (0.0.0.0) 5.91ms
2023-04-12 07:32:09,034 INFO 200 GET /static/js/esphome/c.e7806b41.js (0.0.0.0) 70.95ms
2023-04-12 07:32:09,238 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 2.74ms
2023-04-12 07:34:06,168 INFO 200 GET /devices (0.0.0.0) 2.97ms
2023-04-12 07:39:10,256 INFO 200 GET /devices (0.0.0.0) 13.01ms
2023-04-12 07:46:35,264 INFO 200 GET /devices (0.0.0.0) 4.48ms
2023-04-12 07:51:37,466 INFO 200 GET /devices (0.0.0.0) 3.23ms
2023-04-12 07:56:37,467 INFO 200 GET /devices (0.0.0.0) 2.94ms
2023-04-12 08:01:37,473 INFO 200 GET /devices (0.0.0.0) 2.90ms
2023-04-12 08:06:37,468 INFO 200 GET /devices (0.0.0.0) 2.88ms
2023-04-12 08:09:09,032 INFO 200 GET /devices (0.0.0.0) 2.52ms
2023-04-12 08:14:23,510 INFO 200 GET /devices (0.0.0.0) 2.59ms
2023-04-12 08:19:23,459 INFO 200 GET /devices (0.0.0.0) 2.22ms
2023-04-12 08:24:23,466 INFO 200 GET /devices (0.0.0.0) 2.60ms
2023-04-12 08:29:23,463 INFO 200 GET /devices (0.0.0.0) 3.22ms
2023-04-12 08:34:23,462 INFO 200 GET /devices (0.0.0.0) 2.58ms
2023-04-12 08:39:23,474 INFO 200 GET /devices (0.0.0.0) 2.59ms
2023-04-12 08:44:23,459 INFO 200 GET /devices (0.0.0.0) 2.63ms
2023-04-12 08:49:23,463 INFO 200 GET /devices (0.0.0.0) 3.44ms
2023-04-12 08:54:23,464 INFO 200 GET /devices (0.0.0.0) 3.10ms
2023-04-12 08:59:23,462 INFO 200 GET /devices (0.0.0.0) 3.25ms
2023-04-12 09:04:23,464 INFO 200 GET /devices (0.0.0.0) 2.98ms
2023-04-12 09:09:23,467 INFO 200 GET /devices (0.0.0.0) 2.57ms
2023-04-12 09:14:23,460 INFO 200 GET /devices (0.0.0.0) 2.83ms
2023-04-12 09:19:23,470 INFO 200 GET /devices (0.0.0.0) 2.51ms
2023-04-12 09:24:23,463 INFO 200 GET /devices (0.0.0.0) 2.46ms
2023-04-12 09:29:23,465 INFO 200 GET /devices (0.0.0.0) 3.30ms
2023-04-12 09:34:23,462 INFO 200 GET /devices (0.0.0.0) 3.36ms
2023-04-12 09:39:23,469 INFO 200 GET /devices (0.0.0.0) 2.75ms
2023-04-12 09:44:23,554 INFO 200 GET /devices (0.0.0.0) 2.10ms
2023-04-12 09:49:23,470 INFO 200 GET /devices (0.0.0.0) 2.10ms
[... 7 lines cut ...]
2023-04-12 10:27:44,336 INFO 304 GET / (0.0.0.0) 2.33ms
2023-04-12 10:27:44,480 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 3.26ms
2023-04-12 10:27:44,489 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2?hash=570eb838 (0.0.0.0) 2.49ms
[... cut ...]
2023-04-12 10:27:51,503 INFO Running command 'esphome --dashboard config /config/esphome/esp32-cam.yaml'
2023-04-12 10:27:51,674 INFO 200 GET /static/js/esphome/c.5cd5d307.js (0.0.0.0) 8.62ms
2023-04-12 10:27:53,858 INFO Process exited with return code 0
2023-04-12 10:27:55,482 INFO 304 GET /devices (0.0.0.0) 2.92ms
2023-04-12 10:28:00,439 INFO 304 GET /devices (0.0.0.0) 2.66ms
2023-04-12 10:28:05,469 INFO 304 GET /devices (0.0.0.0) 2.70ms
2023-04-12 10:29:23,579 INFO 200 GET /devices (0.0.0.0) 3.08ms
2023-04-12 10:34:23,460 INFO 200 GET /devices (0.0.0.0) 2.66ms

syssi, 2023-04-12:
This looks like the log of the ESPHome Dashboard. Could you provide the log of the ESHPhome node? It can be downloaded if you open the ESPHome Node tile at the dashboard. There is a "logs" button at the bottm right corner.

Salve87, 2023-04-12:
> This looks like the log of the ESPHome Dashboard. Could you provide the log of the ESHPhome node? It can be downloaded if you open the ESPHome Node tile at the dashboard. There is a "logs" button at the bottm right corner.

Do you mean this button? 
[image]

I got to check if I get an download file on the pc. On the cell nothing shows up

syssi, 2023-04-12:
No. ;-) Do you use the ESPHome Dashboard? It's an HA Add-On accessible via the main navigation on the left.

Salve87, 2023-04-12:
> 

I 

> No. ;-) Do you use the ESPHome Dashboard? It's an HA Add-On accessible via the main navigation on the left.

I have it running, but not for the jk bms esp.
At that time when installing the esp, I was totally new to HA. A friend of mine flashed the esp…

syssi, 2023-04-12:
Do you have a copy of the YAML configuration?

Salve87, 2023-04-12:
> In this case I suggest to start from scratch. Use the BLE configuration example from this project, update the MAC address and flash this configuration using the HA Dashboard to your ESP32. So you have access to the logs in future and you are able to update the device whenever you want.

Ok as suggested, new ESP, new code, new setup: same problem:

This Time I have a log file:
[logs_esp32-jk-bms_logs (1).txt](https://github.com/syssi/esphome-jk-bms/files/11213747/logs_esp32-jk-bms_logs.1.txt)

Any Idea whats wrong, why its not working all of a sudden ??
Thanks

syssi, 2023-04-12:
Could you tell me your BMS model and software/hardware version?

Salve87, 2023-04-12:
> Could you tell me your BMS model and software/hardware version?

sure, sorry forgot
[image]

syssi, 2023-04-12:
Please change the protocol version from `JK02` to `JK02_32S` and compile & flash your ESP with the updated configuration:

```
substitutions:
  name: jk-bms
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: ...
  protocol_version: JK02_32S
```

Does this fix the issue?

Salve87, 2023-04-12:
> JK02_32S

now it won't connect:
19:31:44][D][esp32_ble_tracker:246]: Starting scan...
[19:31:46][W][jk_bms_ble:365]: [C8:47:8C:E6:EA:C5] Not connected
[19:31:49][D][esp32_ble_client:048]: [0] [C8:47:8C:E6:EA:C5] Found device
[19:31:49][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[19:31:49][I][esp32_ble_client:064]: [0] [C8:47:8C:E6:EA:C5] 0x00 Attempting BLE connection
[19:31:50][D][esp-idf:000]: W (269009) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:31:50][D][esp-idf:000]: W (269125) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:31:50][D][esp-idf:000]: W (269219) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:31:50][D][esp-idf:000]: W (269330) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[19:31:50][D][esp-idf:000]: W (269334) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[19:31:50][W][esp32_ble_client:134]: [0] [C8:47:8C:E6:EA:C5] Connection failed, status=133
[19:31:50][D][esp32_ble_tracker:246]: Starting scan...
[19:31:50][D][esp32_ble_client:048]: [0] [C8:47:8C:E6:EA:C5] Found device
[19:31:50][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[19:31:50][I][esp32_ble_client:064]: [0] [C8:47:8C:E6:EA:C5] 0x00 Attempting BLE connection

Salve87, 2023-04-12:
> now it won't connect: 19:31:44][D][esp32_ble_tracker:246]: Starting scan... [19:31:46][W][jk_bms_ble:365]: [C8:47:8C:E6:EA:C5] Not connected [19:31:49][D][esp32_ble_client:048]: [0] [...]

Reatart helped now it’s running, 
Thanks for the support 👍

syssi, 2023-04-12:
All measurements are decoded properly again?

closed 2023-04-12 by syssi