syssi/esphome-jk-bms #310: Connecting multiple Helltec/Neey balancers via BT ?

fabhund, 2023-04-18:
Thank you for the great work on adding support for the Neey balancer.

It works flawless for 1 unit at the time. I would like to connect up to 4 units via BT.
But I cannot seem to figure out how i create the clients/platforms/sensor elements for the additional MAC addresses.

Can anyone, helt me out with an example, maybe for 2 balancers and 2 sensor, so I can get the syntax and repeat it.
I' using the "esp32-heltec-balancer-ble-example.yaml" file
-------YAML code-----
substitutions:
  name: heltec-balancer
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E1:E2:FF

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }
ble_client:
  - mac_address: ${mac_address}
    id: client0

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: heltec_balancer_ble
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"

button:
  - platform: heltec_balancer_ble
    retrieve_settings:
      name: "${name} retrieve settings"
      id: retrieve_settings_button
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"

syssi, 2023-04-18:
```
substitutions:
  name: heltec-balancer
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E1:E2:AA
  mac_address2: C8:47:8C:E1:E2:BB
  mac_address3: C8:47:8C:E1:E2:CC

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1
  - mac_address: ${mac_address3}
    id: client2

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: bms0
  - ble_client_id: client1
    throttle: 5s
    id: bms1
  - ble_client_id: client2
    throttle: 5s
    id: bms2

binary_sensor:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms0
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms1
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms2
    balancing:
      name: "${name} balancing"
    online_status:
      name: "${name} online status"

button:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms0
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms1
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms2
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"
    retrieve_factory_defaults:
      name: "${name} retrieve factory defaults"
```

syssi, 2023-04-18:
A ESP32 is limited to 3 simultaneous BLE client connections.

fabhund, 2023-04-18:
Thank you very much for the example code. Will try it out later today. If ESP "only" supports 3 connections I may have to use 2 EPS, not that big of an issue.

The Neey balancer is the 4.th generation. 
Following is reported by IOS app: (in app store search for "GeeWe BMS" it has an icon as a cartoonified pig !?) 
Product model GW-24SEB
HW-version: HW-3.2.0
Software version: v1.1.1
Production date: 20221201

I do not have my own Android phone, but we tried on a friends phone to install an app via an .apk file and that worked fine also.
https://github.com/NEEY-electronic/NEEY-electronic

Please reach out if I can help in any way. I have access to 4 Neey balancers, ESPhome/HA, and ISO app. (Maybe an android phone can be borrowd)

syssi, 2023-04-18:
Could you double check the product model name?

fabhund, 2023-04-18:
Confirmed a char is missing, the correct product model data.
Also please note i mixed up the software version with app version. Below is updated details.
Product model: GW-24S4EB
Hardware version: HW-3.2.0
Software version: ZH-1.2.4
App version: V.1.1.1

closed 2023-04-19 by syssi

moscito1010, 2023-05-07:
Hello
Can someone link me to instructions for an absolute beginner so that I can get the code on an esp32.
I've tried it for a while now but I don't have all the basics to do it.

I have a NEEY bms 4th 4A and would like to visualize the BMS data in Home Assistant. Unfortunately, I've really never worked with platformIO and Visual Studio. For me it's like sitting in the space shuttle and it should land :-)
Since I don't know at all how to proceed I could use a simple good guide. Preferably a YT post.

syssi, 2023-05-08:
@moscito1010 Please create a new issue. I will provide some guidance there!

trippfam07, 2024-02-20:
Hello all, I am trying this with a node mcu 32s and not having any luck.
Could someone point me in right direction?

This is the error log I am getting...

Processing heltec-balancer (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40302.0 (4.3.2) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
 - toolchain-riscv32-esp 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32 8.4.0+2021r2-patch2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2-patch2
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.o
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/md5/md5.o
Compiling /data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/mdns/mdns_component.o
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp: In member function 'virtual void esphome::heltec_balancer_ble::HeltecBalancerBle::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:194:71: error: 'class esphome::ble_client::BLEClient' has no member named 'get_gattc_if'; did you mean 'gattc_if'?
       auto status = esp_ble_gattc_register_for_notify(this->parent()->get_gattc_if(), this->parent()->get_remote_bda(),
                                                                       ^~~~~~~~~~~~
                                                                       gattc_if
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:194:103: error: 'class esphome::ble_client::BLEClient' has no member named 'get_remote_bda'; did you mean 'remote_bda'?
       auto status = esp_ble_gattc_register_for_notify(this->parent()->get_gattc_if(), this->parent()->get_remote_bda(),
                                                                                                       ^~~~~~~~~~~~~~
                                                                                                       remote_bda
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp: In member function 'bool esphome::heltec_balancer_ble::HeltecBalancerBle::send_command(uint8_t, uint8_t, uint8_t, uint32_t)':
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:768:47: error: 'class esphome::ble_client::BLEClient' has no member named 'get_gattc_if'; did you mean 'gattc_if'?
       esp_ble_gattc_write_char(this->parent_->get_gattc_if(), this->parent_->get_conn_id(), this->char_handle_,
                                               ^~~~~~~~~~~~
                                               gattc_if
src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.cpp:768:78: error: 'class esphome::ble_client::BLEClient' has no member named 'get_conn_id'; did you mean 'conn_id'?
       esp_ble_gattc_write_char(this->parent_->get_gattc_if(), this->parent_->get_conn_id(), this->char_handle_,
                                                                              ^~~~~~~~~~~
                                                                              conn_id
*** [/data/heltec-balancer/.pioenvs/heltec-balancer/src/esphome/components/heltec_balancer_ble/heltec_balancer_ble.o] Error 1
========================== [FAILED] Took 9.05 seconds ==========================