syssi/esphome-jk-bms #322: esp 32 -- jkBMS - Bluetooth

sralus, 2023-05-01:
So I got a new JKBMS B2A20S20P

in the const char* device name = "JK-B2A20S20P"; it had to be changed so that the ESP connects to the JK via bluetooth:
constchar* device name = "JK_B2A20S20P"; so _ instead of -.
My JK-BMS has the hardware ver V11.XW - software ver V11.26 and version V4.10.1
If I now establish a connection, only the cell data is displayed to me, everything else is no longer filled with data.
`{"Cell":{"0":3.409,"1":3.408,"2":3.407,"3":3.407,"4":3.406,"5":3.407,"6":3.408,"7":3.408,"8":3.408,"9":3.407,"10":3.407,"11":3.406,"12":3.407,"13":3.406,"14":3.407,"15":3.407},"Battery":{"Battery_Voltage":0.000,"Delta_Cell_Voltage":0.000,"MOS_Temp":0.000,"Battery_T1":0.000,"Battery_T2":0.000,"Battery_Power":0.000,"Charge_Current":0.000,"Percent_Remain":0,"Cycle_Count":455,"Balance_Current":0.000,"Uptime_Sec":27,"Uptime_Min":9,"Uptime_Hr":20,"Uptime_Day":185,"Charge":"off","Discharge":"off"}}`
 

I flashed the 96=11923-JKBMS_BLE_connection to my ESP32 742 times.
`#include <dummy.h>

#include <dummy.h>

#include "BLEDevice.h"
#include <WiFi.h>
#include <WebServer.h>
#include <PubSubClient.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include "driver/gpio.h"

#define SW_Version "1.1"

bool debug_flg = true;
bool debug_flg_full_log = false;

//########### Settings beginnen #########

//OTA Setting
#define OTA_Hostname "JK-BMS"  // -> hier ändern wenn OTA Name anders sein soll
#define OTA_Passwort "xxxxx"     // -> hier ändern wenn OTA Passwort anders sein soll

// MQTT Setting
const bool MQTT_ENABLE = true;
const char* mqtt_server = "192.168.0.69";     // -> hier die IP des MQTT Server eingeben  
const int mqtt_port = 1883;                     // -> hier den Port einstellen für den MQTT Server 
const char* mqtt_username = "";            // -> hier MQTT Benutzername eintragen
const char* mqtt_passwort = "";  // -> hier MQTT Passwort eingeben       
String mqttname = "BMS_Publisher";          // -> hier wird der MQTT Gerätename festgelegt
const int mqttpublishtime_offset = 5000;        //-> hier einstellen wie oft Danten gesnedet werden sollen 1000 = jede Sekunde 

//W-LAN Setting
const char* ssid = "LSTT";
const char* password = "xxxxxxxxx";

//BMS-BLE Settings
const char* Geraetename = "JK_B2A20S20P";  //JK-B2A24S20P JK-B2A24S15P JK_B2A8S20P
const bool protocol_32 = false;   // true for JK_B2A8S20P else false

//########### Ende der "User" Settings #########

// WebServer
WebServer server(80);

static BLEUUID serviceUUID("ffe0"); // The remote service we wish to connect to.
static BLEUUID    charUUID("ffe1"); //ffe1 // The characteristic of the remote service we are interested in.
BLEClient*  pClient;
BLEScan* pBLEScan;
byte getdeviceInfo[20] = {0xaa, 0x55, 0x90, 0xeb, 0x97, 0x00, 0xdf, 0x52, 0x88, 0x67, 0x9d, 0x0a, 0x09, 0x6b, 0x9a, 0xf6, 0x70, 0x9a, 0x17, 0xfd}; // Device Infos
//byte getInfo[20] = {0xaa, 0x55, 0x90, 0xeb, 0x96, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x10}; 
byte getInfo[20] = {0xaa, 0x55, 0x90, 0xeb, 0x96, 0x00, 0x79, 0x62, 0x96, 0xed, 0xe3, 0xd0, 0x82, 0xa1, 0x9b, 0x5b, 0x3c, 0x9c, 0x4b, 0x5d}; 
//byte balancer_an[20] = {0xaa, 0x55, 0x90, 0xeb, 0x1f, 0x04, 0x01, 0x00, 0x00, 0x00, 0xd3, 0x27, 0x86, 0x3b, 0xe2, 0x12, 0x4d, 0xb0, 0xb6, 0x00};

unsigned long sendingtime = 0;
unsigned long bleScantime = 0;
unsigned long mqttpublishtime = 0;
unsigned long newdatalasttime = 0;
unsigned long ble_connection_time = 0;
[... 379 lines cut ...]
BMS_Publisher/Data/Zelle_13 3.407
BMS_Publisher/Data/Zelle_14 3.406
BMS_Publisher/Data/Zelle_15 3.406
BMS_Publisher/Data/Zelle_16 3.407
BMS_Publisher/Data/Battery_Voltage 0.000
BMS_Publisher/Data/Delta_Cell_Voltage 0.000
BMS_Publisher/Data/MOS_Temp 0.000
BMS_Publisher/Data/Battery_T1 0.000
BMS_Publisher/Data/Battery_T2 0.000
BMS_Publisher/Data/Battery_Power 0.000
BMS_Publisher/Data/Charge_Current 0.000
BMS_Publisher/Data/Percent_Remain 0
BMS_Publisher/Data/Cycle_Count 456
BMS_Publisher/Data/Balance_Current 0.000
BMS_Publisher/Uptime/Sekunde 11
BMS_Publisher/Uptime/Minuten 57
BMS_Publisher/Uptime/Stunden 1
BMS_Publisher/Uptime/Tage 185
BMS_Publisher/Data/Charge off
BMS_Publisher/Data/Discharge off
BMS_Publisher/BLEconnection connected
BMS_Publisher/status online`

I would be happy about an info
Sralus

syssi, 2023-05-01:
Your question is about a foreign implementation!? Let's flash the implementation from this project to your ESP32.

sralus, 2023-05-01:
OK
So I would like to read out my JK-BMS with the ESP32 via Bluetooth and send it to the iobroker on my Raspberry pi via MQTT.
Could you tell me which file I have to flash.
Kind regards

syssi, 2023-05-01:
This one: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml

Please change the MAC address at line 5 to the MAC address of your BMS: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-example.yaml#L5

sralus, 2023-05-01:
i flash my esp32 with arduino ide 2.1.0.
Is it possible to flash your file like this?

syssi, 2023-05-01:
The Arduino IDE isn't an option here. You have to use ESPHome to flash your ESP. See https://esphome.io/guides/installing_esphome.html

closed 2023-05-02 by syssi