syssi/esphome-jk-bms #327: Never work before with Visual Studio/PlattformIO.

moscito1010, 2023-05-08:
Hello
Can someone link me to instructions for an absolute beginner so that I can get the code on an esp32.
I've tried it for a while now but I don't have all the basics to do it.

I have a NEEY bms 4th 4A and would like to visualize the BMS data in Home Assistant. Unfortunately, I've really never worked with plattformIO and Visual Studio. For me it's like sitting in the space shuttle and it should land :-)
Since I don't know at all how to proceed I could use a simple good guide. Preferably a YT post.

syssi, 2023-05-08:
Do you have access to an Linux host? Visual Studio isn't required. A few console commands should be sufficient:

https://raw.githubusercontent.com/syssi/esphome-seplos-bms/main/install.gif

1. Use a x86 based Linux host. Please try to avoid a Raspberry Pi because of the CPU architecture
2. Make sure python 3.x and the python package manage (called `pip`) is installed
3. Install `esphome` using the package manager `pip3 install esphome`
4. The to execute the command line tool `esphome`: `~/.local/bin/esphome`
5. Create a directory for your esphome configuration YAMLs: `mkdir ~/esphome`
6. Download the Heltec Balancer configuration example: `cd ~/esphome; wget https://raw.githubusercontent.com/syssi/esphome-jk-bms/main/esp32-heltec-balancer-ble-example.yaml`
7. Update line 5 (the `mac_address`) to the MAC address of your balancer
8. Create a `secrets.yaml` and add some required secrets:

```
cat > ~/esphome/secrets.yaml <<EOF
wifi_ssid: MY_WIFI_SSID
wifi_password: MY_WIFI_PASSWORD

mqtt_host: MY_MQTT_HOST
mqtt_username: MY_MQTT_USERNAME
mqtt_password: MY_MQTT_PASSWORD
EOF
```

10. Run the YAML using the `esphome` CLI (means: compile & flash the config): `esphome run esp32-heltec-balancer-ble-example.yaml`
11. Have fun! :-)

moscito1010, 2023-05-08:
I'm really blown away. Thank you for your quick help.
In fact, I only have Linux on my various Raspberry, but I'll quickly create a linux partition and install Ubuntu.

moscito1010, 2023-05-08:
oh wait. I have a NEEY balancer that only has bluetooth.
I wanted to use the ESP as a BLE2WIFI bridge.
Did I misunderstand the project? I thought it was about using an ESP32 to implement communication between the balancer and e.g. Home Assistant.

moscito1010, 2023-05-08:
forget it, I think I got it now :-)

syssi, 2023-05-08:
Your understanding is correct. The mentioned YAML talks to your NEEY/Heltec Balancer via BLE and publishs the received measurements via MQTT. The `command` topics can be used to control the balancer (settings / switches).

moscito1010, 2023-05-08:
Something is still wrong.
####
~/esphome$ ~/.local/bin/esphome run esp32-heltec-balancer-ble-example.yaml
INFO Reading configuration esp32-heltec-balancer-ble-example.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
Failed config

wifi: [source esp32-heltec-balancer-ble-example.yaml:24]
   ssid: !secret wifi_ssid
  
   Must be string, got <class 'esphome.helpers.EInt'>. did you forget putting quotes around the value?.
   Password: 12345678
######

I adapted the config on my board with the information from the website https://registry.platformio.org/platforms/platformio/espressif32/boards

#####
substitutions:
   name: heltec-balancer
   device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluet>
   external_components_source: github://syssi/esphome-jk-bms@main
   mac_address: 00:E4:4C:21:D7:53

esphome:
   name: ${name}
   comment: ${device_description}
   projects:
     name: "moscitobox.battery1"
     version: 1.4.0

esp32:
   board: az-delivery-devkit-v4
   frame:
     type: esp-idf
######

the secret.yaml looks like this

###
wifi_ssid: moscitobox
wifi_password: 12345678
mqtt_host: 192.168.178.48
mqtt_username: pi
mqtt_password: 12345678
###

Of course that's not my password :-)

what i should mention i have 5 balancers. do I need an ESP32 for each balancer?

I also use an ESP32 for another project to control my Bluetooth heating valves, so I can connect 6 devices to one ESP.

moscito1010, 2023-05-08:
quotation marks had to be set. It took me a while to understand :-)

moscito1010, 2023-05-09:
Es lies mir keine Ruhe also habe ich es heute früh nochmal probiert.
Ich nutze meinen PC (Ryzen CPU) mit Ubuntu 18 lts. Ich habe apt update und upgrade vorher ausgeführt.

Ich hab Schritt für Schritt nach deiner Anleitung gehandelt aber ich komme zu keinem brauchbaren Ergebnis.
(Schritt 7) Die MAC Adresse vom NEEY habe ich mir in Ubuntu anzeigen lassen und in der yaml eingetragen .
(Schritt 8) Das Erstellen der secrets.yaml ist auch klar.
(Schritt 10) Die secrets.yaml wird aber von ESPHOME später nicht akzeptiert, das WIFI Passwort soll laut Fehlermeldung in Anführungszeichen angegeben werden. Ändere ich das scheitert es daran das 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch2' nicht gefunden wird.

Auf einem Laptop habe ich es auch probiert, das ist exakt das selbe Problem. Ich hab auch meinen Sohn gebeten das ganze mal auf seinem PC zu machen, der hat aber genau das selbe Problem. Er nutzt auch Ubuntu auf seinem Intel PC.
Ich würde meinen das du noch irgendwas zusätzlich installiert hast oder eine andere ESPHOME Version nutz damit das funktioniert.

syssi, 2023-05-09:
I will spin up a ubuntu 18.04 machine and give it a try. Don't give up! You are almost there!

syssi, 2023-05-09:
The most recent ESPHome release requires Python >=3.9. Please use Ubuntu 22 and try again.

moscito1010, 2023-05-09:
ok, i switched to version 22. Now it's working.
My ESP hardware is az-delivery-devkit-v4 and esp32cam.
ESP32CAM doesn't want to work that well but that's no big deal, I have a couple of the az-delivery-devkit-v4.

Now comes task 2 :-)
How can I connect 2 NEEYś or more to one ESP. I have a total of 5 LFP 16S battery packs.
The ESP should be fine with more than one connection, with another esp32 I connected 5 BLE heating valves, that works great.

syssi, 2023-05-09:
Do you know this issue? https://github.com/syssi/esphome-jk-bms/issues/310#issuecomment-1512687283

moscito1010, 2023-05-09:
I read it but don't really believe it. 
I have 5 eq3 BLE heater valves that I connected to an ESP32. This has been working for 2 years without any problems.
Here is the link to the project
https://github.com/softypit/esp32_mqtt_eq3

Here my screenshot
[image]

syssi, 2023-05-09:
Hmm... I'm talking about this limitation: https://esphome.io/components/ble_client.html#ble-client

I guess the `esp32_mqtt_eq3` project doesn't maintain simultaneous BLE connections. Receiving passive advertisements doesn't does not count here either. May be you can talk sequentially to all devices using a single ESP32.

moscito1010, 2023-05-09:
ok, this is probably sufficient for a heating valve.

I took over the multi ble codes and applied them to the remaining sections according to this scheme. This also works so far, I just don't understand the "interval" section.

Here is the result of my "copying"

`substitutions:
  name: heltec-balancer
  device0: NEEY1
  device1: NEEY2
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 00:E4:4C:xx:xx:xx
  mac_address2: 00:E4:4C:xx:xx:xx

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

mqtt:
  broker: 192.168.178.48
  username: pi
  password: xxxxxxxxx
#  id: 

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1

heltec_balancer_ble:
[... 357 lines cut ...]
    buzzer_mode:
      name: "${device0} buzzer mode"
    battery_type:
      name: "${device0} battery type"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY2
    # Not implemented
    # errors:
    #   name: "${device1} errors"
    operation_status:
      name: "${device1} operation status"
    total_runtime_formatted:
      name: "${device1} total runtime formatted"
    buzzer_mode:
      name: "${device1} buzzer mode"
    battery_type:
      name: "${device1} battery type"

interval:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY1
    interval: "${device0} 30min"
    then:
      - button.press: retrieve_settings_button
`

syssi, 2023-05-14:
Could you rephrase your last question? What's the detail you are trying to improve? If you want to reduce the amount of traffic passed to the MQTT server you should use the `throttle` option.

Oh, I got it. I will provide a proper `interval` section later the day. The `interval` key accepts time intervals only like `10min` or `3h`. The `${deivce0} ` variable/substitution is wrong here. As next step the two `retrieve settings buttons` must get a unique identifier.

syssi, 2023-05-14:
```
substitutions:
  name: heltec-balancer
  device0: NEEY1
  device1: NEEY2
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 00:E4:4C:xx:xx:xx
  mac_address2: 00:E4:4C:xx:xx:xx

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

mqtt:
  broker: 192.168.178.48
  username: pi
  password: xxxxxxxxx
#  id: 

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 5s
    id: NEEY1
  - ble_client_id: client1
    throttle: 5s
    id: NEEY2

button:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY1
    retrieve_settings:
      name: "${device0} retrieve settings"
      id: neey0_retrieve_settings_button
    retrieve_device_info:
      name: "${device0} retrieve device info"
    retrieve_factory_defaults:
      name: "${device0} retrieve factory defaults"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY2
    retrieve_settings:
      name: "${device1} retrieve settings"
      id: neey1_retrieve_settings_button
    retrieve_device_info:
      name: "${device1} retrieve device info"
    retrieve_factory_defaults:
      name: "${device1} retrieve factory defaults"

interval:
  - interval: 30min
    then:
      - button.press: neey0_retrieve_settings_button
      - button.press: neey1_retrieve_settings_button
```

closed 2023-05-14 by syssi

moscito1010, 2023-05-14:
that doesn't seem to have been the case.
Here the error message

`andy@andy:~/esphome$ ~/.local/bin/esphome run esp32-heltec-balancer-ble-example.yaml
INFO Reading configuration esp32-heltec-balancer-ble-example.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
Failed config

interval: [source esp32-heltec-balancer-ble-example.yaml:381]
  - interval: 30min
    then: 
      - button.press: 
          
          Couldn't find ID 'neey0_retrieve_settings_button'. Please check you have defined an ID with that name in your configuration.
          id: neey0_retrieve_settings_button
      - button.press: 
          
          Couldn't find ID 'neey1_retrieve_settings_button'. Please check you have defined an ID with that name in your configuration.
          id: neey1_retrieve_settings_button
andy@andy:~/esphome$ ^C
andy@andy:~/esphome$ 
`

syssi, 2023-05-14:
Did you update the two button entities above? Please make sure to assign `id`s here. Please provide your full YAML if you are unsure.

moscito1010, 2023-05-14:
here the code
`substitutions:
  name: heltec-balancer
  device0: NEEY0
  device1: NEEY1
#  device2: NEEY2
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 00:E4:4C:xx:xx:xx
  mac_address2: 00:E4:4C:xx:xx:xx

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: az-delivery-devkit-v4
#  board: esp32cam
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

mqtt:
  broker: 192.168.178.48
  username: pi
  password: XXXXXXXXX
#  id: 

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 15s
    id: NEEY0
[... 288 lines cut ...]
    total_runtime_formatted:
      name: "${device0} total runtime formatted"
    buzzer_mode:
      name: "${device0} buzzer mode"
    battery_type:
      name: "${device0} battery type"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY1
    # Not implemented
    # errors:
    #   name: "${device1} errors"
    operation_status:
      name: "${device1} operation status"
    total_runtime_formatted:
      name: "${device1} total runtime formatted"
    buzzer_mode:
      name: "${device1} buzzer mode"
    battery_type:
      name: "${device1} battery type"

interval:
  - interval: 30min
    then:
      - button.press: neey0_retrieve_settings_button
      - button.press: neey1_retrieve_settings_button`

syssi, 2023-05-15:
Please add these lines to your config:

```
button:
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY1
    retrieve_settings:
      name: "${device0} retrieve settings"
      id: neey0_retrieve_settings_button        # <-----------------
    retrieve_device_info:
      name: "${device0} retrieve device info"
    retrieve_factory_defaults:
      name: "${device0} retrieve factory defaults"
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: NEEY2
    retrieve_settings:
      name: "${device1} retrieve settings"
      id: neey1_retrieve_settings_button        # <-----------------
    retrieve_device_info:
      name: "${device1} retrieve device info"
    retrieve_factory_defaults:
      name: "${device1} retrieve factory defaults"
```

physi, 2023-05-16:
I had the same problem, and disable the 
`api:`
Section solves it, now I see values for 2 Neey in MQ

moscito1010, 2023-05-16:
the just api: commented out?

#api:
?

2 things struck me.
I don't need to press boot button to flash, but only when I flash the yaml file for 2 BMS. If I use the "single" script I have to press the button.
You cannot change the battery type in Homeassistant.

physi, 2023-05-17:
Yes,
# api:
, but the you got no homeassistant  "things"  .
But that was unstable too, the esp lost contact to both neey this night and does not report anything anymore. Before I had only 1 neey and all worked perfectly good.
So I will invest another esp for each neey, that can than also monitor the temperature of each block.

Or if you get it stable up and running, pleas share your code :-)

syssi, 2023-05-17:
`api` enabled or not shouldnt make a difference in general. It looks like a resource issue. Please try to free some resources by removing unimportant/unused components and/or entities.

physi, 2023-05-17:
tried it with only the voltages from 1-16 on 2 Need, does not work, maybe to much to work for one poor Espresso :-)
Will use separate Espresso for each bank, that works.
Thanks for your nice work on this.

Elektrofix-OL, 2023-11-03:
I have the same problem and have now gone back to ESPHome version 2023.5.4, but after an indefinite period of time (3 hours or 7 days) I no longer receive any data.
My Yamel: `substitutions:
  name: speicher
  device0: akku1
  device1: akku2
  device_description: "Monitor and control a Heltec/NEEY 4A balancer via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: 00:e4:4c:21:d9:cd
  mac_address2: 3c:a5:51:85:e8:4c

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.4.0

esp32:
  board: esp32dev
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

logger:
  level: DEBUG
  
# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
#api:
#  encryption:
#    key: "q+W4YeOHnu8kOBLz0LYw7uUlWzzRE+NMk6CpfZE856k="

mqtt:
   broker: !secret mqtt_host
   username: !secret mqtt_username
   password: !secret mqtt_password
   id: mqtt_client

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("GW-", 0) == 0) {
            ESP_LOGI("ble_adv", "New Heltec/NEEY balancer found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
  - mac_address: ${mac_address2}
    id: client1

heltec_balancer_ble:
  - ble_client_id: client0
    throttle: 30s
    id: bms0
[... 296 lines cut ...]
  - platform: heltec_balancer_ble
    heltec_balancer_ble_id: bms1
    # Not implemented
    # errors:
    #   name: "errors ${device1}"
    operation_status:
      name: "operation status ${device1}"
    total_runtime_formatted:
      name: "total runtime formatted ${device1}"
    buzzer_mode:
      name: "buzzer mode ${device1}"
    battery_type:
      name: "battery type ${device1}"

interval:
  - interval: 30min
    then:
      - button.press: retrieve_settings_button0
      - button.press: retrieve_settings_button1
      `
The displays in the LOG:
[image]
Before time to time comes:
[image]
If I unplug the ESP32 and restart it, everything is OK. I've already changed the board, no success. What am I doing wrong?

syssi, 2023-11-04:
@Elektrofix-OL Could you create a new issue please?