syssi/esphome-jk-bms #331: JK BMS BLE Protocol

EstyGee, 2023-05-22:
Hello, it's not about [esphome-jk-bms], but it's about reading daa drom JK BMS by my app...

Now i have this:
[image]

I send to BMS 2 command:
0xffe1:
aa5590eb9700df5288679d0a096b9af6709a17fd
aa5590eb9600f45fe2036dfbe038aabc441234b8

After that i start to recieve data each second, but I was able to decrypt some data. The main thing I need to find is the charge current and the discharge current. Help me please...

First pair: Data from Bluetoothe and after 10 seconds screenshot of app:
[image]
[image]

Second pair: Data from Bluetoothe and after 10 seconds screenshot of app:
[image]
[image]

My English not so good, but I hope you will understood me :)

EstyGee, 2023-05-22:
Oh, I think I found charge current and charge power?
[image]

But where is discharge?

EstyGee, 2023-05-22:
Ok, solved... it can be negative... Thank for everyone (no one) :)

closed 2023-05-22 by EstyGee

syssi, 2023-05-22:
You are on the right track! Keep in mind there are different time zones and everybody needs some rest. :-)

syssi, 2023-05-22:
https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L564-L578