syssi/esphome-jk-bms #336: Reboot AT startup

JonathanJ71, 2023-06-04:
Please help.
When i'm connected to the bms my esp32 reboot all thé Time AT stasting ...
Any Idea ? IS somebody add this issue?
Thanks for help.

syssi, 2023-06-04:
Could you describe your setup and explain the issue in detail? Could you provide some logs?

syssi, 2023-06-25:
Could you provide a photo of your wiring?

hemskgren, 2023-07-31:
I had the same problem with d1_mini (v4), it  looked good in multi-meter and oscilloscope but no response in the debug log!?
I notice in the newer bms software version it exist an option for 'Device Address' in settings and was thinking if that can have an impact!
Insted of explore that I switched to esp32 and it worked instantly!
refernce https://github.com/syssi/esphome-jk-bms/blob/main/esp8266-example-debug.yaml#L13C1-L14C20

closed 2023-08-09 by syssi