syssi/esphome-jk-bms #337: One unit on BLE starts reporting 0 on Delta, Total Voltage and a few others but fine with others and Bluetooth app fine

JohnGalt1717, 2023-06-08:
So I just switched out a BMS because of weird behavior that just started.  If you restart the device by powering it off fully, and then start it up, it will report everything fine. But then after a few minutes it will stop reporting all of the values. The ones above show 0 whereas each cell etc. are fine.

Switched to a newer unit and it is fine with the exact same settings, just mac change.

Not sure that this is this project, but given that the JK BMS iOS app shows everything fine, I gotta believe that it's related?

syssi, 2023-06-08:
Could you check the software and hardware version of your BMS and provide a log of the ESPHome node. I guess the `protocol_version` isn't pinned and flips between the different frame formats.

syssi, 2023-06-10:
@JohnGalt1717 Friendly reminder.

JohnGalt1717, 2023-06-10:
Sorry. Day job. Will put it back in and pull the data today.

JohnGalt1717, 2023-06-11:
[image]

Before [16S] - [Boot]
Before [2M22S] - [Boot]
Before [2M26S] - [Button to turn it off]
Before [1H46M51S] - [Boot]
Before [1H46M55S] - [APP open discharge]
Before [1H46M55S] - [APP close discharge]
Before [1H46M56S] - [APP open charge]
Before [1H46M57S] - [APP close charge]
Before [2H1M32S] - [APP open discharge]
Before [2H1M32S] - [APP close discharge]
Before [2H1M36S] - [APP open charge]
Before [2H1M46S] - [APP close charge]
Before [4H28M32S] - [Boot]
Before [4H28M36S] - [Button to turn it off]
Before [4H32M25S] - [APP open charge]
Before [4H32M36S] - [APP close charge]
Before [4H32M37S] - [APP open discharge]
Before [4H33M12S] - [APP close discharge]
Before [4H33M17S] - [APP open charge]
Before [4H33M19S] - [APP close charge]
Before [5D21H19M59S] - [Cell overcharge protection is released]
Before [5D21H23M16S] - [Cell 01 over charge protection]
Before [6D22H37M45S] - [Cell overcharge protection is released]
Before [6D22H38M13S] - [Cell 03 over charge protection]
Before [6D22H38M49S] - [Cell overcharge protection is released]
Before [6D22H39M17S] - [Cell 03 over charge protection]
Before [6D22H39M52S] - [Cell overcharge protection is released]
Before [6D22H40M10S] - [Cell 03 over charge protection]
Before [6D22H42M54S] - [Cell overcharge protection is released]
Before [6D22H43M2S] - [Cell 03 over charge protection]
Before [6D22H59M34S] - [Cell overcharge protection is released]
Before [6D23H0M2S] - [Cell 01 over charge protection]
Before [6D23H0M23S] - [Cell overcharge protection is released]
Before [6D23H0M48S] - [Cell 01 over charge protection]
Before [6D23H1M2S] - [Cell overcharge protection is released]
Before [6D23H1M7S] - [Cell 01 over charge protection]
Before [7D22H15M49S] - [Cell overcharge protection is released]
Before [7D22H16M24S] - [Cell 01 over charge protection]
Before [7D22H50M37S] - [Cell overcharge protection is released]
Before [7D22H52M29S] - [Cell 01 over charge protection]
Before [7D23H11M59S] - [Cell overcharge protection is released]
Before [7D23H14M48S] - [Cell 01 over charge protection]
Before [7D23H15M16S] - [Cell overcharge protection is released]
Before [7D23H15M21S] - [Cell 01 over charge protection]
Before [7D23H16M37S] - [Cell overcharge protection is released]
Before [7D23H18M20S] - [Cell 01 over charge protection]
Before [7D23H28M24S] - [Cell overcharge protection is released]
Before [7D23H30M10S] - [Cell 01 over charge protection]
Before [7D23H31M49S] - [Cell overcharge protection is released]
Before [7D23H33M34S] - [Cell 01 over charge protection]

JohnGalt1717, 2023-06-11:
I don’t have it hooked up to esp anymore. I can do so if you need it.

syssi, 2023-06-11:
Please pin to protocol version to `JK02_32S` + update the ESP and try again:

```
substitutions:
  # ...
  # Defaults to "JK02" (hardware version >= 6.0 and < 11.0)
  # Please use "JK02_32S" if you own a new JK-BMS >= hardware version 11.0 (f.e. JK-B2A8S20P hw 11.XW, sw 11.26)
  # Please use "JK04" if you have some old JK-BMS <= hardware version 3.0 (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02_32S
```

I guess this will solve your issue.

JohnGalt1717, 2023-06-11:
Why did it all of a sudden freak out?  It has been working for months until I updated to the latest esp home version and even that for a week or more.

syssi, 2023-06-11:
Because I've implemented an auto-detection in the past to guess the correct protocol version but it turns out my auto-detection wasn't a good idea / isn't stable. If some specific bytes of the new frame looks like the old frame the protocol version gets changed to the old format.

I will remove this auto-detection in one of the next releases an the manual selecftion of the proper protocol version will be necessary. See

https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#L450-L459

closed 2023-06-21 by JohnGalt1717