syssi/esphome-jk-bms #344: B2a8s20p ble not all sensor correct

hemskgren, 2023-06-23:
Hi 
My b2a8s20p look to match one existing confirm to work. Using JK02_32S.

[image]

* Cell volt is correct but all resistance is 0, same as manual in app.
* all 'tempeture' is off/wrong
* all 'Total' look like static and wrong.
* 'min cell voltage' look to show delta. 

More is wrong but above is easy to see.

I have test uart but not get it to work thru the 4 pin gps port.

syssi, 2023-06-23:
If it's a version 9 device you should use `protocol_version:  JK02`. Please give it another try!

hemskgren, 2023-06-27:
I will test it during the weekend.

hemskgren, 2023-07-02:
Yes, it look OK with JK02

closed 2023-07-02 by hemskgren