syssi/esphome-jk-bms #346: JK-BA1A8S10P BLE ESP32 config.

Chickenbreast0, 2023-07-05:
Hi Syssi,

I just wanted to inform you, that the BLE ESP32 configuration works with two JK-BA1A8S10P BMS devices.
I am using theJK02_32S protocol. HW ID: V11.XW, SW ID: 11.261. 

Cheers

syssi, 2023-07-05:
Thanks for your feedback! Nice to see you here!

syssi, 2023-07-05:
Could you double check your device model name? It should be `JK-B1A8S10P`, right?

closed 2023-07-05 by syssi

Chickenbreast0, 2023-07-10:
FYI for further questions:
I discovered the BLE MAC ID using a PlayStore app called BLE Scanner on my mobile. The app scans for all BT devices in the environment and also shows the MAC ID of the devices.

mrmaximas, 2025-06-24:
@Chickenbreast0 can you show your config? thanks!

syssi, 2025-06-24:
@mrmaximas Which hardware/firmware version do you own? If it's v11 too you should use this configuration:

https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v11-example.yaml

mrmaximas, 2025-06-25:
@syssi hi! thanks for you hard work. for now i'm testing JK-B1A8S10P via BLE
JK_B1A8S10P
Hardware Ver: V15H
Software Ver: V15.38
Seems to be working, I'll post my results a little later. For now I'm thinking how to eliminate sending the debug topic to mqtt (it's an unnecessary entity for me).

syssi, 2025-06-25:
Please use the `v15` example for your device: https://github.com/syssi/esphome-jk-bms/blob/main/esp32-ble-v15-example.yaml

You can disable the log topic:

```
mqtt:
  log_topic: null
```

syssi, 2025-06-25:
If you don't use Home Assistant you could disable the auto-discovery messages too:

```
mqtt: 
  discovery: false
```

mrmaximas, 2025-06-25:
@syssi 🙏🏻

mrmaximas, 2025-06-25:
@syssi  i have "ERROR No remote or local logging method configured (api/mqtt/logger) with this config"

logger:
  level: INFO
  logs:
    esp32_ble_tracker: INFO
    esp32_ble_client: INFO

mqtt:
    broker: 192.168.8.10
    id: mqtt_client
    log_topic: null
    discovery: false

mrmaximas, 2025-06-25:
@syssi thanks, trying using api with the

```
api:
  reboot_timeout: 0
```

Failed config

api: [source /config/bms.yaml:47]
  
  Don't know what '0' means as it has no time *unit*! Did you mean '0s'?.
  reboot_timeout: 0

compiled and running successfully with
```
api:
  reboot_timeout: 0s
```
thanks!
p.s. it is possible to use BLE in observation only mode? I don't aim to control bms, I'm only interested in data collection to influxdb then grafana.

mrmaximas, 2025-06-25:
@syssi I don't want to create a separate post, but can you please tell me why  jk-bms/sensor/jk-bms_balancing/state topic has three values 0, 1 and 2? What is the difference?

[image]

mrmaximas, 2025-06-26:
@syssi thanks!  
jk-bms/sensor/jk-bms_balancing/state topic has three values 0, 1 and 2 >> this is balancing current direction.

PR #347 (2023-07-05): Add JK-B1A8S10P, hw 11.xw, sw 11.261 as supported
Closes: #346