syssi/esphome-jk-bms #348: Not showing right values

MrYoshii, 2023-07-10:
Hi i have 3 BMS and the BLE ESP32.
Yesterday a low temp alarm went off after i checked i saw that BMS2 had 0C.
But not only temp was at 0. A lot of other values are also zero or wrong. 

[image]

I tried restart everything but nothing helped.
I checked the log from the ESP32 and there are also wrong values.
`[09:28:06][D][jk_bms_ble:646]: Unknown168: 00.00.00.00.00.02.FB.1D.00.00.C0.45.04.00.3C.00.00.00.5A.F6.00.01.64.00.00.00.B1.05.F6.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (46)
[09:28:06][D][jk_bms_ble:660]: Unknown189: 01 64
[09:28:06][D][jk_bms_ble:664]: Unknown192: 00
[09:28:06][D][jk_bms_ble:666]: Unknown193: 00 B1 (0x00 0x8D)
[09:28:06][D][jk_bms_ble:668]: Unknown195: 05 F6 (0x21 0x40)
[09:28:11][D][sensor:127]: 'battery2 cell voltage 1': Sending state 3.31800 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 1': Sending state 0.00000 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 2': Sending state 3.31800 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 2': Sending state 0.00000 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 3': Sending state 3.31600 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 3': Sending state 0.00000 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 4': Sending state 3.31600 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 4': Sending state 0.25500 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 5': Sending state 3.31600 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 5': Sending state 0.00000 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 6': Sending state 3.31800 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 6': Sending state 3.31800 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 7': Sending state 3.31500 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 7': Sending state 0.00500 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell voltage 8': Sending state 3.31500 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 cell resistance 8': Sending state 0.00600 Ω with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 min cell voltage': Sending state 3.31500 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 max cell voltage': Sending state 3.31800 V with 3 decimals of accuracy
[09:28:11][I][jk_bms_ble:516]: Enabled cells: 00 00 00 00
[09:28:11][D][sensor:127]: 'battery2 average cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 delta cell voltage': Sending state 0.00000 V with 3 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 max voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 min voltage cell': Sending state 1.00000  with 0 decimals of accuracy
[09:28:11][D][jk_bms_ble:540]: Unknown112: 00 00
[09:28:11][I][jk_bms_ble:545]: Wire resistance warning: 00 00 00 00
[09:28:11][D][sensor:127]: 'battery2 total voltage': Sending state 0.00000 V with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 current': Sending state 0.00000 A with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 power': Sending state 0.00000 W with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 charging power': Sending state 0.00000 W with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 discharging power': Sending state 0.00000 W with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 temperature sensor 1': Sending state 0.00000 °C with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 temperature sensor 2': Sending state 0.00000 °C with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 power tube temperature': Sending state 0.00000 °C with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[09:28:11][D][text_sensor:067]: 'battery2 errors': Sending state ''
[09:28:11][D][sensor:127]: 'battery2 balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 capacity remaining': Sending state 17629.18555 Ah with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 total battery capacity setting': Sending state 0.00000 Ah with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 charging cycles': Sending state 26543.00000  with 0 decimals of accuracy
[09:28:11][D][sensor:127]: 'battery2 total charging cycle capacity': Sending state 259.66800 Ah with 3 decimals of accuracy
[09:28:11][D][jk_bms_ble:630]: Unknown158: 37 26 (always 0x64 0x00?)
[09:28:11][D][jk_bms_ble:633]: Unknown160: 00 00 (always 0xC5 0x09?)
[09:28:11][D][sensor:127]: 'battery2 total runtime': Sending state 15335656.00000 s with 0 decimals of accuracy
[09:28:12][D][text_sensor:067]: 'battery2 total runtime formatted': Sending state '177d 11h'
[09:28:12][D][jk_bms_ble:646]: Unknown168: 00.00.00.00.00.02.0A.1E.00.00.C0.45.04.00.3C.00.00.00.5A.F6.00.01.64.00.00.00.B6.05.F6.00.01.01.00.00.00.00.00.00.00.00.00.00.00.00.00.00 (46)
[09:28:12][D][jk_bms_ble:660]: Unknown189: 01 64
[09:28:12][D][jk_bms_ble:664]: Unknown192: 00
[09:28:12][D][jk_bms_ble:666]: Unknown193: 00 B6 (0x00 0x8D)
[09:28:12][D][jk_bms_ble:668]: Unknown195: 05 F6 (0x21 0x40)`
When i direct connect my phone to the bms the values are ok, 
I tryed to update the esp but i just get errors.
`INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/battery-2.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.0.9
INFO Compiling app...
Processing battery2 (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.0.1)
-- Project sdkconfig file /config/esphome/.esphome/build/battery2/sdkconfig.battery2
-- Configuring incomplete, errors occurred!
See also "/data/battery2/.pioenvs/battery2/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /data/cache/platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 3.90 seconds ==========================`
I'm not sure what's going on. If you need more, let me know. 
Thanks in advance

syssi, 2023-07-10:
Could you provide the hardware and firmware version of the devices? I assume you should pin the protocol version. The build issue is a issue of the ESPHome dashboard/ docker setup. The different build and cache directories are distributed across different filesystems which isn't supported by the build system. You could avoid this issue if you install ESPHome manually on a second (x86) host & compile+flash the device using the command line:

```
esphome run battery.yaml
```

MrYoshii, 2023-07-10:
Firmware: 2022.12.3 (Jan 4 2023, 18:55:21)
Hardware: 1.3.0 
AZDelivery NodeMCU ESP32 S

syssi, 2023-07-13:
My question was about the hardware and firmware version of your JK-BMS.

MrYoshii, 2023-07-14:
oh sry
[image]
It worked for at least 7 Months now.

MrYoshii, 2023-07-15:
also i can not get it to reflash...
i have now tryed esphome on windows and linux debian and i still get the same error??

```
INFO ESPHome 2023.6.5
INFO Reading configuration battery-2.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.0.9
INFO Compiling app...
Processing battery2 (board: nodemcu-32s; framework: espidf; platform: platformio/espressif32@5.3.0)
--------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50001.0 (5.0.1)
 - tool-cmake @ 3.16.4
 - tool-ninja @ 1.7.1
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (5.0.1)
-- Project sdkconfig file /home/login/esphome-jk-bms/.esphome/build/battery2/sdkconfig.battery2
-- Configuring incomplete, errors occurred!
See also "/home/login/esphome-jk-bms/.esphome/build/battery2/.pioenvs/battery2/CMakeFiles/CMakeOutput.log".

fatal: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git
CMake Error at /home/login/.platformio/packages/framework-espidf/tools/cmake/tool_version_check.cmake:36 (message):

  Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
  /home/login/.platformio/packages/toolchain-xtensa-esp32/bin/xtensa-esp32-elf-gcc

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  /home/login/.platformio/packages/framework-espidf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)
  /home/login/.platformio/packages/framework-espidf/tools/cmake/build.cmake:380 (include)
  /home/login/.platformio/packages/framework-espidf/tools/cmake/build.cmake:599 (__build_process_project_includes)
  /home/login/.platformio/packages/framework-espidf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:3 (project)

======================================================== [FAILED] Took 3.60 seconds ========================================================
```

syssi, 2023-07-15:
Could you provide your complete YAML? I guess the IDF platform is pinned to `latest`. Please use `recommended` instead or remove the line.

MrYoshii, 2023-07-16:
I did change that IDF platform thing and now I could reflash the ESP but still same issue.
Here my YAML:
```
substitutions:
  name: battery2
  device_description: "Monitor and control a JK-BMS via bluetooth"
  external_components_source: github://syssi/esphome-jk-bms@main
  mac_address: C8:47:8C:E5:5E:BC
  # Defaults to "JK02". Please use "JK04" if you have some old JK-BMS version (f.e. JK-B2A16S hw 3.0, sw. 3.3.0)
  protocol_version: JK02

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-jk-bms"
    version: 1.3.0

esp32:
  board: nodemcu-32s
  framework:
#  The arduino platform crashs
#    type: arduino
#    version: latest
    type: esp-idf
    version: recommended

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:

# The MQTT component is ESP-IDF compatible since ESPHome version 2022.4.0. If
# ESPHome suggests to use the arduino framework instead because of missing ESP-IDF
# framework support you should update your setup.
# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

esp32_ble_tracker:

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 5s
    id: bms0

binary_sensor:
  - platform: jk_bms_ble
    balancing:
      name: "${name} balancing"
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

button:
  - platform: jk_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    retrieve_device_info:
      name: "${name} retrieve device info"

number:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    balance_trigger_voltage:
      name: "${name} balance trigger voltage"
    cell_count:
      name: "${name} cell count"
    total_battery_capacity:
      name: "${name} total battery capacity"
    cell_voltage_overvoltage_protection:
      name: "${name} cell voltage overvoltage protection"
    cell_voltage_overvoltage_recovery:
      name: "${name} cell voltage overvoltage recovery"
    cell_voltage_undervoltage_protection:
      name: "${name} cell voltage undervoltage protection"
    cell_voltage_undervoltage_recovery:
      name: "${name} cell voltage undervoltage recovery"
    balance_starting_voltage:
      name: "${name} balance starting voltage"
    voltage_calibration:
      name: "${name} voltage calibration"
    current_calibration:
      name: "${name} current calibration"
    power_off_voltage:
      name: "${name} power off voltage"
    max_balance_current:
      name: "${name} max balance current"
    max_charge_current:
      name: "${name} max charge current"
    max_discharge_current:
      name: "${name} max discharge current"

sensor:
  - platform: jk_bms_ble
    jk_bms_ble_id: bms0
    min_cell_voltage:
      name: "${name} min cell voltage"
    max_cell_voltage:
      name: "${name} max cell voltage"
    min_voltage_cell:
      name: "${name} min voltage cell"
    max_voltage_cell:
      name: "${name} max voltage cell"
    delta_cell_voltage:
      name: "${name} delta cell voltage"
    average_cell_voltage:
      name: "${name} average cell voltage"
    cell_voltage_1:
      name: "${name} cell voltage 1"
    cell_voltage_2:
      name: "${name} cell voltage 2"
    cell_voltage_3:
      name: "${name} cell voltage 3"
    cell_voltage_4:
      name: "${name} cell voltage 4"
    cell_voltage_5:
      name: "${name} cell voltage 5"
    cell_voltage_6:
      name: "${name} cell voltage 6"
    cell_voltage_7:
      name: "${name} cell voltage 7"
    cell_voltage_8:
      name: "${name} cell voltage 8"
    cell_resistance_1:
      name: "${name} cell resistance 1"
    cell_resistance_2:
      name: "${name} cell resistance 2"
    cell_resistance_3:
      name: "${name} cell resistance 3"
    cell_resistance_4:
      name: "${name} cell resistance 4"
    cell_resistance_5:
      name: "${name} cell resistance 5"
    cell_resistance_6:
      name: "${name} cell resistance 6"
    cell_resistance_7:
      name: "${name} cell resistance 7"
    cell_resistance_8:
      name: "${name} cell resistance 8"
    total_voltage:
      name: "${name} total voltage"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    charging_power:
      name: "${name} charging power"
    discharging_power:
      name: "${name} discharging power"
    temperature_sensor_1:
      name: "${name} temperature sensor 1"
    temperature_sensor_2:
      name: "${name} temperature sensor 2"
    power_tube_temperature:
      name: "${name} power tube temperature"
    state_of_charge:
      name: "${name} state of charge"
    capacity_remaining:
      name: "${name} capacity remaining"
    total_battery_capacity_setting:
      name: "${name} total battery capacity setting"
    charging_cycles:
      name: "${name} charging cycles"
    total_charging_cycle_capacity:
      name: "${name} total charging cycle capacity"
    total_runtime:
      name: "${name} total runtime"
    balancing_current:
      name: "${name} balancing current"
    errors_bitmask:
      name: "${name} errors bitmask"

switch:
  - platform: jk_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    balancer:
      name: "${name} balancer"

  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

  - platform: gpio
    name: "Battery_2 Fan"
    pin: 16
    inverted: False
    restore_mode: RESTORE_DEFAULT_OFF

text_sensor:
  - platform: jk_bms_ble
    errors:
      name: "${name} errors"
    total_runtime_formatted:
      name: "${name} total runtime formatted"
```

syssi, 2023-07-16:
Please change this setting:

```
# from
substitutions:
  protocol_version: JK02

# to
substitutions:
  protocol_version: JK02_32S
```

And reflash your ESP again. This should fix the issue!

MrYoshii, 2023-07-16:
Thanks, that fixed the issue. But I'm not sure why it randomly stopped working?

syssi, 2023-07-16:
I tried to auto-detect the protocol version but the mechanism doesn't work reliable. I will remove the auto-detection soon.

closed 2023-07-16 by syssi