syssi/esphome-jk-bms #349: 8 cells minimum is wrong

pedro042, 2023-07-13:
Hi, I've BD4A8S4P which supports 4-8 cells. In settings is possible to configure min. 8 cells. It's wrong. Please don't limit count of cells to 8 or more.
The BMS works well, I set count of cells in the Android app.
Thanks.

syssi, 2023-07-13:
Thanks for your report. I will push a fix!

closed 2023-07-13 by syssi

pedro042, 2023-07-13:
Thanks; actually this BMS can handle 4 LifePo4 (minimum, because of the voltage needed to power the BMS) or 3 Li-Ion :-) (3 Li-Ion cells can power this BMS). Maybe one more limit decrease - 4 -> 3
https://www.aliexpress.com/item/1005004590744267.html - see the table, unfortunately as JPG picture only

syssi, 2023-07-13:
Good catch! I will prepare another fix asap.

syssi, 2023-07-16:
Fixed.

commit d8d5bc66 (2023-07-13): Fix cell count limits (Closes: #349) (#350)