syssi/esphome-jk-bms #355: Bluetooth Connection Issue

JohnGalt1717, 2023-07-24:
I have one of my esp32s that has a bms hooked up that works fine with my phone and a second jkbms without issue that gives this in the logs:

```
[10:27:23][D][esp32_ble_tracker:245]: Starting scan...
[10:27:23][D][esp32_ble_client:048]: [0] [C8:47:8C:F7:8A:36] Found device
[10:27:23][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[10:27:23][I][esp32_ble_client:064]: [0] [C8:47:8C:F7:8A:36] 0x00 Attempting BLE connection
[10:27:24][D][esp-idf:000]: W (17273) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[10:27:24][D][esp-idf:000]: W (17494) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[10:27:24][D][esp-idf:000]: W (17587) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[10:27:25][D][esp-idf:000]: W (18124) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x3e

[10:27:25][D][esp-idf:000]: W (18128) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x3e

[10:27:25][D][esp-idf:000]: W (18129) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x3e

[10:27:25][W][esp32_ble_client:134]: [0] [C8:47:8C:F7:8A:36] Connection failed, status=133
```

It will periodically connect, then loses connection, and then never connects again until I restart the esp32.

This worked until recently. And I've tried "recommended" for the esp-idf version, and I've also tried pinning 5.4.0 as the platform version with the same results. Using latest results in errors on build about git.

I also periodically see this in the logs:

```
[10:27:41][W][esp32_ble_client:134]: [0] [C8:47:8C:F7:8A:36] Connection failed, status=133
[10:27:41][D][esp32_ble_tracker:245]: Starting scan...
[10:27:41][W][jk_bms_ble:365]: [C8:47:8C:F7:8A:36] Not connected
[10:27:41][W][jk_bms_ble:365]: [C8:47:8C:E5:B6:C9] Not connected
[10:27:41][D][esp32_ble_client:048]: [1] [C8:47:8C:E5:B6:C9] Found device
[10:27:41][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[10:27:41][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[10:27:41][D][esp-idf:000]: E (35033) BT_BTM: BTM_BleScan scan not active

[10:27:41][D][esp-idf:000]: W (35035) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6
```

When it does connect I get this:

```
[10:28:26][I][jk_bms_ble:1158]: Device info frame (320 bytes) received
[10:28:26][I][jk_bms_ble:1210]:   Vendor ID: JK-B2A24S20P
[10:28:26][I][jk_bms_ble:1211]:   Hardware version: 10.X-W
[10:28:26][I][jk_bms_ble:1212]:   Software version: 10.02
[10:28:26][I][jk_bms_ble:1213]:   Uptime: 40668600 s
[10:28:26][I][jk_bms_ble:1214]:   Power on count: 17
[10:28:26][I][jk_bms_ble:1215]:   Device name: Battery 1
[10:28:26][I][jk_bms_ble:1216]:   Device passcode: 1234
[10:28:26][I][jk_bms_ble:1217]:   Manufacturing date: 220403
[10:28:26][I][jk_bms_ble:1218]:   Serial number: 1120303195
[10:28:26][I][jk_bms_ble:1219]:   Passcode: 0000
[10:28:26][I][jk_bms_ble:1220]:   User data: Input Userdata
[10:28:26][I][jk_bms_ble:1221]:   Setup passcode: 1717
[10:28:27][I][jk_bms_ble:891]: Settings frame (320 bytes) received
```

Which appears to use the default version for communications according to the docs. (JK_02). Anything else results in weird responses.

I'm also seeing periodic restarts of the esp32s occurring too on this one and a second esp32s that I have running for 2 more bmses.  Is this related?

syssi, 2023-07-24:
Which ESPHome release do you use? Please make sure the Android app isn't connected / force closed if you try to talk to the BMS using the ESP. The bluetooth module is able to handle a single connection only. Are you sure the ESP is close enough to the BMS?

JohnGalt1717, 2023-07-24:
2023.7.0 of ESPHome.

No android or ios app running at all and my idevices aren't in range.

The ESP32s are 3 feet away with direct line of sight and worked fine until very recently.

syssi, 2023-07-24:
Hmm..  I've no ideas. I cannot update my setup to ESPHome 2023.7.0 at the moment.

hemskgren, 2023-07-25:
+1 on this, I was running esphome 2023.3.2 before and suddenly after some days the second bms did not connect. Use to do reload and if that not solve the problem power of for 20 seconds. As it did not fix the problem I try latest esphome 2023.7.0 but still same problem.  Then I found this open issue and remember when I have done some BMS test before on esphome 2023.6.x and got problem with second BMS after esphome minor release update.  
I tested to downgrade per release and it start to work on esphome 2023.6.4 again.

jouking, 2023-07-26:
ok, so my problem is detected.
We will have to wait for you to update to version 2023.7.X to fix it.
Thank you.

syssi, 2023-07-26:
@jouking I don't if the problem has already been resolved upstream.

jouking, 2023-07-26:
> @jouking I don't if the problem has already been resolved upstream.

But what is the solution? to downgrade?

I understood that you were going to update to 2023.7.X to solve the problem.

syssi, 2023-07-26:
For now downgading ESPHome to version 2023.6.4 is a confirmed workaround.

JohnGalt1717, 2023-07-26:
Based on that information, It must be this issue that was fixed with 2023.6.5 that caused it:

Fix bulk and single Bluetooth parser coexistence [esphome#5073](https://github.com/esphome/esphome/pull/5073) by [@bdraco](https://github.com/bdraco)

JohnGalt1717, 2023-07-26:
I've created a new ESPHome issue for it: https://github.com/esphome/issues/issues/4747

jouking, 2023-08-01:
Updated Esphome to version 2023.7.1 and the problem is still the same.

syssi, 2023-08-10:
I've updated my setup today (without applying the suggested `sdkconfig_options` yet):

```
$ esphome run attic-bms-ble.yaml
INFO ESPHome 2023.7.1
INFO Reading configuration attic-bms-ble.yaml...
INFO Updating https://github.com/syssi/esphome-jk-bms.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing attic-bms-ble (board: wemos_d1_mini32; framework: espidf; platform: platformio/espressif32@5.3.0)
```

Let's see what happens. :-)

jouking, 2023-08-10:
I have the same problem with the BT configuration but on another ESP32 Ethernet. It also fails me and this one fails me more times than the BT.

I have this configuration, any ideas to improve communication?:

esphome:
  name: wt32
  platform: ESP32
  board: esp-wrover-kit

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxx"

ota:
  password: "xxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Optional manual IP
  manual_ip:
    static_ip: 192.168.1.172
    gateway: 192.168.1.254
    subnet: 255.255.255.0   

#captive_portal:
  
external_components:
  - source: github://syssi/esphome-jk-bms@main
    refresh: 0s

uart:
  id: uart_0
  baud_rate: 115200
  rx_buffer_size: 384
  tx_pin: GPIO3
  rx_pin: GPIO1

jk_modbus:
  id: modbus0
  uart_id: uart_0

jk_bms:
  id: bms0
  jk_modbus_id: modbus0

#uart:
#  tx_pin: 27
#  rx_pin: 26
#  baud_rate: 256000
#  parity: NONE
#  stop_bits: 1

sensor:
  - platform: jk_bms
    min_voltage_cell:
      name: "jkbms_eth min voltage cell"
    max_voltage_cell:
      name: "jkbms_eth max voltage cell"
[... 158 lines cut ...]
    password:
      name: "jkbms_eth Password"
    device_type:
      name: "jkbms_eth Device type"
    software_version:
      name: "jkbms_eth Software version"
    manufacturer:
      name: "jkbms_eth Manufacturer"

binary_sensor:
  - platform: jk_bms
    balancing:
      name: "jkbms_eth balancing"
    balancing_switch:
      name: "jkbms_eth balancing switch"
    charging:
      name: "jkbms_eth charging"
    charging_switch:
      name: "jkbms_eth charging switch"
    discharging:
      name: "jkbms_eth discharging"
    discharging_switch:
      name: "jkbms_eth discharging switch"
    dedicated_charger_switch:
      name: "jkbms_eth dedicated charger switch"

magnetus26, 2023-08-10:
im on esphome  2023.6.4  using JK04 protocoland its working fine.
 will this become an issue if I update esphome to  2023.7.1?
thanks
Miguel

kev300, 2023-08-17:
Interesting. I'm running on 2023.7 for a month now and never had an issue. :grimacing: 

But I did change the scan interval back than...

`
esp32_ble_tracker:
  scan_parameters:
    interval: 500ms
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("JK-", 0) == 0) {
            ESP_LOGI("ble_adv", "New JK-BMS found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0

jk_bms_ble:
  - ble_client_id: client0
    protocol_version: ${protocol_version}
    throttle: 0s
    id: bms0
`

syssi, 2023-08-17:
I cannot reproduce the issue too.

magnetus26, 2023-08-18:
hi all have anyone tried the new esphome version 2023.8.1 to see if the issue has been solved?

thanks
Miguel

syssi, 2023-08-18:
I've updated my setup to `2023.8.1` today. Let's see! :-)

hippieua, 2023-08-21:
> I've updated my setup to `2023.8.1` today. Let's see! :-)

2023.8.2 installed.

Im using board: esp32-c3

Lots of Not connected messages in log

after adding 
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10" 

it became more stable

magnetus26, 2023-10-15:
hi all have anyone tried the new esphome version `2023.9.3` to see if the issue has been solved?

thanks
Miguel

syssi, 2023-10-15:
No critical issues on my side:

[image]

JohnGalt1717, 2023-10-16:
It appears to be much better now.

closed 2023-10-16 by JohnGalt1717